realsenseai/realsense-ros #1555: Invalid frame for RGB image in rs_camera.launch (/camera/color/image_raw not available)

kashishdhal, 2020-12-08:
Hello @doronhi,

With reference to #1523 and  #1536 I am creating this issue. I do not see **/camera/color/image_raw**  when I launch rs_camera.launch. However, when I change the resolution of the RGB image in the launch I am able to successfully see those topics.

Please see the output of **rs-enumerate-devices** which you asked me to do in the pull request. Please note that I am using an ARM computer (RPI 4 ), I don't know if that is making  a difference here.

<pre><font color="#8AE234"><b>ubuntu@gnclab</b></font>:<font color="#729FCF"><b>~</b></font>$ rs-enumerate-devices
Device info: 
    Name                          : 	Intel RealSense L515
    Serial Number                 : 	f0245848
    Firmware Version              : 	01.05.01.03
    Recommended Firmware Version  : 	01.05.02.00
    Physical Port                 : 	/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Product Id                    : 	0B64
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	L500
    Asic Serial Number            : 	0003acce764f
    Firmware Update Id            : 	0003acce764f

Stream Profiles supported by L500 Depth Sensor
 Supported modes:
    stream       resolution      fps       format   
    Confidence	  1024x768	@ 30Hz	   RAW8
    Confidence	  640x480	@ 30Hz	   RAW8
    Confidence	  320x240	@ 30Hz	   RAW8
    Infrared	  1024x768	@ 30Hz	   Y8
    Infrared	  640x480	@ 30Hz	   Y8
    Infrared	  320x240	@ 30Hz	   Y8
    Depth	  1024x768	@ 30Hz	   Z16
    Depth	  640x480	@ 30Hz	   Z16
    Depth	  320x240	@ 30Hz	   Z16

Stream Profiles supported by RGB Camera
 Supported modes:
    stream       resolution      fps       format   
    Color	  1920x1080	@ 30Hz	   RGB8
    Color	  1920x1080	@ 30Hz	   Y16
    Color	  1920x1080	@ 30Hz	   BGRA8
    Color	  1920x1080	@ 30Hz	   RGBA8
    Color	  1920x1080	@ 30Hz	   BGR8
    Color	  1920x1080	@ 30Hz	   YUYV
    Color	  1920x1080	@ 15Hz	   RGB8
    Color	  1920x1080	@ 15Hz	   Y16
    Color	  1920x1080	@ 15Hz	   BGRA8
    Color	  1920x1080	@ 15Hz	   RGBA8
    Color	  1920x1080	@ 15Hz	   BGR8
    Color	  1920x1080	@ 15Hz	   YUYV
    Color	  1920x1080	@ 6Hz	   RGB8
    Color	  1920x1080	@ 6Hz	   Y16
    Color	  1920x1080	@ 6Hz	   BGRA8
    Color	  1920x1080	@ 6Hz	   RGBA8
    Color	  1920x1080	@ 6Hz	   BGR8
    Color	  1920x1080	@ 6Hz	   YUYV
    Color	  1280x720	@ 60Hz	   RGB8
    Color	  1280x720	@ 60Hz	   Y16
    Color	  1280x720	@ 60Hz	   BGRA8
    Color	  1280x720	@ 60Hz	   RGBA8
    Color	  1280x720	@ 60Hz	   BGR8
    Color	  1280x720	@ 60Hz	   YUYV
    Color	  1280x720	@ 30Hz	   RGB8
    Color	  1280x720	@ 30Hz	   Y16
    Color	  1280x720	@ 30Hz	   BGRA8
    Color	  1280x720	@ 30Hz	   RGBA8
    Color	  1280x720	@ 30Hz	   BGR8
    Color	  1280x720	@ 30Hz	   YUYV
    Color	  1280x720	@ 15Hz	   RGB8
    Color	  1280x720	@ 15Hz	   Y16
    Color	  1280x720	@ 15Hz	   BGRA8
    Color	  1280x720	@ 15Hz	   RGBA8
    Color	  1280x720	@ 15Hz	   BGR8
    Color	  1280x720	@ 15Hz	   YUYV
    Color	  1280x720	@ 6Hz	   RGB8
    Color	  1280x720	@ 6Hz	   Y16
    Color	  1280x720	@ 6Hz	   BGRA8
    Color	  1280x720	@ 6Hz	   RGBA8
    Color	  1280x720	@ 6Hz	   BGR8
    Color	  1280x720	@ 6Hz	   YUYV
    Color	  960x540	@ 60Hz	   RGB8
    Color	  960x540	@ 60Hz	   Y16
    Color	  960x540	@ 60Hz	   BGRA8
    Color	  960x540	@ 60Hz	   RGBA8
    Color	  960x540	@ 60Hz	   BGR8
    Color	  960x540	@ 60Hz	   YUYV
    Color	  960x540	@ 30Hz	   RGB8
    Color	  960x540	@ 30Hz	   Y16
    Color	  960x540	@ 30Hz	   BGRA8
    Color	  960x540	@ 30Hz	   RGBA8
    Color	  960x540	@ 30Hz	   BGR8
    Color	  960x540	@ 30Hz	   YUYV
    Color	  960x540	@ 15Hz	   RGB8
    Color	  960x540	@ 15Hz	   Y16
    Color	  960x540	@ 15Hz	   BGRA8
    Color	  960x540	@ 15Hz	   RGBA8
    Color	  960x540	@ 15Hz	   BGR8
    Color	  960x540	@ 15Hz	   YUYV
    Color	  960x540	@ 6Hz	   RGB8
    Color	  960x540	@ 6Hz	   Y16
    Color	  960x540	@ 6Hz	   BGRA8
    Color	  960x540	@ 6Hz	   RGBA8
    Color	  960x540	@ 6Hz	   BGR8
    Color	  960x540	@ 6Hz	   YUYV

Stream Profiles supported by Motion Module
 Supported modes:
    stream       resolution      fps       format   
    Accel	 N/A		@ 400Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 200Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 100Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 400Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 200Hz	   MOTION_XYZ32F
    Gyro	 N/A		@ 100Hz	   MOTION_XYZ32F
</pre>

doronhi, 2020-12-14:
Thanks @kashishdhal for the info. I referred some people to this issue and hope to have an answer soon enough.

RealSenseSupport, 2021-01-04:
Hi @kashishdhal 

As per the l515 datasheet, 640x480 is not a supported RGB resolution.

https://dev.intelrealsense.com/docs/lidar-camera-l515-datasheet

Thanks

kashishdhal, 2021-01-05:
Hello @RealSenseSupport ,

Thanks for the reply! This is the reason that I opened a pull request #1536 to update the launch file but it was closed by @doronhi

I was under the impression that it may help other users. If this resolution is not supported please accept my pull request and replace the dimensions into launch file.

Thanks!
Kashish Dhal

doronhi, 2021-01-07:
@kashishdhal I understand now that you are correct, as well as @krazycoder2k was back in #1297 .
I prefer a more generic than #1536 so I submitted #1614.
I hope you will approve.

kashishdhal, 2021-01-07:
@doronhi , looks good to me, it solves the problem, approved from my side but you need to modify the launch file as well. There is no purpose we should have invalid frame dimensions in the launch file.

closed 2021-01-07 by kashishdhal

doronhi, 2021-01-11:
There is no point in keeping default values matching all the different devices when the librealsense2 SDK keeps track of them.
Removed all default profile value.