realsenseai/realsense-ros #1556: I want to call the callback function when messages with the same timestamp come from both rgb and depth topics

Hiroaki-K4, 2020-12-09:
Hi.
I want to call the callback function when messages with the same timestamp come from both rgb(/camera/color/image_raw
) and depth(/camera/depth/image_rect_raw) topics.
I set enable_sync and align_depth in rs_camera.launch to true. I then used message_filters to synchronize the two topics.
```
message_filters::Subscriber<sensor_msgs::Image> rgb_sub(nodeHandle_, cameraTopicName, cameraQueueSize);
message_filters::Subscriber<sensor_msgs::Image> depth_sub(nodeHandle_, depthTopicName, depthQueueSize);
message_filters::TimeSynchronizer<sensor_msgs::Image, sensor_msgs::Image> sync(rgb_sub, depth_sub, 10);
sync.registerCallback(&TkdnnObjectDetector::cameraCallback, this);
```
After doing this, I tried to run it, but the cameraCallback function was never called. Do you have any advice?
Thanks!

MartyG-RealSense, 2020-12-09:
Hi @Hiroaki-K4   There was a previous case where a RealSense user is also setting up a sync callback after using message_filters.  In their particular script, they call **async**.registerCallback instead of sync.  It would make logical sense to use async, since callbacks are **asynchronous**.

Hiroaki-K4, 2020-12-10:
Hi @MartyG-RealSense.
I chaged sync to async. But I got an error. which file should I include for async?
```
error: 'async' was not declared in this scope
```
Thanks!

MartyG-RealSense, 2020-12-10:
This is not one of my specialist knowledge subjects unfortunately.  The next thing to try may be to go back to the last known code that did not have errors (using sync instead of async) and then examining the structure of the code inside the brackets -  **(&TkdnnObjectDetector::cameraCallback, this)** - to see whether that is the cause of nothing happening when sync.regsterCallback is called.

For example, what happens if you drop the & sign from the callback:

**sync.registerCallback(TkdnnObjectDetector::cameraCallback, this);**

Hiroaki-K4, 2020-12-11:
ok. I solved.
I used to declare message_filters in the class init (), but when I moved it to main (), it worked.
Thank you!

closed 2020-12-11 by Hiroaki-K4

MartyG-RealSense, 2020-12-11:
Great news that you found a solution - thanks very much for the update!