realsenseai/realsense-ros #1563: How to check cloud-point is organized or not?

ranjitkathiriya, 2020-12-10:
To tell the difference between organized and unorganized point clouds, you can check if the isdense boolean variable of the incoming point cloud coming from the camera is set to true or not. If it is true, then it is organized, Otherwise, it is not.

ranjitkathiriya, 2020-12-10:
and 

2. How to find which ROS topic from your camera is publishing organized or depth-registered pointcloud.

MartyG-RealSense, 2020-12-10:
Hi @ranjitkathiriya   The link below describes that you should be able to know if a point cloud is unorganized or organized just from the type of roslaunch instruction that you are using (**rs_camera** for unorganized and **rs_rgbd** for organized).

https://github.com/IntelRealSense/realsense-ros/issues/893

The link below states that when using rs_rgbd, the organized cloud is available from the **depth_image_proc** nodelet.

https://github.com/IntelRealSense/realsense-ros/issues/851#issuecomment-516426314

Information about **depth_image_proc** can be found here:

http://wiki.ros.org/depth_image_proc

In regard to **rs_camera**, the RealSense ROS wrapper documentation's section on launch parameters states that using the **pointcloud** filter will add a pointcloud topic **/camera/depth/color/points**

ranjitkathiriya, 2020-12-10:
Then why in is_dense: True for launching rs_camera.

roslaunch realsense2_camera rs_camera.launch filters:=pointcloud enable_infra:=true enable_accel:=true enable_gyro:=true enable_gyro:=true

[image]

rostopic echo /camera/depth/color/points # For checking cloudpoint details

MartyG-RealSense, 2020-12-10:
**is_dense** is set to true for Pointcloud2 if there are no invalid points, as defined in the link below:

http://docs.ros.org/en/api/sensor_msgs/html/msg/PointCloud2.html

An unorganized cloud is dense, so is_dense = true because it has no invalid points.  An organized cloud has invalid points and so is_dense = false.   The link below that discusses is_dense describes that in an organized cloud, invalid points are present because they are used to fill up space.

https://answers.ros.org/question/234455/pointcloud2-and-pointfield/?answer=234472#post-id-234472

ranjitkathiriya, 2020-12-10:
Thanks for the link and explanation about is_dense and organized cloud. Your explanation was really helpful to me. Thanks

closed 2020-12-10 by ranjitkathiriya