realsenseai/realsense-ros #1566: nodelets error - T265 and D435i

srnetsr, 2020-12-12:
Hi,

I am using separate `launch/includes/nodelet.launch.xml` to run two cameras; t265 and d435i

But there is an error;

WARNING [140459179501312] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x2, error: No such device, err. num: 19
ERROR [140459179501312] (tm-device.cpp:2036) Bulk request error RS2_USB_STATUS_NO_DEVICE
WARNING [140459397613312] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x2, error: No such device, err. num: 19
ERROR [140459397613312] (tm-device.cpp:2036) Bulk request error RS2_USB_STATUS_NO_DEVICE
WARNING [140459538319104] (messenger-libusb.cpp:42) control_transfer returned error, index: 0, error: No data available, number: 61
 
Is there anyone who can help?

srnetsr, 2020-12-14:
@doronhi @mattcurfman 
Sir, do you have any idea about this issue?
[Link](https://dev.intelrealsense.com/docs/ros-wrapper) is saying "Multiple camera is currently not supported for T265 cameras."
Could you confirm if for me?

Thank you.

doronhi, 2020-12-16:
Regarding the Link you provide, I'll have to check what exactly it refers to because obviously T265 and a D435i can work together and we even provide the `rs_d400_and_t265.launch` file to demonstrate and ease the process.

Is the T265 works when it's the only device connected? Does it work with realsense-viewer?

Did you install the udev rules and dkms?

srnetsr, 2020-12-16:
@doronhi 
After setting all correctly, I confirmed it is working with two cameras.
I think the `Link` seems to be old and need to be edited.
Sorry for the confusion.

closed 2020-12-16 by srnetsr