realsenseai/realsense-ros #1567: Control "/camera/depth/color/points" rate (ROS2)

SBarzz, 2020-12-14:
Hi,
I'm using realsense ros with D435i in ROS2.
Currently when setting:
```
        depth_fps: 30.0
        depth_height: 240
        depth_width: 424
```
The rate of the topic "/camera/depth/color/points" is ~10Hz.
(Tested using: ros2 topic hz /camera/depth/color/points)
The same configuration in ROS1 provided 30hz
Is there a way to increase the rate?
Thanks

MartyG-RealSense, 2020-12-14:
Hi @SBarzz   Could you provide information about your configuration please:

-  Computer / computing device
- Operating system
-  ROS2 version (Dashing, Eloquent, Foxy, etc)

Also, can you confirm please that you are using the **ROS2 Eloquent** wrapper.

https://github.com/IntelRealSense/realsense-ros/tree/eloquent

I ask this because there is an older ROS2 wrapper for RealSense that is no longer supported.

https://github.com/intel/ros2_intel_realsense

SBarzz, 2020-12-14:
> Hi @SBarzz Could you provide information about your configuration please:
> 
> * Computer / computing device
> * Operating system and version (e.g Ubuntu 20.04)
> * ROS2 version (Dashing, Eloquent, Foxy, etc)
> * Librealsense SDK version
> 
> Also, can you confirm please that you are using the **ROS2 Eloquent** wrapper.
> 
> https://github.com/IntelRealSense/realsense-ros/tree/eloquent
> 
> I ask this because there is an older ROS2 wrapper for RealSense that is no longer supported.
> 
> https://github.com/intel/ros2_intel_realsense

Computer / computing device: Both on Dell G5 and Jetson TX2
Operating system and version: Ubuntu 18.04
ROS2 version: Eloquent
Librealsense SDK version: 
RealSense ROS v3.1.1 - ROS2 Eloquent (From this repository)
LibRealSense v2.40.0

Thanks

MartyG-RealSense, 2020-12-15:
Hi @SBarzz  It is important to match the RealSense ROS wrapper version as closely as possible with the librealsense SDK recommended for that particular wrapper.  The release otes for ROS2 Eloquent wrapper 3.1.1 recommend librealsense **2.38.1** instead of the more recent 2.40.0 that you are using.

SBarzz, 2020-12-15:
> Hi @SBarzz It is important to match the RealSense ROS wrapper version as closely as possible with the librealsense SDK recommended for that particular wrapper. The release notes for ROS2 Eloquent wrapper 3.1.1 recommend librealsense **2.38.1** instead of the more recent 2.40.0 that you are using.
> 
> The recently released ROS2 Eloquent wrapper **3.1.2** supports SDK 2.40.0. I would recommend rebuilding the ROS2 wrapper using the 3.1.2 version.
> 
> https://github.com/IntelRealSense/realsense-ros/releases/tag/3.1.2

Upgraded to 3.1.2 with librealsense v2.40.0
But still the rate of "/camera/depth/color/points" is slower then ROS1:
See this image:

[image]

doronhi, 2020-12-16:
I also see that `ros2 topic hz` gives a lower rate than expected.
This [answers.ros.org issue](https://answers.ros.org/question/350753/ros2-topic-hz-provides-wrong-rate-for-larger-msgs/) may be relevant.
I added a script to check the rate of arriving messages in #1571.
Running it using `./topic_hz.py /camera/depth/color/points` I get the expected 30Hz.

SBarzz, 2020-12-16:
> I also see that `ros2 topic hz` gives a lower rate than expected.
> This [answers.ros.org issue](https://answers.ros.org/question/350753/ros2-topic-hz-provides-wrong-rate-for-larger-msgs/) may be [...]

Hi,
I used your script in order to check the rate.
The first measurement does start from 30 Hz but immediately after a few seconds the rate drops and stabilizes at 16 Hz.

[image]

I don't think the problem is in the measurement of "ros2 topic" because I see that there is an effect on the rates of algorithms that receive this data as input.

Btw,
These are the parameters I use

```
camera:
    camera:
      ros__parameters:
        device_type: ''
        serial_no: ''
        usb_port_id: ''
        
        align_depth: true
        initial_reset: true

        color_fps: 30.0
        color_height: 240
        color_width: 424

        depth_fps: 30.0
        depth_height: 240
        depth_width: 424
        
        infra_fps: 30.0
        infra_height: 240
        infra_width: 424

        gyro_fps: 400.0
        accel_fps: 250.0

        enable_accel: true
        enable_color: true
        enable_depth: true
        enable_gyro: true
        enable_infra1: false
        enable_infra2: false

        pointcloud_texture_index: 0
        pointcloud_texture_stream: RS2_STREAM_COLOR

        stereo_module:
          emitter_enabled: 1
        unite_imu_method: 'linear_interpolation'

        rosbag_filename: ''

```

doronhi, 2020-12-16:
You already use a very low image resolution. What kind of machine you use? Could it possibly be overloaded? Can you `top` and see the realsense2_camera's CPU usage? Do you have more than 1 machine on your ROS network?
What frame rate do you get with realsense-viewer? (press the little (i) icon on the right side of the image)

SBarzz, 2020-12-16:
> You already use a very low image resolution. What kind of machine you use? Could it possibly be overloaded? Can you `top` and see the realsense2_camera's CPU usage? Do you have more than 1 machine [...]

Actually now using your script I get 30 Hz, but I don't know what changed, maybe its cpu usage...
I'm using Jetson TX2.
The realsense2 cpu usage is ~90%

But still I'm having a problem that I have node which subscribes to this topic in a low rate. 
I calculated the time difference between two consecutive messages with this code:

```
      double dt = (rclcpp::Time(rosPc2->header.stamp) - rclcpp::Time(m_lastMessageTime)).nanoseconds();
      RCLCPP_INFO("dt : %f", dt);
      m_lastMessageTime = rosPc2->header.stamp;
```

And I'm getting  prints that dt = 233540700.0 nanoseconds ~= 0.23 seconds
I expect it to be around 0.033 (30 hz)
Any idea why?

Is there any reason why this publisher:
`_pointcloud_publisher = _node.create_publisher<sensor_msgs::msg::PointCloud2>("depth/color/points", 1);`

is not using the "rmw_qos_profile_sensor_data" like the other publishers?
Can it be related to my issue?

doronhi, 2020-12-20:
There is no reason that I can think of right now why the pointcloud publisher is not using "rmw_qos_profile_sensor_data" like image publishers. More over, I did find [some discussions](https://answers.ros.org/answers/354750/revisions/) according to which it can be related.

MartyG-RealSense, 2020-12-26:
Hi @SBarzz   Do you require further assistance with this case, please?   Thanks!

SBarzz, 2020-12-29:
> Hi @SBarzz Do you require further assistance with this case, please? Thanks!

No, thanks for the support!

closed 2020-12-29 by MartyG-RealSense