realsenseai/realsense-ros #1569: the timestamp of IMU have two kinds of base number, sometime the secs of timestamp is like 1608015601, sometime the secs of timestamp is like 3216031523

TouchDeeper, 2020-12-15:
Hello,
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.09.00 | 
| Operating System & Version |   Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0-128-generic             | 
| Platform                                 | PC |
| SDK Version                            |  2.40.0                      | 
| Language                            | C                         | 
| Segment			|  Robot                  | 
| installation Method			|  backend installation refer to https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md                  | 

I find a strange phenomenon. the timestamp of IMU has two kinds of the base number. sometimes I start the realsense node, the secs of the timestamp is like 1608015601, but there are also sometimes I start the realsense node, the secs of the timestamp is like 3216031523.  The `Global time` of Motion Module showed in `realsense-viewer` is like 1608015601, and I guess the timestamp like 1608015601 is what I want since if the timestamp is like 3216031523, I will have timestamp related error when using Kalibr to process camera-imu joint calibration. I have posted the [isue](https://github.com/ethz-asl/kalibr/issues/401) in Kalibr repository.
Could you please provide some advice to keep the secs of the timestamp of IMU in 1608015601 like number? Thans a lot.
The launch file I used to start the realsense node is:
```
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default="/home/td/realsense/realsense_ws/src/realsense-ros-2.2.20/realsense2_camera/conifg/kalibr.json"/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>

  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="false"/>

  <arg name="infra_width"         default="640"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="false"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="250"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="false"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
[... 215 lines cut ...]
[ INFO] [1608015987.214250726]: Motion Module was found.
[ INFO] [1608015987.214271857]: num_filters: 0
[ INFO] [1608015987.214279094]: Setting Dynamic reconfig parameters.
 15/12 15:06:27,248 WARNING [139851357734656] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
process[throttle_imu-6]: started with pid [14750]
process[rviz-7]: started with pid [14802]
[ INFO] [1608015987.662909437]: Done Setting Dynamic reconfig parameters.
[ INFO] [1608015987.663250072]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1608015987.663455853]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1608015987.663473843]: setupPublishers...
[ INFO] [1608015987.665978762]: Expected frequency for infra1 = 30.00000
[ INFO] [1608015987.688157299]: Expected frequency for infra2 = 30.00000
[ INFO] [1608015987.709051303]: Start publisher IMU
[ INFO] [1608015987.709581796]: setupStreams...
[ INFO] [1608015987.730978192]: insert Infrared to Stereo Module
[ INFO] [1608015987.731019356]: insert Infrared to Stereo Module
[ INFO] [1608015987.731035117]: insert Gyro to Motion Module
[ INFO] [1608015987.731046205]: insert Accel to Motion Module
 15/12 15:06:27,743 WARNING [139851357734656] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
[ INFO] [1608015987.827078781]: SELECTED BASE:Depth, 0
[ INFO] [1608015987.832709921]: RealSense Node Is Up!
 15/12 15:06:27,931 WARNING [139851357734656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 15/12 15:06:27,982 WARNING [139851357734656] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

```

MartyG-RealSense, 2020-12-15:
Hi @TouchDeeper   I hope that the comment I have linked to below and the couple of comments below it will answer your question.

https://github.com/IntelRealSense/realsense-ros/issues/1454#issuecomment-711172609

TouchDeeper, 2020-12-15:
@MartyG-RealSense 
Hi
Thanks for your reply. Adding `<arg name="initial_reset"             default="true"/>` solve my issue.
I want to know what the `initial_reset` do. In addition, I recommend you can write a parameter explanation document that explant what teach parameter do, this document will be very popular and useful. Thanks again.

MartyG-RealSense, 2020-12-15:
**initial_reset** performs a hardware reset of the camera at launch, similar to the **Hardware Reset** option in the RealSense Viewer and the **hardware_reset()** programming instruction in librealsense.

The RealSense ROS wrapper front page has a section with a list of launch commands that includes initial_reset:

https://github.com/IntelRealSense/realsense-ros#launch-parameters

TouchDeeper, 2020-12-16:
@MartyG-RealSense Thank you for your reply.

I'm sorry I forget the parameters page. I have calibrated the realsense d435i successfully.  I recorded this process and the problems encountered and solutions. To thank you, I shared this [document](https://github.com/TouchDeeper/BetaGo/blob/ztd/betago_calibration/doc/rs_d435i_calibration.md) in my repository. I hope it will help other people.

MartyG-RealSense, 2020-12-16:
Excellent resource - thanks very much for sharing that with the RealSense community    :)

Do you need assistance with anything else, please?   Thanks!

closed 2020-12-16 by TouchDeeper