realsenseai/realsense-ros #1570: Param '/t265/tracking_module/frames_queue_size' has value 256 that is not in range [0, 32]. Removing this parameter from dynamic reconfigure options.

srnetsr, 2020-12-15:
With running `rs_d400_and_t265.launch` using t265 and d435i, I have 

`Param '/t265/tracking_module/frames_queue_size' has value 256 that is not in range [0, 32]. Removing this parameter from dynamic reconfigure options.`

What is this mean?
Is there anyone know about it?

srnetsr, 2020-12-28:
@doronhi 
Could you advise me to resolve this issue?

doronhi, 2020-12-31:
The message means that the device has a default value that is incompatible with the allowed option's range.
I am not sure where the value of 256 comes from. I expect somewhere in librealsense2 but I'm not sure.
I am not aware of issues caused by it and you can fix this problem by adding the following lines to your launch file:
```
  <rosparam>
      /camera/tracking_module/frames_queue_size: 16
  </rosparam>
```
That will fix the value before the device is started.

Again, I could be wrong but I think you can just as well ignore the warning.

srnetsr, 2021-01-04:
Thanks.
If it is not causing any error, then I will ignore it.

closed 2021-01-04 by srnetsr