realsenseai/realsense-ros #1577: Noetic Release

fmessmer, 2020-12-17:
sorry, for opening yet another "Noetic Release" issue

I might be missing something, but `realsense2_camera` is still not released into `noetic`
only `librealsense` is released
http://repositories.ros.org/status_page/ros_noetic_default.html?q=realsense

there have been numerous issues before, but all of them were already closed - **without the packages being released**
- https://github.com/IntelRealSense/realsense-ros/issues/1214
- https://github.com/IntelRealSense/realsense-ros/issues/1423

MartyG-RealSense, 2020-12-17:
Hi @fmessmer   Official support for ROS1 Noetic was added to the RealSense ROS wrapper in wrapper version **2.2.0**.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.20

fmessmer, 2020-12-17:
I got that from the related issues...question remains whether `realsense2_camera` package will be available as debian package?

MartyG-RealSense, 2020-12-17:
I will refer the above question about Debian packages for 2.2.20 to @doronhi the wrapper developer then, as I am not involved in the development side of RealSense.

mirkocomparetti-synesis, 2020-12-18:
Hello, this will be useful for us as well.

Thanks!

doronhi, 2020-12-20:
The failing debian build is a very important issue and hopefully will be attended very shortly.

dgrnbrg, 2020-12-27:
This package is also important to us, as Realsense is supported on noetic.

fmessmer, 2021-01-04:
I see a new tag [`2.2.21`](https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.21) has been created which requires `librealsense2` in version `2.41.0` - will this combination be successfully released into `noetic`?
can someone give a status update or roadmap for the successful release of `realsense2_camera` into `noetic`?

MartyG-RealSense, 2021-01-04:
Hi @fmessmer   I will refer your question to @doronhi the RealSense ROS wrapper developer.

doronhi, 2021-01-04:
librealsense2 version 2.41.0 and realsense2_camera version 2.2.21 are in the process of being released into noetic.
These are the PRs, waiting to be merged and released:
https://github.com/ros/rosdistro/pull/27865
https://github.com/ros/rosdistro/pull/27868

MartyG-RealSense, 2021-01-10:
Hi @fmessmer   Do you require further assistance with this case, please?  Thanks!

fmessmer, 2021-01-11:
thanks, @doronhi @MartyG-RealSense 
I'll close this issue once the next noetic sync went through and the packages become available in `main`

kentevan27, 2021-01-18:
Are syncs typically monthly?  I'm waiting on this one to go through as well

fmessmer, 2021-01-18:
no clear cycle for the releases....but they are announced here: https://discourse.ros.org/c/release/16/l/latest

Plopao, 2021-01-21:
Marking the method #1 as "not working on Noetic" would be a pleasant addition until the solution is pushed.

To those still looking to install the package:
I had success installing and running the package with a simple git clone and catkin_make.

MartyG-RealSense, 2021-01-21:
Great to hear that your installation in https://github.com/IntelRealSense/realsense-ros/issues/1636 was successful @Plopao

fmessmer, 2021-01-21:
the packages is also already available in the `ros-testing` repository...it's really just missing a sync...

kentevan27, 2021-01-21:
There's a sync scheduled for tomorrow apparently

fmessmer, 2021-01-21:
:point_right: https://discourse.ros.org/t/preparing-for-noetic-sync-2021-01-22/18570 :tada:

fmessmer, 2021-01-27:
sync is out and thus the packages are available for noetic
thanks

closed 2021-01-27 by fmessmer