realsenseai/realsense-ros #1579: Is there any difference between the depth information of realsense-viewer and realsense-ros?

Hiroaki-K4, 2020-12-18:
Hi.
I am using realsense for 3D object detection. The data taken with the real sense-viewer is used as the training data. At the time of inference, we use the depth information from realsense-ros.
So I have a queation about depth information by realsense. Is there any difference between the depth information of realsense-viewer and realsense-ros? For example, Maximum depth distance, etc.
Thanks!

MartyG-RealSense, 2020-12-18:
Hi @Hiroaki-K4   Aside from technical differences in how RealSense ROS handles data compared to librealsense (since the ROS wrapper aims for data to be compliant with ROS standards), RealSense Viewer also applies a range of post-processing filters to the data by default.  

These can be turned off manually.  You can do so individually for each filter or entirely by left-clicking on the blue icons - signifying an On state - to turn them off) in the **Post Processing** section of the options side-panel under the **Stereo Module** section.  

The Viewer's post-processing filter set has a **Threshold Filter** that sets the minimum and maximum observable depth distance, excluding data from the image that is beyond this distance.  When enabled, this filter imposes a maximum depth distance of 4 meters by default.  

In RealSense ROS, filters have to be configured manually - for example, by defining them in the roslaunch instruction.

[image]

Hiroaki-K4, 2020-12-21:
@MartyG-RealSense 
Thank you for reply. I understood!

closed 2020-12-21 by Hiroaki-K4