realsenseai/realsense-ros #1581: Image delay when using global_time_enabled

r91andersson, 2020-12-18:
This is actually a copy from the original issue created in the liberalsense repo.

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.09.00 and 5.12.08.200 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    | Linux 4.9.140-tegra    | 
| Platform                                 | NVIDIA Xavier AGX  |
| SDK Version                            |  v2.40.0                          | 
| Language                            | C++                          | 
| Segment			|  Robot                          | 
| Jetpack Version             | v.4.4 |
| VL4 Version                   | L4T-32.4.3 |
| Backend Camera Driver installation | Native kernel with the patch applied as in the tutorial, RSUSB=False|
### Issue Description
When parameter **global_time_enabled** is enabled, we're experiencing a huge delay in retrieval of the images. The timestamp however doesn't say that the image is as old as it is in reality.
I've been testing and can confirm that it's the same for both firmware 05.12.09.00 and 5.12.08.200.

### Test Description:
We have a movable camera module (where the camera is attached onto) on a linear rail. This linear rail has a motor that can drive the whole camera module forward and backward at a constant speed. We have a digital encoder mounted on the linear rail motor that has a resolution of 0.0000469398m/tick. In this way, we can save the absolute position of the camera module for each image taken.

### Test Procedure:
**Step 1.**
We drive the camera module to a known absolute encoder position equal to 6500. We refer and name this position to **encoder_trigger_image_position**

**Step 2.**
When the camera module is still at the **encoder_trigger_image_position**, we take and save an image. This image is named and will be refered to our **ground truth image**.

**Step 3.**
We drive the camera module to the intial position equal to 0. 

**Step 4**
We set the camera module in a linear motion by setting the speed to 1000 tick/s (=0.0469398023 m/s).

**Step 5**
Just when the camera module passes the **encoder_trigger_image_position** we grab the latest image and save this image and name it **speed_1000**. We are aware of that we can potentially miss 1 sample here, but the fact that we're sampling the camera stream at 30Hz, we shouldn't expect too much drift.

**Step 6**
We repeat the steps 3-6, but change the speed to 2000 (=0.0938796047 m/s)

**Step 7**
We do an ocular inspection of the 3 images.
The **ground truth image** is where we want the other images to be as close as possible to. 
We have a green pin sticking up from the soil, so we can easily see if the images are close to each other by comparing this pin in each of the images to the  **ground truth image**.

### Test Result:
**Test run 1:** 
When we run the test with the parameter **global_time_enabled** **enabled**, we can see a huge drift in the images compared to the ground truth image.
Ground truth:
[image]
Speed 1000:
[image]
Speed 2000:
[image]

**Test run 2**
When we run the test with the parameter **global_time_enabled** **disabled**, we see a much smaller drift in the images compared to the ground truth image.
Ground truth:
[image]
Speed 1000:
[image]
Speed 2000:
[image]

It's much more obvious and clear when downloading the images and then previewing them fast, by looking at ground truth image, then fast switch to speed_1000, then last speed_2000.

The camera config we ran the setup with (only difference is we toggled the Global Time Enabled):
[image]

### Expected Result:
It should be no/very little difference (when looking at drift of the green pin compared to the ground truth image) between running test run 1 and test run 2

doronhi, 2020-12-20:
Thanks for your elaborate tests, described both here and in your post [IntelRealSense/librealsense#8003](https://github.com/IntelRealSense/librealsense/issues/8003)
I added an inner bug report, DSO-16278, although it may take me a while to attend.
In the meanwhile, could you note what version of realsense2_camera you use.
Trying to understand the test itself, the point **encoder_trigger_image_position** is at 6500 ticks (i.e. 0.3051087m), right? So to take the image marked as **Speed 1000** you start the camera moving at  time t0 (ROS time) and then save the image with the inner timestamp of t1=t0+6.5(sec) (at 1000 ticks/sec)? Or something else?
If so, do you take the acceleration rate into account? How did you calculate the 200ms offset (mentioned in [your remark](https://github.com/IntelRealSense/librealsense/issues/8003#issuecomment-747054148) - Did you look which image most closely resembles **ground truth image** and marked it's timestamp compared to the expected timestamp?
I'm sure it's all taken care of as there is a difference between runs with global timestamp enabled being the only difference. I just want a thorough understanding as I'll need to construct my own test somehow.
The global timestamp converts the HW timestamp of the camera to the system timestamp of the computer. Do you use a single computer for controlling both the movable-camera-module and the the camera?

r91andersson, 2021-01-07:
@doronhi Sorry for the late reply, Christmas holiday!

You're almost right, but the speed is not expressed in ticks/s, but rather RPM.

The librealsense SDK version is v.2.40.0, and the ros-wrapper version 2.2.20.

To cancel out the acceleration factor, I have the encoder trigger position at a point when the speed is stable (=almost zero acceleration). 

The 200ms offset is just a rough calculation. I measured the offset for the pin in the different images, I've noticed that the offset roughly is about 2cm for the speed 2000 RPM (=0.0938796047 m/s) which roughly translates to the 200ms delay.

I've been using two computers for this setup. One that controls the movable camera rig, and one (Xavier) that runs the camera drivers. However, I have both computers synced with chronyc, where the computer that controls the rig acts as the master, and the Xavier (where the cameras are connected to) acts as the client that syncs against the master. I have also verified that the sync is correct and no drift is seen between the two systems.

doronhi, 2021-02-11:
I made a time modification in #1684. Does it make a difference in your case?

RealSenseSupport, 2021-03-09:
@r91andersson , has the latest update resolved your issue? I'll close this issue unless your problem still exists. Should the problem arise again later, please open another issue.

RealSenseSupport, 2021-03-12:
Hello @r91andersson, I'll close this today but if this or any other issue arises please feel free to open another ticket.

closed 2021-03-12 by RealSenseSupport