realsenseai/realsense-ros #1582: topic of realsense l515 loss a lot of data !

zhongxiangzju, 2020-12-18:
Hi, I try to record rgbd images of a moving object using realsense l515. After I launch sensor using
`roslaunch realsense2_camera rs_camera.launch`, topics are listed. But the frame rate is not matched with my setting which is 30 fps. Here is a screenshot of ros output.
[image]
Is it a bug ?

zhongxiangzju, 2020-12-19:
Name of the subscribed topic is `/camera/aligned_depth_to_color/image_raw`, and output is from `rostopic hz /camera/aligned_depth_to_color/image_raw`

doronhi, 2020-12-20:
It seems by the realsense2_camera log that exposure time value is set to 2000. That is 200(milliseconds) which means approximately 5 images per second.

xiaotaw, 2020-12-30:
@zhongxiangzju  I got 14 Hz with exposure time = 2000, and 27 Hz with exposure time = 166.

closed 2021-04-02 by RealSenseSupport