realsenseai/realsense-ros #1583: Missing packages for ubuntu 20.04 focal - noetic

mirkocomparetti-synesis, 2020-12-18:
Hello,
Running `sudo apt install ros-noetic-realsense2-description ros-noetic-realsense2-camera` on a new ubuntu desktop 20.04 machine with ROS noetic fails with
```
E: Impossible to find package ros-noetic-realsense2-description
E: Impossible to find package ros-noetic-realsense2-camera
```
while on ubuntu 18.04 with melodic it works just fine.
Which repo am I missing?

Thanks!

MartyG-RealSense, 2020-12-18:
Hi @mirkocomparetti-synesis   It sounds as though your problem is related to one that was posted on this forum yesterday.

https://github.com/IntelRealSense/realsense-ros/issues/1577

mirkocomparetti-synesis, 2020-12-18:
You are right. Sorry I didn't see it as I was looking for "20.04"...

Then this can be marked as duplicate.

Thanks

MartyG-RealSense, 2020-12-18:
You are very welcome.   I will close this case as a duplicate with your consent then.  Thanks very much!

closed 2020-12-18 by MartyG-RealSense