realsenseai/realsense-ros #1585: T265 Wrong coordinate frame movement

iamsavva, 2020-12-20:
I am using T265 cameras in ROS, and I have an odd relationship between the `odom_frame` and the `pose_frame`.

`odom_frame` is static, seems to be chosen based on gravity. When I pull up rviz and select `odom_frame` as fixed frame, blue (z) axis of the `odom_frame` is pointing out of the plane. Originally, with the T265 camera on the table vertically, `odom_frame` and `pose_frame` coincide.

according to this image:
[image]

x-axis should be towards the right camera. However, if I move the camera in this direction, in rviz the `pose_frame` moves in the (-y) direction (opposite of the green arrow) relative to the `odom_frame`. Shouldn't it move in the x axis though, along the red arrow?

If I move the camera up (which according to the image is positive y) in rviz `pose_frame` moves in positive z.
If I move the camera back (which according to the image is positive z) in rviz `pose_frame` moves in negative x.

Am I missing something? My tf tree is untouched, I only `roslaunch` the camera with `rs_t265.launch`. Would appreciate any help!

iamsavva, 2020-12-21:
These observations are compliant with REP-103, but not the images provided on all intel websites. I guess this has been updated at some point to match ROS conventions?

RealSenseSupport, 2021-02-15:
Hi,

I believe you are correct. The images are probably outdated. Thanks for catching this.

Will you be needing further help with this?

Thanks

iamsavva, 2021-02-27:
> Hi,
> 
> I believe you are correct. The images are probably outdated. Thanks for catching this.
> 
> Will you be needing further help with this?
> 
> Thanks

no, that's all, I'm closing the issue. But it would be quite useful if somebody was to at least link to this issue or make a note on those old posts that the data there is outdated.

closed 2021-02-27 by iamsavva