realsenseai/realsense-ros #1592: required clearing for depth related question

ranjitkathiriya, 2020-12-22:
Hi,

The command I have runed:
`roslaunch realsense2_camera rs_rgbd.launch filters:=colorizer align_depth:=true`

1. I would like to get depth in mm(Millimetre) of a given pixel-like X and Y?
2. How to consider the minimum range of real sense?
3. What is min and max value over here?
4. I have tried this code: correct if I have committed any mistake 
```
#!/usr/bin/env python
import rospy
from cv_bridge import CvBridge, CvBridgeError
from sensor_msgs.msg import Image
import numpy as np
import cv2

def convert_depth_image(ros_image):
    bridge = CvBridge()
    # Use cv_bridge() to convert the ROS image to OpenCV format
    try:
        # Convert the depth image using the default passthrough encoding
        depth_image = bridge.imgmsg_to_cv2(ros_image, desired_encoding="passthrough")
        depth_array = np.array(depth_image, dtype=np.float32)
        center_idx = np.array(depth_array.shape) / 2

        print('center depth:', depth_array[int(center_idx[0]), int(center_idx[1])]) # Changed Over here
        #
        # cv2.imshow("Depth_Sub", depth_image)
        # if cv2.waitKey(1) & 0xFF == ord('q'):
        #     rospy.signal_shutdown('Quit')
        #     cv2.destroyAllWindows()

    except CvBridgeError:
        print('Error!!')
    # Convert the depth image to a Numpy array

def pixel2depth():
    rospy.init_node('pixel2depth', anonymous=True)
    rospy.Subscriber("/camera/aligned_depth_to_color/image_raw", Image, callback=convert_depth_image, queue_size=1)
    rospy.spin()

if __name__ == '__main__':
    pixel2depth()
```

I have changed this line : print('center depth:', depth_array[int(center_idx[0]), int(center_idx[1])]) # because of flot issue while dividing by 2.

5. The output I am receiving is in the list form example like: 
center depth: [225. 255.  29.]
center depth: [213. 255.  41.]
center depth: [224. 255.  30.]
center depth: [225. 255.  29.]
center depth: [228. 255.  26.]
center depth: [223. 255.  31.]
center depth: [219. 255.  35.]
center depth: [216. 255.  38.]
center depth: [213. 255.  41.]

What is this please provide more clarification on this above lists and How can I get the distance in mm? 

Thanks and Regards

MartyG-RealSense, 2020-12-22:
Hi @ranjitkathiriya   The tutorial in the link below that uses Python and ROS provides a method for converting 2D xy to 3D xyz that may be simpler:

https://medium.com/@yasuhirachiba/converting-2d-image-coordinates-to-3d-coordinates-using-ros-intel-realsense-d435-kinect-88621e8e733a

ranjitkathiriya, 2020-12-22:
Actually, I found this code in this issue and It was uploaded by @doronhi. 

https://github.com/IntelRealSense/realsense-ros/issues/714

In this comment so I want an explanation of the outputs: like what is this :: center depth: [213. 255. 41.]
https://github.com/IntelRealSense/realsense-ros/issues/714#issuecomment-561036465

MartyG-RealSense, 2020-12-22:
I guess you are referring to the edited **my_depth_subscriber.txt** text file version of the original script that was provided by @doronhi 

https://github.com/IntelRealSense/realsense-ros/issues/714#issuecomment-561036465

Apparently the values that are output from the script are in millimeters once converted to OpenCV format (e.g 231 X x 255 Y  x 41 Z mm).

ranjitkathiriya, 2020-12-22:
Okay, That's the explanation I wanted to thank you so much. 

here Z is in form of mm or I have to apply some OpenCV Technique and calculate mm from these three coordinates?

MartyG-RealSense, 2020-12-22:
I have checked the code many times but I cannot be sure whether the XYZ value output after conversion to OpenCV is in mm or needs to undergo further conversion to display the distance in mm.   

@doronhi   Can you advise on this please?    Thanks!

doronhi, 2020-12-23:
From the README.md file:
> colorizer: will color the depth image. On the depth topic an RGB image will be published, instead of the 16bit depth values .

So, while using the `filter:=colorizer` option, you no longer get the depth but RGB values.
Remove that filter and you'll get depth in millimeters.

ranjitkathiriya, 2020-12-23:
Hi @doronhi,

Thanks for providing me the explanation. I will check and update you back thanks

closed 2020-12-28 by ranjitkathiriya

ranjitkathiriya, 2021-01-08:
Thanks @doronhi it's working