realsenseai/realsense-ros #1596: I want to publish rgb data and depth data from a bag file.

Hiroaki-K4, 2020-12-23:
Hi.
I want to publish rgb data and depth data from a bag file recorded using realsense-viewer. Is there a good way to receive a file like [image_publisher]() and publish rgb and depth data?
Thanks!

MartyG-RealSense, 2020-12-23:
Hi @Hiroaki-K4   There are differences between bag files recorded in ROS and bag files recorded in the RealSense SDK.  This means that an SDK-recorded bag is not guaranteed to play back in ROS, and vice versa.

https://github.com/IntelRealSense/librealsense/issues/3020#issuecomment-452321766

If you are using ROS, you could record topics published by the RealSense ROS wrapper into a rosbag file within ROS using the **rosbag record** function.

https://wiki.ros.org/rosbag/Tutorials/Recording%20and%20playing%20back%20data

Doronhi the RealSense ROS wrapper developer provides an example in the link below of usng rosbag record to record and play back topics published by the RealSense ROS wrapper:

https://github.com/IntelRealSense/realsense-ros/issues/1076#issuecomment-586702026

As Doronhi mentions in the above link, you should be able to view the played-back data in the **RViz** interface.

MartyG-RealSense, 2020-12-30:
Hi @Hiroaki-K4   Do you require further assistance with this case, please?   Thanks!

Hiroaki-K4, 2021-01-05:
Hi @MartyG-RealSense 
Sorry for the late reply. I understand how to record and play back with rosbag. Thank you!

closed 2021-01-05 by Hiroaki-K4

MartyG-RealSense, 2021-01-05:
Great to hear - thanks for the update!