realsenseai/realsense-ros #1598: realsense2_camera: Cannot locate rosdep definition for [ament_lint_common]

paddygoat, 2020-12-29:
Description:	Ubuntu 20.04.1 LTS
ROS2 Eloquent

tegwyn@tegwyn-CF-H2AS01EDA:~/ros2_ws$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Add distro "dashing"
Skip end-of-life distro "eloquent"
Add distro "foxy"
Skip end-of-life distro "groovy"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Add distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in /home/tegwyn/.ros/rosdep/sources.cache
tegwyn@tegwyn-CF-H2AS01EDA:~/ros2_ws$ rosdep install -i --from-path src --rosdistro $ROS_DISTRO -y
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
realsense2_camera: Cannot locate rosdep definition for [ament_lint_common]
realsense2_description: Cannot locate rosdep definition for [ament_lint_common]
realsense2_camera_msgs: Cannot locate rosdep definition for [ament_lint_common]
tegwyn@tegwyn-CF-H2AS01EDA:~/ros2_ws$

MartyG-RealSense, 2020-12-30:
Hi @paddygoat    Given that you are using Ubuntu **20.04** instead of 18.04, would it be possible for you to upgrade your ROS2 version from Eloquent to **Foxy**?   I ask this because the first RealSense ROS2 wrapper for Foxy and Ubuntu 20.04 has just been released for use with librealsense **2.41.0**.

https://github.com/IntelRealSense/realsense-ros/tree/foxy

paddygoat, 2020-12-30:
Yes, I was kind of hoping that this would be the solution. Many thanks @MartyG-RealSense

paddygoat, 2021-01-01:
Hi Marty !!!
I installed and tested ROS2 Foxy.
Followed instructions to install wrapper: 
Step 3: Install Intel® RealSense™ ROS2 wrapper from Sources
Re-boot.
Colcon build fail:

tegwyn@tegwyn-CF-H2AS01EDA:~$ cd ~/ros2_ws
tegwyn@tegwyn-CF-H2AS01EDA:~/ros2_ws$ colcon build
Starting >>> realsense2_camera_msgs
--- stderr: realsense2_camera_msgs                         
CMake Error at CMakeLists.txt:13 (find_package):
  By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "ament_cmake", but CMake did not find one.

  Could not find a package configuration file provided by "ament_cmake" with
  any of the following names:

    ament_cmakeConfig.cmake
    ament_cmake-config.cmake

  Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
  "ament_cmake_DIR" to a directory containing one of the above files.  If
  "ament_cmake" provides a separate development package or SDK, be sure it
  has been installed.
---
Failed   <<< realsense2_camera_msgs [0.40s, exited with code 1]
                                
Summary: 0 packages finished [1.10s]
  1 package failed: realsense2_camera_msgs
  1 package had stderr output: realsense2_camera_msgs
  2 packages not processed
tegwyn@tegwyn-CF-H2AS01EDA:~/ros2_ws$

MartyG-RealSense, 2021-01-01:
Hi @paddygoat   Could you tell me the computer or computing device that you are using please so that I can advise the appropriate installation method for librealsense 2.41.0? For example, if you are using a PC then you can build librealsense for Ubuntu 20.04 using the instructions in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

Once the new librealsense version has been installed then you should rebuild the RealSense ROS2 Foxy wrapper.

paddygoat, 2021-01-02:
@MartyG-RealSense  .... I am using a Panosonic CF-H2 PC running Ubuntu 20.04.
I used the link : https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md to install librealsense 2.40.0 with no problems and tested ok.
The problem I now have is that it needs to be upgraded to librealsense 2.41.0 for compatibility with ROS2 and Ubuntu 20.04 and  librealsense 2.41.0 is only in '**pre-release**' and not '**release**'. I'm not expert with Github, so need to know how to upgrade using the pre-release please !!!!
Many thanks and happy new year.

MartyG-RealSense, 2021-01-02:
You can build librealsense 2.41.0 with **CMake** from source code instead of packages using the **RSUSB** installation method.  This method requires an internet connection but is not dependent on Linux versions or kernel versions and does not require patching.

Instructions for the RSUSB method can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

For installation on your PC, use the instruction below for step 5 to install librealsense without including Python.

**cmake ../ -DFORCE_RSUSB_BACKEND=ON  -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

closed 2021-01-16 by MartyG-RealSense

MartyG-RealSense, 2023-09-28:
Hi @jhordanhiro  The ROS wrapper link that you posted leads to the ROS2 version of the RealSense ROS wrapper.  If you are using ROS1 Noetic then you should build the ROS1 wrapper at the link below.

https://github.com/IntelRealSense/realsense-ros/tree/ros1-legacy

However, the official ROS1 wrapper does not support D405 (the ROS2 wrapper for Humble and Iron does).  There is a custom 'forked' version of the wrapper with D405 support created by a RealSense user at the link below.

https://github.com/rjwb1/realsense-ros/tree/development

It is also possible to modify the official ROS1 wrapper to add D405 support using a method described at https://github.com/IntelRealSense/realsense-ros/issues/2582#issuecomment-1371133133