realsenseai/realsense-ros #1600: Issue of the inter_cam_sync_mode

Joeyyzhao, 2020-12-31:
Hi, I try to enable the RS2_OPTION_INTER_CAM_SYNC_MODE on the l515, but I found that the depth topic message (topic name:/camera/depth/image_rect_raw) always miss some data periodically (throw "no message" when run "rostopic hz" command), no matter what frequency of trigger signal I give (1HZ~10HZ I tried).

RealSenseSupport, 2021-01-08:
Hi @Joeyyzhao 

Can you help us understand if you've setup multiple L515 cameras at the same time and you've implemented the multicam setup as outlined in this guide:  https://www.intelrealsense.com/lidar-camera-l515-multi-camera-configuration/ , https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup?_ga=2.226724253.1687202588.1610048669-1479303487.1578600965

RealSenseSupport, 2021-01-14:
Hi @Joeyyzhao

Can you help us understand if you've setup multiple L515 cameras at the same time and you've implemented the multicam setup as outlined in this guide: https://www.intelrealsense.com/lidar-camera-l515-multi-camera-configuration/ , https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup?_ga=2.226724253.1687202588.1610048669-1479303487.1578600965

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-21 by RealSenseSupport