realsenseai/realsense-ros #1603: Pixels of the camera image are not constant

Schloern93, 2021-01-02:
Hello,
I use Ros eloquent.
I make an object detection using the topic "/D435/color/image_raw" and border the found objects with bounding boxes.
Now I want to use the depth information of the D435 camera to detect the depth of each detected bounding box. 
I try to do this using the topic ""/D435/depth/color/points"".

Idea: 
1. the camera image was set as follows.
- Width: 640
- Height: 480
2. the detected bounding boxes return me the pixels at which they were drawn in the camera image (2D boxes) (used topic "/D435/color/image_raw").
2. now I try to find the same pixels by using the PointCloud2 message to get the depth information. Through the PointCloud2 message I can then detect the depth of the pixels and enter this as additional information in the bounding boxes so that I can display them in 3D space.
3. 
Here is my callback to evaluate the PointCloud2 message:

      void callback_depth_image_d435 (const sensor_msgs::msg::PointCloud2::SharedPtr point_cloud2_msgs)
      {
      //sensor_msgs::msg::PointCloud2& cloud = ;
      pcl::PointCloud<pcl::PointXYZ> point_cloud;

      // convert point clouds2 msgs data to points
      pcl::fromROSMsg(*point_cloud2_msgs, point_cloud);

      // resize vector every callback
      point_cloud2_points.resize(point_cloud.points.size());     

      for(unsigned long i = 0; i < point_cloud.points.size(); i++)
      {
        // get points
        point_cloud2_points.at(i).x = point_cloud.points.at(i).x;
        point_cloud2_points.at(i).y = point_cloud.points.at(i).y;
        point_cloud2_points.at(i).z = point_cloud.points.at(i).z;
      }
    }

The problem now is that the number of points from the PointCloud2 message keeps changing (I expected the number of pixels to remain), so I have the same number of points in the PointCloud message all the time. 
Is there a way to infer in the PointCloud2 message the same pixel I found in the topic "/D435/color/image_raw"?
Also, since I haven't worked with PointClouds that much, I would be interested to know why the number of points changes conditionally.

Thanks for the help in advance

MartyG-RealSense, 2021-01-02:
Hi @Schloern93   Are you using **roslaunch realsense2_camera rs_camera.launch** as your roslaunch instruction or are you using **roslaunch realsense2_camera rs_rgbd.launch** please?

**rs_camera** will provide an "unorganized" point cloud and **rs_rgbd** will provide an "organized" point cloud that is more efficient and has one-to-one mapping between a pixel and a point.

https://answers.ros.org/question/234455/pointcloud2-and-pointfield/?answer=234472#post-id-234472

Schloern93, 2021-01-04:
Thanks for the quick reply. Your link helped a lot to see more about PointClouds.
I use the Ros eloquent branches. I have only the following launch files available.
- rs_d400_and_t265_launch.py
- rs_launch.py
- rs_multi_camera_launch.py

Is it possible to organize the PointCloud message by setting a parameter?

MartyG-RealSense, 2021-01-04:
The documentation for the RealSense ROS2 Eloquent wrapper states that a point cloud can be enabled using the launch instruction below:

**ros2 launch realsense2_camera rs_launch.py enable_pointcloud:=true**

****************

This adds the pointcloud topic **/camera/depth/color/points**

The documentation adds:: "The depth FOV and the texture FOV are not similar. By default, pointcloud is limited to the section of depth containing the texture.  You can have a full depth to pointcloud, coloring the regions beyond the texture with zeros, by setting **allow_no_texture_points** to true"

Schloern93, 2021-01-04:
Hi, @MartyG-RealSense Thanks for the quick reply.
I have already tried that. I use the launch file "rs_d400_and_t265_launch.py"

And these are the parameters I have set in the rs_d400_and_t265_launch.py
Index 
D435 -> 1
T265 -> 2

local_parameters = [
{'name': 'camera_name1',           'default': 'D435','description': 'camera unique name'},
{'name': 'device_type1',                'default': 'd4.', 	'description': 'choose device by type'},
{'name': 'camera_name2',                'default': 'T265', 	'description': 'camera unique name'},
{'name': 'device_type2',                'default': 't265', 	'description': 'choose device by type'},
{'name': 'enable_fisheye12',            'default': 'false', 'description': 'topic for T265 wheel odometry'},
{'name': 'enable_fisheye22',            'default': 'false', 'description': 'topic for T265 wheel odometry'},
{'name': 'color_width1',                'default': '640', 'description': 'color image width'},
{'name': 'color_height1',               'default': '480', 'description': 'color image height'},
{'name': 'enable_color1',               'default': 'true', 'description': 'enable color stream'},
{'name': 'depth_width1',                'default': '640', 'description': 'depth image width'},
{'name': 'depth_height1',               'default': '480', 'description': 'depth image height'},
{'name': 'enable_depth1',               'default': 'true', 'description': 'enable depth stream'},
{'name': 'enable_pointcloud1',          'default': 'true',	'description': 'enable pointcloud'},
{'name': 'allow_no_texture_points1',    'default': 'true', 'description': ''},
 ]

The PointCloud message I receive through the topic "/D435/depth/color/points" is still an unsorted PointCloud2 message. Did I forget anything else?
Thanks in advance

Schloern93, 2021-01-05:
Is this a common mistake? Then I would open a new issue.

MartyG-RealSense, 2021-01-05:
I have referred your case to the RealSense team at Intel.  I appreciate your patience in the meantime.

doronhi, 2021-01-07:
The pointcloud published by realsense2_camera node in unorganized.
There are plans to add the option to publish an organized point cloud. It should be very soon but I can't commit right now.
In the ROS1 environment there is node, activated by the rs_rgbd.launch file, that registers to the aligned-depth and color images and publishes an organized point cloud. Maybe it is available in eloquent as well and a launch file can be created - I didn't check.

Schloern93, 2021-01-12:
Thanks for the answer. I have now solved my problem by using the package "depth_image_proc" to get my sorted PointCloud. 
https://github.com/allenh1/image_pipeline/tree/eloquent
This works so far quite well for me.

MartyG-RealSense, 2021-01-12:
Thanks very much @Schloern93 for the update - it is great to hear that you achieved a solution!

closed 2021-01-15 by RealSenseSupport