realsenseai/realsense-ros #1604: (synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!

srnetsr, 2021-01-04:
Do you have any idea about this error?

(synthetic-stream.cpp:48) Exception was thrown during user processing callback: Out of frame resources!

What is this and what cause this problem?

RealSenseSupport, 2021-01-05:
@srnetsr Which camera you are working with?

srnetsr, 2021-01-14:
@RealSenseCustomerSupport 
As you can see above,

srnetsr, 2021-01-14:
Hi, @MartyG-RealSense 
I saw your reply in #1161 
You mentioned this error message and  reducing frame rate and resolution as a solution.
FYI, I am not using NVIDIA machienes.

Please refer to the above parameter values and advise me to resolve this issue?

MartyG-RealSense, 2021-01-15:
Hi @srnetsr   Looking at your code in the opening comment, I note that you have set **infra_fps** to a value of **45**.  45 FPS is not supported as a default option on RealSense cameras.  On a D435 camera on a USB 3 connection, the supported FPS settings for infrared at 640x480 resolution are:

**6, 15, 30, 60, 90**

[image]

Could you therefore try setting infra_fps to '30' please?

MartyG-RealSense, 2021-01-29:
Case closed due to no further comments received.

closed 2021-01-29 by MartyG-RealSense

AndreV84, 2021-03-03:
we can not get ros2 color topic passing anything with the error like that
it streams the depth, however
```
(synthetic-stream.cpp:48) Exception was thrown during user processing callback: 1kernel_oth

MartyG-RealSense, 2021-03-03:
Hi @AndreV84   If you are using ROS2, have you checked the list of topics being published with **ros2 topic list** to see whether the color topics are on the list?

AndreV84, 2021-03-03:
yes, the color topic was published, but empty

MartyG-RealSense, 2021-03-04:
@AndreV84   Can you confirm whether you are using the current RealSense ROS2 wrapper please and not the old obsolete ROS2 wrapper in the link below that is no longer supported.

https://github.com/intel/ros2_intel_realsense

AndreV84, 2021-03-05:
@MartyG-RealSense 
yes, but we run from docker, also on arm64

MartyG-RealSense, 2021-03-05:
@AndreV84   I researched your problem carefully but could not find a solution.  Is the problem limited to just color or do other types of stream such as depth have empty topics too?

AndreV84, 2021-03-05:
it goit addressed by adding more swap memory
as it turned out it was due to cuda memory allocation error it seems without extra swap it was posible to get only part of topics published

MartyG-RealSense, 2021-03-05:
Okay, great news that you were able to find a solution!  Thanks for the update @AndreV84

srnetsr, 2021-03-05:
@MartyG-RealSense 
I am sorry for my late response.
I am using **90** fps for infra_fps, and tried small numbers as well.
However, the error keep displaying..
I am just ignoring this message, but I do not know why it is coming up.

MartyG-RealSense, 2021-03-06:
Hi @srnetsr  Could you test whether resetting the camera at launch makes a difference please by adding the command below to the end of your roslaunch instruction:

**initial_reset:=true**

mzahana, 2021-06-13:
@AndreV84 & @MartyG-RealSense 
I am getting the same error (or similar)
```
(synthetic-stream.cpp:48) Exception was thrown during user processing callback:
```
**setup**
* Camera: D435i with firmware `5.12.7` (I also tried the latest D435i firmware `5.12.13.5` as well as `5.12.11`
* Using Xavier NX with `l4t-r32.5.0` 
* Using a docker container with Nvidia base image `l4t-r32.5.0` (inspired by [jetson-containers](https://github.com/dusty-nv/jetson-containers) )
* Compiled libearlsense `v2.45.0`  (with `-DFORCE_RSUSB_BACKEND=ON -DBUILD_WITH_CUDA=true `) and realsense-ros `2.3.0` from source
* I run the default `rs_camera.launch`

I get a continuous stream of the above error. The RGB ROS topic `/camera/color/image_raw` is there but empty, but the depth image topic is there and has messages at 30Hz.

When I use `realsense_viewer`, it seems RGB and depth images are being streamed, but the motion module is not recognized.

I am not sure what the problem is with the ROS side. Any help is appreciated

MartyG-RealSense, 2021-06-13:
Hi @mzahana  

@AndreV84 has a greater knowledge of Docker than I do.

However, please try adding **initial_reset:=true** to your roslaunch instruction to see whether it makes a positive difference.  For example:

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true**

mzahana, 2021-06-13:
@MartyG-RealSense thanks for your reply. Unfortunately , `init_reset` has no effect in my case.

AndreV84, 2021-06-13:
@mzahana what exactly is the command line you are executing the docker image with?
what are the exact steps to reproduce the errors?
you may try executing with --privileged , also you may mount video devices to the container somehow to see if it works

mzahana, 2021-06-13:
Here is the docker run command
```
docker run --runtime nvidia -it --network host -e DISPLAY=$DISPLAY --restart always \    --user=${CONTAINER_USER_NAME} \    -v /tmp/.X11-unix/:/tmp/.X11-unix \    -v /dev:/dev \    --group-add=dialout \    --group-add=tty \    --tty=true \    --device=/dev/ttyUSB0 \    --device=/dev/ttyTHS0 \    -v ${WORKSPACE_DIR}:/home/${CONTAINER_USER_NAME}/shared_volume \    --workdir="/home/${CONTAINER_USER_NAME}" \    --name=${CONTAINER_NAME} \    --privileged \    ${DOCKER_REPO} \    bash -c "${CMD}"
```

AndreV84, 2021-06-13:
which ROS distribution are you using? ROS1? 
Melodic? Noetic?
OS which version is used? 18.04 ubuntu 20.04? within the container? outside of the container?
You may also try ROS2
Regarding the execution of jetson command you may like to add to /etc/docker/daemon.json the content as follows
```
 to your /etc/docker/daemon.json configuration file before attempting to build the containers:

{
    "runtimes": {
        "nvidia": {
            "path": "nvidia-container-runtime",
            "runtimeArgs": []
        }
    },

    "default-runtime": "nvidia"
}
```
normally on Nx realsense d435i works, but sometimes powered usb hub is handy, also it might conflict with other usb devices, also the latest firmware is better to use to avoid bugs etc.
Generally dockerization works, but it is tricky to implement;
If it doesn't work for you with ROS1 try switching to ROS2

AndreV84, 2021-06-13:
latest firmware is 5.12.14_50 as I recollect
e.g. it could be installed from 2.47 sdk

AndreV84, 2021-06-13:
@mzahana what is the usb speed value returned by 
rs-fw-update -l
is it 2.0 usb speed or 3.2 usb speed?

mzahana, 2021-06-14:
@AndreV84 
* The `/etc/docker/daemon.json` is the same as the one you mentioned
* here is the output of `rs-fw-update -l`
```sh
connected devices:
1) Name: Intel RealSense D435I, serial number: 048122071215, update serial number: 046523051844, firmware version: 05.12.07.100, USB type: 3.2
```
* I have a swap of 4Gb, on an SD card. I don't have an SSD

AndreV84, 2021-06-14:
@mzahana swap 4gb might not be sufficient for dockerized ros node with realsense sdk
also you might need to install systemwide realsense sdk which you probably have done
try assigning at least 8-12 gb of swap - it could fit on sdcard

AndreV84, 2021-06-14:
@mzahana 
is the rs-fw-update -l output from docker or from systemwisde in the message above?

AndreV84, 2021-06-14:
does the ros node work from systemwide install on the same device?

mzahana, 2021-06-14:
@AndreV84 the `rs-fw-update -l` is run system wide.
I have not tried to use ROS system wide on Xavier NX
I already have everything else I need running inside a container, except realsense ros

AndreV84, 2021-06-14:
@mzahana what is the output of rs-fw-update -l from the container?

mzahana, 2021-06-14:
Now, I am trying to install the same librealsense  system wide from source (same version as in the container); it was installed from apt-get. I am not sure if that is an issue.

AndreV84, 2021-06-14:
@mzahana typically we would try both from source but also from apt;for the container use we used from source as it would allow to reduce amount of issues;
Could you share the output of
```
rs-fw-update -l
```
if executing it from inside the container?

AndreV84, 2021-06-14:
have you had a chance to try ros2 container? with realsense node? it could be 18.04 or upgraded to 20.04 container

mzahana, 2021-06-20:
@AndreV84 Sorry for the late reply. The Xavier NX stopped working and I had to re-flash it. This time I am using Jetpack 4.4 instead of the latest. I read somewhere that things worked better with Jetpack 4.4

The output  of `rs-fw-update -l ` inside the container gives
```
connected devices:
1) Name: Intel RealSense D435I, serial number: 048122071215, update serial number: 046523051844, firmware version: 05.12.07.100, USB type: 3.2
```

The output  of `rs-fw-update -l ` outside the container gives
```
connected devices:
1) Name: Intel RealSense D435I, serial number: 048122071215, update serial number: 046523051844, firmware version: 05.12.07.100, USB type: 3.2
```

I don't use ROS 2 and I don' have it.

I installed the same librealsense version (2.45.0) outside/insde the container, and I still get the same error
```
UMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
[... 114 lines cut ...]
 20/06 16:51:23,169 ERROR [546859889024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 
 20/06 16:51:23,203 ERROR [546859889024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 
 20/06 16:51:23,212 WARNING [547264389504] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 16:51:23,237 ERROR [546859889024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 
 20/06 16:51:23,270 ERROR [546859889024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 
 20/06 16:51:23,303 ERROR [546859889024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 
 20/06 16:51:23,337 ERROR [546859889024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 
 20/06 16:51:23,363 WARNING [547264389504] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 16:51:23,396 ERROR [546859889024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 
 20/06 16:51:23,414 WARNING [547264389504] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 16:51:23,456 ERROR [546859889024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 
^C[camera/realsense2_camera-3] killing on exit
[camera/realsense2_camera_manager-2] killing on exit
 20/06 16:51:23,516 ERROR [546859889024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 
 20/06 16:51:23,566 WARNING [547264389504] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 16:51:23,577 ERROR [546859889024] (synthetic-stream.cpp:48) Exception was thrown during user processing callback: 
^C 20/06 16:51:23,617 WARNING [547264389504] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

```

I am not sure what to try next. I still need to run D435i with realsense-ros inside the container, as everything else is working fine.

Any hints?

Thanks.

AndreV84, 2021-06-20:
@mzahana 
Did you add the swap at least somewhat 15GB?
In Dockerfile you could just update lines to install ROS 2 instead of ROS1, then setup corresponding packages

AndreV84, 2021-06-20:
does realsense viewer run fine from withtin the docker container?

mzahana, 2021-06-20:
@AndreV84 I am using a 16GB SD card for boot and 250GB SSD.
Here is the output of `free`
```
              total        used        free      shared  buff/cache   available
Mem:        7949852     1521752     3949068       39924     2479032     6213464
Swap:       3974904           0     3974904

```
I guess the swap is 4GB.
I followed [this nice video from JetsonHacks](https://www.youtube.com/watch?v=ZK5FYhoJqIg) to use the SSD
Now, I am not sure if the swap is on the SD card or the SSD. How to increase it in this case?

Now, I am ssh-ing into the Xavier, so realsense-viewer won't likely work remotely. I can try this tomorrow, though.

AndreV84, 2021-06-20:
@mzahana it seems that you got the OS onto the ssd drive
you may do these steps
```
Turn off zram:

cd /etc/systemd
sudo mv nvzramconfig.sh nvzramconfig.sh.orig
sudo reboot
​ Add swap file on nvme and verify:

sudo fallocate -l 32G /home/nvidia/nvme/swapfile
sudo chmod 600 /home/nvidia/nvme/swapfile
sudo mkswap /home/nvidia/nvme/swapfile
sudo swapon /home/nvidia/nvme/swapfile
sudo swapon -s
Add the line below to /etc/fstab so swap file will be automounted going forward:

/home/nvidia/nvme/swapfile swap swap defaults 0 0

```
for remote realsense-viewer you may install turboVNC + VirtualGL, or else. even if jetson just has some display atatched it wopuld be possible to use realsense-viewer redirecting it to the display while over ssh, but it would require to add nomachine or vino to see also the output. While withotu conencted HDMI display - VirtuaoGL can be used

mzahana, 2021-06-20:
@AndreV84 There is no `nvme` inside my home folder. Do I just create it?

AndreV84, 2021-06-20:
@mzahana you may either just create folder following the steps exactly
or adjust paths as per your convinience
the only limitation is that there should be enough space as for now

mzahana, 2021-06-20:
@AndreV84 I just increased the swap size to 32GB, and I still get the same error :(

mzahana, 2021-06-20:
@AndreV84 Do I need to do any kernel patching?

AndreV84, 2021-06-20:
@mzahana you may try ros2 realsnse foxy wrapper within a container
https://github.com/IntelRealSense/realsense-ros/releases/tag/3.2.0

mzahana, 2021-06-20:
@AndreV84 The point is I am not going to use ROS 2, even if it worked there as everything else is working fine with ROS 1 in the container.
I am going to try to use a powered USB hub to see if it's a power issue.

AndreV84, 2021-06-20:
we did the kernel patching but it was only required for d455 on NX due to specific software compatibility
typically RSUSB baqckend is used that doesn't require patching;
Does systemwide ROS1 rtealsense node  work if running fropm outside of the container jusy for brief test?

mzahana, 2021-06-20:
@AndreV84 I will need to install ROS 1 system wide and test.

mzahana, 2021-06-20:
@AndreV84 So, I installed ROS1 system wide and realsense-ros pkg. I get the following Warning/ERROR in the beginning, then it stops appearing. Otherwise I am now able to get 30Hz of both color and depth topics
```
WARNING [546786234752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 21:12:30,828 WARNING [546786234752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 21:12:30,980 WARNING [546786234752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 21:12:31,232 WARNING [546786234752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 21:12:31,283 WARNING [546786234752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 21:12:31,434 WARNING [546786234752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 21:12:31,485 WARNING [546786234752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 21:12:31,636 WARNING [546786234752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 21:12:31,687 WARNING [546786234752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 21:12:31,838 WARNING [546786234752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 21:12:31,889 WARNING [546786234752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 21:12:31,920 ERROR [546391974272] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 20/06 21:12:32,040 WARNING [546786234752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 21:12:32,091 WARNING [546786234752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 21:12:32,142 WARNING [546786234752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 21:12:32,194 WARNING [546786234752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 21:12:32,244 WARNING [546786234752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 21:12:32,295 WARNING [546786234752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 20/06 21:12:32,346 WARNING [546786234752] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

```
What could be missing inside the container then, if it works fine outside it?

AndreV84, 2021-06-20:
@mzahana 
is the rgb topic still empty in the docker container?
you could also try to install in the container realsense sdk  from apt repository

mzahana, 2021-06-20:
@AndreV84 I installed both realsense SDK and realsense-ros from apt repos, and I was able to get color/depth topics inside the container. However, I am not able to get pointcloud inside the container, but I get it outside the container.
```
roslaunch realsense2_camera rs_camera.launch filters:="pointcloud"
```
I get,
```
[ INFO] [1624217370.319602053]: publishing unordered pointcloud.
[ WARN] [1624217370.792778998]: No stream match for pointcloud chosen texture Process - Color
```

That was the main reason that I switched to realsense sDK and ros pkg from source instead from apt repos.
If I can get pointcloud inside the container, then problem solved.

AndreV84, 2021-06-20:
@mzahana 
1. try different usb ports of the NX ; some ports will do better than others [  it is the case with zed camera]
2. try using the source code realsense sdk. could you confirm it also got the rgb topics published inside the container?
3. try with/ without powered/ unpowered udb hubs with various cables

AndreV84, 2021-06-20:
it is also a warning
it does actually do pointcloud but some frames are dropped
https://github.com/IntelRealSense/realsense-ros/pull/591

mzahana, 2021-06-21:
@AndreV84 I switched back to realsense `2.45.0` from source and I made one more step that I missed, copying the udev rules (inside the container), but still I get the same error,and only depth topic has messages. The color topic is published by empty.

I also tried to run `realsense-viewer`. Outside the container (with realsense sdk installed from source) it works fine. However, inside the container it doesn't and it gives the following error
```
Segmentation fault (core dumped)
```

Just a side note, I build realsense sdk using the following command
```
cmake ../  -DFORCE_RSUSB_BACKEND=ON -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true -DBUILD_WITH_CUDA=true
```

mzahana, 2021-06-21:
@AndreV84 
The only solution that worked for me so far, is to run the realsense ros node outside the container and access the topics inside. That worked with all topics at 30 Hz including pointcloud.

AndreV84, 2021-06-21:
@mzahana so it worked from apt repos inside the container, but did not work from sources inside the container? right?

mzahana, 2021-06-21:
@AndreV84 yes, worked from apt repos inside the container, but partially. Pointcloud doesn’t work, and color image is published at reduced rate of 16 hz. Outside the container, compiled from source, works fine with all topics at 30hz including pointcloud

AndreV84, 2021-06-21:
@mzahana Probably  pointcloud should work for apt repos sdk implementation; the messages you quoted are warnings about skipped frames; you still should be able t use the pointcloud.

AndreV84, 2021-06-22:
try running ralsense from inside of the container from apt repository realsense manager, or try rviz to see pointcloud; it should appear

mzahana, 2021-06-22:
I think I reached to a working state inside the container. what I observed is that if I build a docker image with root user, it works. Previously, I created a user inside the container and for some reason things didn't work well with realsense. Now, I am using the root user inside the container, have reaalsense sdk 2.45.0 and realsense-ros 2.3.0 compiled from source inside the container, and I get all topics I need. There are still some warning that appear in the  start only
```
 22/06 19:23:44,770 WARNING [546652017024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/06 19:23:44,821 WARNING [546652017024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/06 19:23:44,872 WARNING [546652017024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/06 19:23:44,922 WARNING [546652017024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/06 19:23:45,064 WARNING [546652017024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/06 19:23:45,114 WARNING [546652017024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/06 19:23:45,165 WARNING [546652017024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/06 19:23:45,216 WARNING [546652017024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 22/06 19:23:45,268 WARNING [546652017024] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
```
Otherwise, it's in a working state.
I am also able to run rviz and realsense-viewer inside the container.

@AndreV84 Thanks for your assistance.

mzahana, 2021-06-22:
I also would like to mention that I am using containers based on [jetson-containers](https://github.com/dusty-nv/jetson-containers). Very nice repo!

MartyG-RealSense, 2021-06-22:
Hi @mzahana  If **control_transfer returned** warnings are generated a small number of times during launch, or at a rate of one or two per minute after launch, then they can be safely ignored.

AndreV84, 2021-07-20:
@MartyG-RealSense 
do you know about any use cases
of d435i working with esp32?

[4 more replies]