realsenseai/realsense-ros #1605: Nothing shows up in rviz d435i

majinshaoyuindustry, 2021-01-04:
Hello,
I am running on realsense 2.40. and realsense-ros 2.2.20
Installation and all the commands seems to work well, then I tried to init the pointcloud example and here is the return:

c:\catkin_ws>roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
... logging to C:\Users\ssu\.ros\log\014b7102-4ee2-11eb-a67a-48f17fc2b9f5\roslaunch-MSI-10004.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://127.0.0.1:51375/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path:
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/rosbag_filename:
 * /camera/realsense2_camera/serial_no:
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method:
 * /camera/realsense2_camera/usb_port_id:
 * /rosdistro: noetic
 * /rosversion: 1.15.9

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [11856]
ROS_MASTER_URI=http://127.0.0.1:11311
setting /run_id to 014b7102-4ee2-11eb-a67a-48f17fc2b9f5
process[rosout-1]: started with pid [16792]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [888]
process[camera/realsense2_camera-3]: started with pid [13992]
[ERROR] [1609801822.608764500]: Skipped loading plugin with error: XML Document 'c:\opt\ros\noetic\x64\share\test_nodelet/test_nodelet.xml' has no Root Element. This likely means the XML is malformed or missing..
[ERROR] [1609801822.613452300]: Skipped loading plugin with error: XML Document 'c:\opt\ros\noetic\x64\share\test_nodelet_topic_tools/test/test_nodelets.xml' has no Root Element. This likely means the XML is malformed or missing..
[ INFO] [1609801822.624265800]: Initializing nodelet with 12 worker threads.
[ INFO] [1609801823.244338300]: RealSense ROS v2.2.20
[ INFO] [1609801823.244503900]: Built with LibRealSense v2.40.0
[ INFO] [1609801823.245418300]: Running with LibRealSense v2.40.0
[ INFO] [1609801823.280496100]:
[ INFO] [1609801823.688707800]: Device with serial number 034422074961 was found.

[ INFO] [1609801823.688825400]: Device with physical ID \\?\usb#vid_8086&pid_0b3a&mi_03#6&1a077f8f&0&0003#{e5323777-f976-4f5b-9b55-b94699c46e44}\global was found.
[ INFO] [1609801823.689420500]: Device with name Intel RealSense D435I was found.
[ WARN] [1609801823.689668800]: Error extracting usb port from device with physical ID: \\?\usb#vid_8086&pid_0b3a&mi_03#6&1a077f8f&0&0003#{e5323777-f976-4f5b-9b55-b94699c46e44}\global
Please report on github issue at https://github.com/IntelRealSense/realsense-ros
[ INFO] [1609801823.689927900]: Device USB type: 3.2
[ INFO] [1609801823.691540700]: getParameters...
[ INFO] [1609801823.789961600]: setupDevice...
[ INFO] [1609801823.790054800]: JSON file is not provided
[ INFO] [1609801823.790760900]: ROS Node Namespace: camera
[ INFO] [1609801823.791186800]: Device Name: Intel RealSense D435I
[ INFO] [1609801823.791388100]: Device Serial No: 034422074961
[ INFO] [1609801823.791434800]: Device physical port: \\?\usb#vid_8086&pid_0b3a&mi_03#6&1a077f8f&0&0003#{e5323777-f976-4f5b-9b55-b94699c46e44}\global
[ INFO] [1609801823.791727400]: Device FW version: 05.12.10.00
[ INFO] [1609801823.791992200]: Device Product ID: 0x0B3A
[ INFO] [1609801823.792244200]: Enable PointCloud: On
[ INFO] [1609801823.792462900]: Align Depth: Off
[ INFO] [1609801823.792650300]: Sync Mode: On
[ INFO] [1609801823.792940100]: Device Sensors:
[ INFO] [1609801823.836903600]: Stereo Module was found.
[ INFO] [1609801823.870625700]: RGB Camera was found.
[ INFO] [1609801823.870929800]: Motion Module was found.
[ INFO] [1609801823.871485800]: Add Filter: pointcloud
[ INFO] [1609801823.872124000]: num_filters: 1
[ INFO] [1609801823.872263300]: Setting Dynamic reconfig parameters.
[ INFO] [1609801826.576498100]: Done Setting Dynamic reconfig parameters.
[ INFO] [1609801826.576899900]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1609801826.577569600]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1609801826.577667700]: setupPublishers...
[ INFO] [1609801826.582148700]: Expected frequency for depth = 30.00000
[ INFO] [1609801826.607574900]: Expected frequency for color = 30.00000
[ INFO] [1609801826.623599100]: setupStreams...
[ INFO] [1609801826.661406600]: insert Depth to Stereo Module
[ INFO] [1609801826.661540600]: insert Color to RGB Camera
[ INFO] [1609801826.857030100]: SELECTED BASE:Depth, 0
[ INFO] [1609801826.869757800]: RealSense Node Is Up!

Then I fired up rviz but nothing showed up. The camera works fine with the viewer though.

[image]

Please help!
Thanks!

majinshaoyuindustry, 2021-01-04:
UPDATE: Here is my rviz return:

c:\catkin_ws>rviz
[ INFO] [1609804129.910808500]: rviz version 1.14.4
[ INFO] [1609804129.910933300]: compiled against Qt version 5.15.0
[ INFO] [1609804129.911107500]: compiled against OGRE version 1.12.7 (Rhagorthua)
[ INFO] [1609804129.920127600]: Forcing OpenGl version 0.
[ INFO] [1609804132.657486900]: Stereo is NOT SUPPORTED
[ INFO] [1609804132.657585100]: OpenGl version: 4.6 (GLSL 4.6).

Is it possible that my rviz got a problem from the "Stereo is NOT SUPPORTED" ?
I didn't even install rviz, I guess it is part of ROS?

jpizarrom, 2021-01-05:
Hello, at least for ros2 foxy I used static_transform_publisher nodes and real. policy "Best Effort" for images in rviz2  

[image]

```
import os

from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
import launch_ros.actions

def generate_launch_description():
    return LaunchDescription([

        launch_ros.actions.Node(
            package='tf2_ros', node_executable='static_transform_publisher', node_name='bl_base', output='screen',
            arguments=['0', '0', '0', '0', '0', '0', 'map', 'base_link']),
    
        launch_ros.actions.Node(
            package='tf2_ros', node_executable='static_transform_publisher', node_name='bl_camera_color', output='screen',
            arguments=['0', '0', '0', '0', '0', '0', 'base_link', 'camera_color_optical_frame']),
        
        launch_ros.actions.Node(
            package='tf2_ros', node_executable='static_transform_publisher', node_name='bl_camera_depth', output='screen',
            arguments=['0', '0', '0', '0', '0', '0', 'base_link', 'camera_depth_optical_frame']),

        # Rviz
        launch_ros.actions.Node(
            package='rviz2', node_executable='rviz2', output='screen',
            arguments=['--display-config', './default.rviz']
            ),
    ])
```

majinshaoyuindustry, 2021-01-05:
> Hello, at least for ros2 foxy I used static_transform_publisher nodes and real. policy "Best Effort" for images in rviz2
> 
> [image]
> 
> custom_launch.py
> 
> ```python
> import os
> 
> from ament_index_python.packages import get_package_share_directory
> from launch import LaunchDescription
> import launch_ros.actions
> 
> def generate_launch_description():
>     return LaunchDescription([
> 
>         launch_ros.actions.Node(
>             package='tf2_ros', node_executable='static_transform_publisher', node_name='bl_base', output='screen',
>             arguments=['0', '0', '0', '0', '0', '0', 'map', 'base_link']),
>     
>         launch_ros.actions.Node(
>             package='tf2_ros', node_executable='static_transform_publisher', node_name='bl_camera_color', output='screen',
>             arguments=['0', '0', '0', '0', '0', '0', 'base_link', 'camera_color_optical_frame']),
>         
>         launch_ros.actions.Node(
>             package='tf2_ros', node_executable='static_transform_publisher', node_name='bl_camera_depth', output='screen',
>             arguments=['0', '0', '0', '0', '0', '0', 'base_link', 'camera_depth_optical_frame']),
> 
>         # Rviz
>         launch_ros.actions.Node(
>             package='rviz2', node_executable='rviz2', output='screen',
>             arguments=['--display-config', './default.rviz']
>             ),
>     ])
> ```

Hi thanks for showing me a working model. 
Pardon me that I am very new to ROS. I am running on: ROS noetic rosversion: 1.15.9. Maybe that is ROS1 compared to yours (meanwhile my rviz would be considered as rviz1 comparing to yours as rviz2)? I also found out that I have that warning: "Error extracting usb port from device with physical ID", not sure if this was the problem too.

MartyG-RealSense, 2021-01-05:
Thanks very much @jpizarrom for offering advice to @majinshaoyuindustry 

Hi @majinshaoyuindustry  It looks as though you are using the point cloud example from the RealSense ROS wrapper documentation.

https://github.com/IntelRealSense/realsense-ros#point-cloud

Could you try the procedure below please:

1.  Left-click on the **Add** button at the bottom of the RViz side-panel.

[image]

2.  On the panel that appears in the center window of Rviz after clicking Add, select the **By Topic** tab.

3.  You should now see a list of topics.  Select the topic **depth/color/points/PointCloud2**

If all has gone well, the images should now be displayed in RViz.

majinshaoyuindustry, 2021-01-05:
@MartyG-RealSense  
Hello Marty,

Thank you for showing me that, it works well now! Weird that if I selected pointcloud2 from the "by display" it won't work.
I do have one more question: 
I was trying to do transformation over on the camera and potentially aligning two 435i together. I went through all the related topic and the suggestions made from you (Vicalib, Connix-Center, and this tutorial from Dorodnic [https://github.com/IntelRealSense/realsense-ros/wiki/Showcase-of-using-2-cameras](url) ).

It seems that there isn't any pointers/examples that is Windows friendly. Is that true?
ie. the "set_cams_transforms.py" requires Termios which is Unix only. Is there a walk-around for WIN10?
I guess from the multiple IMUs' data we can construct a matrix for affine transformation?

Much appreciated!

MartyG-RealSense, 2021-01-05:
librealsense has an affine transformation instruction called **rs2_transform_point_to_point**, so that may be a good option for doing a live affine transform that can be accessed from a librealsense Windows application.

https://github.com/IntelRealSense/librealsense/issues/5583#issuecomment-570805578

doronhi, 2021-01-07:
The 'set_cams_transforms.py' is just used to manually play with the transformation between frame ids `cam_1_link` and `cam_2_link`. I wrote it a while back when ROS was not available for windows.
You can set that transformation using the ros command `static_transform_publisher` - see [static_transform_publisher wiki](http://wiki.ros.org/tf#static_transform_publisher).

majinshaoyuindustry, 2021-01-07:
@MartyG-RealSense @doronhi 

Thank you both very much for the pointers!
I had spent quite some time with the transformation functions you shared, bear with my lack of knowledge but I couldn't find even just a simple  example for PC taking advantages of the IMUs from multiple d435i, or these transformation methods are just for manual/estimations?
Thanks!

MartyG-RealSense, 2021-01-08:
Hi @majinshaoyuindustry   If you were working in librealsense, the **multi streaming model** of accessing streams ('endpoints') from different cameras would be an appropriate method to use.

If ROS is your preferred method then @doronhi will be able to advise you better.  Doronhi suggested using **static_transform_publisher** though,  and there is a non-RealSense case where someone was advised to combine multiple Kinect depth cameras in RViz using static_transform_publisher by setting up a new 'multi_map' frame in the launch file and then setting it in RViz in the **Fixed Frame** option.

https://answers.ros.org/question/261331/rviz-multiple-cameras-kinects-same-global-frame/

doronhi, 2021-01-12:
The kinect example shared by @MartyG-RealSense is very simple and works the same way as the example given here. It means that you should set a transformation from every frame you want seen to a single transformation which is selected as the "Fixed frame" in RViz.
In both examples, the situation is static - The relative position of the cameras is known and is set to a static transformation. From you trying to use the IMU I gather that the cameras relative position isn't constant. If that is the case bear in mind that the IMU just measures temporal acceleration and rotational velocity, and perhaps to that accurately for long periods of time. You'll need other tools, like "rtabmap" or others, to estimate the camera's position and orientation and then keep updating the static transform accordingly.

MartyG-RealSense, 2021-01-18:
Hi @majinshaoyuindustry   Do you require further assistance with this case, please?   Thanks!

majinshaoyuindustry, 2021-01-18:
Thank you for you help @MartyG-RealSense I'm good for now!

closed 2021-01-19 by MartyG-RealSense