realsenseai/realsense-ros #1607: L515 IMU Timestamp Disorder

xiaotaw, 2021-01-05:
**Info**
librealsense: may be 2.41.0 or an older version, because I apt-get upgrade the system yesterday.
```bash
$ locate librealsense2.so
/opt/ros/melodic/lib/librealsense2.so
/opt/ros/melodic/lib/librealsense2.so.2.40
/opt/ros/melodic/lib/librealsense2.so.2.40.0
/usr/lib/x86_64-linux-gnu/librealsense2.so
/usr/lib/x86_64-linux-gnu/librealsense2.so.2.41
/usr/lib/x86_64-linux-gnu/librealsense2.so.2.41.0
```
realsense-ros: 2.2.20
```bash
$ head -n 5 /opt/ros/melodic/share/realsense2_camera/package.xml 
<?xml version="1.0"?>
<package format="2">
  <name>realsense2_camera</name>
  <version>2.2.20</version>
  <description>RealSense Camera package allowing access to Intel T265 Tracking module and SR300 and D400 3D cameras</description>
```
firmware: 01.05.02.00

**Problem**
 For every 1 hour, the imu's timestamp suddenly drop to start_timestamp when record begins.

**Reproduce**
1. link camera to computer, and record imu data into a bag for a more than 1 hour.
2. play the bag, and run a c++ program (the code comes from imu_utils), like: 
```c++
void
imu_callback( const sensor_msgs::ImuConstPtr& imu_msg )
{
    //    m_buf.lock( );
    //    imu_buf.push( imu_msg );
    //    m_buf.unlock( );
    double time = imu_msg->header.stamp.toSec( );
    gyr_x->pushRadPerSec( imu_msg->angular_velocity.x, time );
    gyr_y->pushRadPerSec( imu_msg->angular_velocity.y, time );
    gyr_z->pushRadPerSec( imu_msg->angular_velocity.z, time );
    acc_x->pushMPerSec2( imu_msg->linear_acceleration.x, time );
    acc_y->pushMPerSec2( imu_msg->linear_acceleration.y, time );
    acc_z->pushMPerSec2( imu_msg->linear_acceleration.z, time );

    const double offset = 1.60972e+09;
    if ( start )
    {
        start_t = time;
        start   = false;
        std::cout << "\noffset(s): " << offset << std::endl;
    }
    else
    {
        double time_min = ( time - start_t ) / 60;
        if ( time_min > max_time_min )
            end = true;
    }
    static int cnt = 0;
    cnt ++;
    if (cnt % (400*60) == 0){
        std::cout << "start_time(s) (-offset): " << start_t - offset;
        std::cout << "  now(s) (-offset): " << time - offset;
        std::cout << "  duration(min): " << (time - start_t) / 60 << std::endl;
    }
}

ros::Subscriber sub_imu = n.subscribe( IMU_TOPIC, //
                                           20000000,
                                           imu_callback,
                                           ros::TransportHints( ).tcpNoDelay( ) );

[... 286 lines cut ...]
start_time(s) (-offset): 1334.75  now(s) (-offset): 2294.46  duration(min): 15.9952
start_time(s) (-offset): 1334.75  now(s) (-offset): 2371.91  duration(min): 17.2861
start_time(s) (-offset): 1334.75  now(s) (-offset): 2463.96  duration(min): 18.8201
start_time(s) (-offset): 1334.75  now(s) (-offset): 2558.21  duration(min): 20.391
start_time(s) (-offset): 1334.75  now(s) (-offset): 2659.65  duration(min): 22.0817
start_time(s) (-offset): 1334.75  now(s) (-offset): 2747.97  duration(min): 23.5537
start_time(s) (-offset): 1334.75  now(s) (-offset): 2834.82  duration(min): 25.0012
start_time(s) (-offset): 1334.75  now(s) (-offset): 2918.78  duration(min): 26.4005
start_time(s) (-offset): 1334.75  now(s) (-offset): 3012.15  duration(min): 27.9567
start_time(s) (-offset): 1334.75  now(s) (-offset): 3085.47  duration(min): 29.1788
start_time(s) (-offset): 1334.75  now(s) (-offset): 3172.69  duration(min): 30.6323
start_time(s) (-offset): 1334.75  now(s) (-offset): 3286.24  duration(min): 32.5249
start_time(s) (-offset): 1334.75  now(s) (-offset): 3418.22  duration(min): 34.7245
start_time(s) (-offset): 1334.75  now(s) (-offset): 3533.29  duration(min): 36.6423
start_time(s) (-offset): 1334.75  now(s) (-offset): 3638.72  duration(min): 38.3995
start_time(s) (-offset): 1334.75  now(s) (-offset): 3740.41  duration(min): 40.0944
start_time(s) (-offset): 1334.75  now(s) (-offset): 3836.63  duration(min): 41.698
start_time(s) (-offset): 1334.75  now(s) (-offset): 3951.82  duration(min): 43.6178
start_time(s) (-offset): 1334.75  now(s) (-offset): 4101.06  duration(min): 46.1052
start_time(s) (-offset): 1334.75  now(s) (-offset): 4234.03  duration(min): 48.3214
start_time(s) (-offset): 1334.75  now(s) (-offset): 4364.31  duration(min): 50.4926
start_time(s) (-offset): 1334.75  now(s) (-offset): 4499.71  duration(min): 52.7494
start_time(s) (-offset): 1334.75  now(s) (-offset): 4615.07  duration(min): 54.672
start_time(s) (-offset): 1334.75  now(s) (-offset): 4715.93  duration(min): 56.3531
```

xiaotaw, 2021-01-05:
launch file:
```vim
<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>

  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="false"/>

  <arg name="infra_width"         default="640"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="false"/>
  <arg name="enable_infra2"       default="false"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="1280"/>
  <arg name="color_height"        default="720"/>
  <arg name="enable_color"        default="false"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="400"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="true"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default="linear_interpolation"/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>
  <arg name="allow_no_texture_points"   default="false"/>

  <group ns="$(arg camera)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
      <arg name="external_manager"         value="$(arg external_manager)"/>
      <arg name="manager"                  value="$(arg manager)"/>
      <arg name="output"                   value="$(arg output)"/>
      <arg name="serial_no"                value="$(arg serial_no)"/>
      <arg name="usb_port_id"              value="$(arg usb_port_id)"/>
      <arg name="device_type"              value="$(arg device_type)"/>
[... 28 lines cut ...]
      <arg name="depth_fps"                value="$(arg depth_fps)"/>
      <arg name="infra_fps"                value="$(arg infra_fps)"/>
      <arg name="color_fps"                value="$(arg color_fps)"/>
      <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
      <arg name="accel_fps"                value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="publish_tf"               value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
    </include>
  </group>
</launch>

```

doronhi, 2021-01-12:
I'll look into it. I opened an inner tracking file: RS5-10187

doronhi, 2021-01-14:
The issue is caused because global_timestamp is off (in L515 it is currently off by default. It is On by default in D4xx series). To turn it on, add the following lines to your launch file:
```
  <rosparam>
      /camera/motion_module/global_time_enabled: true
      /camera/l500_depth_sensor/global_time_enabled: true
      /camera/rgb_camera/global_time_enabled: true
  </rosparam>
```

xiaotaw, 2021-01-18:
Thank you, @doronhi ! I will try it and give a feedback as soon as possible.

closed 2021-01-18 by xiaotaw

MartyG-RealSense, 2025-02-07:
Hi @Muyun99  You could try enabling global time in the ROS2 launch instruction:

`ros2 launch realsense2_camera rs_launch.py global_time_enabled:=true`

You could also try enabling it after launch has completed with a **ros2 param set** instruction.  If you have a RealSense 400 Series camera then in the current ROS2 wrapper the command may be something like this:

`ros2 param set /camera/camera depth_module.global_time_enabled true`

If you have an L515 camera and are using an old version of the ROS2 wrapper such as 2.50.0 then the command may be like this:

r`os2 param set /camera depth_module.global_time_enabled true`

Muyun99, 2025-02-07:
> Hi [@Muyun99](https://github.com/Muyun99) You could try enabling global time in the ROS2 launch instruction:
>
> `ros2 launch realsense2_camera rs_launch.py global_time_enabled:=true`
>
> You could also [...]

@MartyG-RealSense Thx for your reply, i fix this issue using the following commad

`ros2 launch realsense2_camera rs_launch.py camera_namespace:='camera' camera_name:='l515' align_depth.enable:=true motion_module.global_time_enabled:=true depth_module.global_time_enabled:=true rgb_camera.global_time_enabled:=true initial_reset:=true`

MartyG-RealSense, 2025-02-07:
@Muyun99  You are very welcome.  Thanks so much for sharing what worked for you!

MartyG-RealSense, 2026-01-09:
What I meant by suggesting it is that global time creates a common timestamp for the different streams, and the enable_sync setting does its best to send frames with the same timestamp.  So enable_sync might be a substitute for having global time enabled, even though the warning message about hardware timestamps will still be present in the log.

Because ROS has its own in-built time synching mechanism, having global time enabled is not vital for ROS users because ROS' own mechanism makes global time redundant.