realsenseai/realsense-ros #1609: Obtaining orientation of multiple D435i cameras

MrOCW, 2021-01-06:
Hi @doronhi, I am using the opensource_tracking.launch file to get the orientation of the camera. Is there a way to get the orientation of the camera with respect to a fixed frame instead of the starting frame when the launch file begins?

Also, is there a way to obtain orientations of 2 cameras?

I have changed the launch file to resemble the format of rs_multiple_devices.launch and 2 cameras has appeared in rviz but I am unable to obtain the orientations from the odom topic.

Thank you

doronhi, 2021-01-06:
This file is basically a demo for combining realsense2_camera node with other open source node navigation tools.
The orientation of the camera with respect to the starting frame is given by `rtabmap_ros` I think. The easiest way to get it relative to a given frame is to publish a static transform between the first frame and the given frame and do the conversion.

Regarding 2 cameras, you didn't specify if there is a known relation between them or not. If not, then it's simply duplicating everything and avoiding duplication of names and mixed topics.
If you also want `rtabmap_ros` to take a known relative position between the cameras into account... well, I am sorry. I would guess it's possible but I don't have much experience with it and as much as I would like to - it doesn't seems likely for me to look into it in the near future.

Good luck

MartyG-RealSense, 2021-01-06:
Hi @doronhi  There is an article where someone used the D435i SLAM tutorial as a foundation and added to it, showing how to then get a relative position fro a recorded map using RTABMAP.

https://github.com/IntelRealSense/librealsense/issues/8094#issuecomment-753979162

closed 2021-01-09 by MrOCW