realsenseai/realsense-ros #1611: Error when installing dependencies on ROS2 Eloquent

rafaelRinconFo, 2021-01-06:
Hello there.

I'm currently trying to install the package for ROS2 Eloquent in UBUNTU 18.04 with the same instructions you've stated. However, I get the following logs:

predator-vision@predator:~$ cd ~/ros2_ws
predator-vision@predator:~/ros2_ws$ sudo apt-get install python-rosdep -y
[sudo] password for predator-vision: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python-rosdep is already the newest version (0.20.0-1).
The following package was automatically installed and is no longer required:
  linux-hwe-5.4-headers-5.4.0-42
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
predator-vision@predator:~/ros2_ws$ sudo rosdep init
ERROR: default sources list file already exists:
	/etc/ros/rosdep/sources.list.d/20-default.list
Please delete if you wish to re-initialize
predator-vision@predator:~/ros2_ws$ sudo rosdep init
Wrote /etc/ros/rosdep/sources.list.d/20-default.list
Recommended: please run

	rosdep update

predator-vision@predator:~/ros2_ws$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Add distro "dashing"
Skip end-of-life distro "eloquent"
Add distro "foxy"
Skip end-of-life distro "groovy"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Add distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in /home/predator-vision/.ros/rosdep/sources.cache
predator-vision@predator:~/ros2_ws$ echo $ROS_DISTRO
eloquent
predator-vision@predator:~/ros2_ws$ rosdep install -i --from-path src --rosdistro $ROS_DISTRO -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
realsense2_camera: Cannot locate rosdep definition for [diagnostic_updater]
realsense2_description: Cannot locate rosdep definition for [realsense_camera_msgs]

As you can see, everything seems to be fine until the rosdep install -i --from-path src --rosdistro $ROS_DISTRO -y command. From this point and on, I can't continue with the installation.

doronhi, 2021-01-07:
The requirement for realsense_camera_msgs is a bug. It should be `realsense2_camera_msgs`. But I don't think it will stop you from building later on.
The main issue, I think, is that "eloquent" is EOL.
You should add the option `--include-eol-distros` to `rosdep update`.

rafaelRinconFo, 2021-01-07:
Hello there!!!

After that command line was added, it worked perfectly. Maybe it should be added in the documentation. 

Thank you very much.

RealSenseSupport, 2021-01-08:
Hello @rafaelRinconFo ,
It sounds like your issue has been resolved, if it is and you don't have any further questions, we'll close this in five business days. If you still need assistance please let us know or if you have any new issues feel free to open a new ticket.

doronhi, 2021-01-12:
> Hello there!!!
> 
> After that command line was added, it worked perfectly. Maybe it should be added in the documentation.
> 
> Thank you very much.

Glad it worked for you.
I updated the README.md file accordingly (#1613)

rafaelRinconFo, 2021-01-12:
> > Hello there!!!
> > After that command line was added, it worked perfectly. Maybe it should be added in the documentation.
> > Thank you very much.
> 
> Glad it worked for you.
> I updated the README.md file accordingly (#1613)

This can be closed now. Your help was very useful.

closed 2021-01-12 by rafaelRinconFo

gaultierlecaillon, 2021-03-02:
> The requirement for realsense_camera_msgs is a bug. It should be `realsense2_camera_msgs`. But I don't think it will stop you from building later on. As a matter of fact, you can remove it from the realsense2_description/package.xml file.
> The main issue, I think, is that "eloquent" is EOL.
> You should add the option `--include-eol-distros` to `rosdep update`.
> After that, `rosdep install -i --from-path src --rosdistro $ROS_DISTRO -y` should work.

Thank you so much. I followed the ros2 eloquent installation (https://docs.ros.org/en/eloquent/Installation/Linux-Install-Binary.html) for ubuntu 18.04 and I had the same issue.

So, I just tape
`rosdep update --include-eol-distros`
Then
`rosdep install -i --from-path ros2-linux/share --rosdistro eloquent -y`
and it works great

Thanks you !