realsenseai/realsense-ros #1615: Fail to make infrared camera stream in RGB for D455

wwtinwhu, 2021-01-07:
Hello,  my launch code is as follows.
`roslaunch realsense2_camera rs_camera.launch enable_infra:=true infra_rgb:=true`
There are two problems. First,  infra image is not in RGB,  still in infrad mode. Second, there is  one infrad image output, the topic is /camera/infra/image_raw. What i  want is /camera/infra1/image_rect_raw and /camera/infra2/image_rect_raw.

MartyG-RealSense, 2021-01-07:
Hi @wwtinwhu   The RealSense ROS wrapper documentation states for **infra_rgb** that "This feature is only supported by Realsense sensors with RGB streams available from the infra cameras".

On the D455 model, it must have firmware version **5.12.8.200** or newer installed in order for the RGB8 color mode to be available from the left infrared imager. 

https://github.com/IntelRealSense/librealsense/issues/7897#issuecomment-736370516

Do you have firmware **5.12.8.200** or newer installed in your camera please?

wwtinwhu, 2021-01-08:
> Hi @wwtinwhu The RealSense ROS wrapper documentation states for **infra_rgb** that "This feature is only supported by Realsense sensors with RGB streams available from the infra cameras".
> 
> On the D455 model, it must have firmware version **5.12.8.200** or newer installed in order for the RGB8 color mode to be available from the left infrared imager.
> 
> [IntelRealSense/librealsense#7897 (comment)](https://github.com/IntelRealSense/librealsense/issues/7897#issuecomment-736370516)
> 
> Do you have firmware **5.12.8.200** or newer installed in your camera please?

Hi @MartyG-RealSense , I check the firmware version of my D455 in the Intel RealSense Viewer. The version is 05.12.10.00. So the answer is yes.

MartyG-RealSense, 2021-01-08:
Thanks very much for the information.

Reading the documentation on the front page of the RealSense ROS wrapper, it seems that **/camera/infra/image_raw** is a topic that is used on the L515 camera model.  So otherwise, I believe that the separate **infra1** and **infra2** topics would be used for a model such as D455.

If the color is coming from the left infrared sensor then I would expect **infra1** to be the appropriate topic to use.

/camera/infra1/camera_info
/camera/infra1/image_rect_raw

What image do you get if you visualize **/camera/infra1/image_rect_raw** after having enabled infra_rgb please?

wwtinwhu, 2021-01-08:
> Thanks very much for the information.
>
> Reading the documentation on the front page of the RealSense ROS wrapper, it seems that **/camera/infra/image_raw** is a topic that is used on the L515 camera [...]

Yes, you are right. Separate infra1 and infra2 topics can be used for a model for D455.  When I tried code as follows, I can get the topic /camera/infra1/image_rect_raw and /camera/infra2/image_rect_raw.
`roslaunch realsense2_camera rs_camera.launch enable_infra1:=true` 
And visualization is like this. 
[image]
By the way, Is this two images in infrared mode? I think it is more like in gray mode.
However, when I tried code as follows, these two topics are not shown.
`roslaunch realsense2_camera rs_camera.launch enable_infra1:=true enable_infra2:=true infra_rgb:=true`

MartyG-RealSense, 2021-01-08:
The way I read the instructions is that your original roslaunch instruction is correct, using **enable_infra** instead of individually enabling infra1 and infra2:

**roslaunch realsense2_camera rs_camera.launch enable_infra:=true infra_rgb:=true**

But try visualizing **/camera/infra1/image_rect_raw** in RViz after using that roslaunch instruction.

wwtinwhu, 2021-01-09:
> The way I read the instructions is that your original roslaunch instruction is correct, using **enable_infra** instead of individually enabling infra1 and infra2:
>
> **roslaunch realsense2_camera [...]

That's the problem. I was mentioned in the first floor. When I tried code as follows, as the instructions.
`roslaunch realsense2_camera rs_camera.launch enable_infra:=true infra_rgb:=true`
There is no topic named /camera/infra1/image_rect_raw or /camera/infra2/image_rect_raw. The topic shown related to it is /camera/infra/image_raw. I suppose this is a bug for the D455. 
But for the current use, I can use the code as follows to get the stereo image in infrared mode.
`roslaunch realsense2_camera rs_camera.launch enable_infra1:=true enable_infra2:=true`.

------------------------------------------------------------------------------------------------------------------------
For the another question you haven't answered me @MartyG-RealSense . Why this infrared image look more like in the gray mode?

MartyG-RealSense, 2021-01-09:
It is normal for infrared to be greyscale in the default **Y8** infrared format.  On devices where color from the left infrared imager is supported (D415 and D455), color modes such as **RGB8** can be used for images with color in them.

Looking through the wrapper release notes, it states the following information in the notes for wrapper **2.2.20** from November 2020:

**"get rgb stream from infrared sensor (applies to D415) - use infra_rgb parameter.**

It only mentions support in this feature for D415 and not D455.

wwtinwhu, 2021-01-10:
> It is normal for infrared to be greyscale in the default **Y8** infrared format. On devices where color from the left infrared imager is supported (D415 and D455), color modes such as **RGB8** can be used for images with color in them.
> 
> Looking through the wrapper release notes, it states the following information in the notes for RealSense ROS wrapper **2.2.20** from November 2020:
> 
> **get rgb stream from infrared sensor (applies to D415) - use infra_rgb parameter.**
> 
> It only mentions support of the infra_rgb parameter for D415 and not D455.

I get the two points  you mentioned. Thank you very much for your patient answer.  Looking forward to the new version that supports getting rgb stream from infrared sensor for D455.  Would you mind informing me when it's done ?  then, I will close this issue.

MartyG-RealSense, 2021-01-10:
I will highlight this case to @doronhi the RealSense ROS wrapper developer.

Hi @doronhi  Would it be possible for the **infra_rgb** parameter to support D455 in a future wrapper update now that D455 has support for color from the left infrared imager with firmware **5.12.8.200** and newer?

wwtinwhu, 2021-01-10:
> I will highlight this case to @doronhi the RealSense ROS wrapper developer.
> 
> Hi @doronhi Would it be possible for the **infra_rgb** parameter to support D455 in a future wrapper update now that D455 has support for color from the left infrared imager with firmware **5.12.8.200** and newer?

Thanks! @MartyG-RealSense @doronhi

doronhi, 2021-01-11:
It seems there is a confusion here.
First, there is no difference in the wrapper, regarding this aspect at least, between D415 and D455. The feature works for both cameras. (Tested with Ubuntu 16, ROS kinetic)
As can be seen when running `rs-enumerate-devices` the infrared sensor with RGB is "Infrared" for example, the line:
`Infrared	  640x480	@ 30Hz	   RGB8`
as opposed to the "Infrared1", "Infrared2" sensors for Y8 and Y16 formats, as seen in the printed line:
 `Infrared 1	  640x480	@ 30Hz	   Y8`

That is why when referring to the rgb format of the infrared sensor we omit the sensor's index and also why you get the "rgb" image in the `/camera/infra/image_raw` topic and the Y8 format in the `/camera/infra1/camera_info` and `/camera/infra2/camera_info` topics.

Regarding the issues you describe in your first post:
1. The topic `/camera/infra/image_raw` comes from the same physical sensor as `/camera/infra1/image_raw`.
2. The image is in RGB format but not as pretty as the image from the RGB sensor. You can also see them in the `realsense-viewer`

wwtinwhu, 2021-01-12:
> It seems there is a confusion here.
> First, there is no difference in the wrapper, regarding this aspect at least, between D415 and D455. The feature works for both cameras. (Tested with Ubuntu 16, [...]

Thanks for your explanation @doronhi . I get that.  But I find another problem that why the color of the light is strange in RGB8 mode? 
[image]

MartyG-RealSense, 2021-01-12:
Hi @wwtinwhu    That coloration of the lights has been encountered by another RealSense user in RGB8 mode on D455:

https://github.com/IntelRealSense/librealsense/issues/7934#issuecomment-741612769

wwtinwhu, 2021-01-13:
> Hi @wwtinwhu That coloration of the lights has been encountered by another RealSense user in RGB8 mode on D455:
> 
> [IntelRealSense/librealsense#7934 (comment)](https://github.com/IntelRealSense/librealsense/issues/7934#issuecomment-741612769)

Thanks.

closed 2021-01-13 by wwtinwhu