realsenseai/realsense-ros #1619: Bond broken

Alex-Beh, 2021-01-13:
Hello, I have  the issue `bond broken, exiting` but the error is inconsistent. Sometimes happen but not often. 

Besides, the following error may occur sometimes:
```
[FATAL] [1610527366.955373954]: ASSERTION FAILED                                                                                                                                             
        file = /opt/ros/melodic/include/ros/publisher.h                                                                                                                                                                   line = 103                                                                                                                                                                                          
        cond = false                                                                                                                                                                                        
        message =                                                                                                                                                                                  
[FATAL] [1610527366.958677887]: Call to publish() on an invalid Publisher                                                                                                                                
[FATAL] [1610527366.959914414]:    
```             

Then the D435 camera won't launch properly and I cannot get the Image data from the rostopic.            

> 
> realsense_ros branch: 2.2.14
> Device FW version: 05.12.06.00

MartyG-RealSense, 2021-01-13:
Hi @Alex-Beh    As a starting point in investigating your case, may I ask which librealsense version you are using please?   Thanks!

Alex-Beh, 2021-01-13:
Hello, 
```
[ INFO] [1610529465.152207594]: RealSense ROS v2.2.14                                                                                                                                                       
[ INFO] [1610529465.154425292]: Running with LibRealSense v2.35.2 
```

MartyG-RealSense, 2021-01-13:
Thanks very much for the information.  Your RealSense ROS wrapper version is correctly matched with the appropriate librealsense SDK version.

The infrequency at which the error occurs would suggest that there is not a serious problem with your RealSense ROS installation and so it may be a temporary 'hiccup' when it does occur.

I located a past case of this error in the RealSense ROS wrapper.  

https://github.com/IntelRealSense/realsense-ros/issues/1068

As it is an infrequent issue for you, I would recommend adding the camera reset instruction below to the end of your roslaunch instruction and see whether the error occurs again in future.

**initial_reset:=true**

doronhi, 2021-01-14:
I recommend upgrading your realsense2_camera. There is a good chance that this bug is already fixed.

Alex-Beh, 2021-01-18:
I try `initial_reset:=true` but sometimes the error still occurs.

> upgrading your realsense2_camera

Do you mean upgrade the ROS wrapper to the latest version 2.2.21?

MartyG-RealSense, 2021-01-18:
Hi @Alex-Beh   Yes, realsense2_camera = the RealSense ROS wrapper

If you plan to upgrade the wrapper version and you use installation **Method 2**, you should first update librealsense.  This will then enable you to know which RealSense ROS wrapper version you should install.

For example, if you plan to install RealSense ROS wrapper 2.2.21 with Method 2 then librealsense version 2.41.0 should first be installed.  The recommended librealsense version for each ROS wrapper version is listed in the release notes for a particular wrapper version.

https://github.com/IntelRealSense/realsense-ros/releases

MartyG-RealSense, 2021-01-24:
Hi @Alex-Beh   Do you require further assistance with this case, please?   Thanks!

Alex-Beh, 2021-01-25:
The error is not consistent. So far it didn't happen but I am not sure where it has been solved completely.

Besides, sometimes the following error will printed in the terminal output:
` 25/01 19:44:31,410 ERROR [281472534372736] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130`

Do you have any idea on this?

MartyG-RealSense, 2021-01-25:
The error **(uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130** usually occurs in logs when **control_transfer returned** warnings are being generated.  If you are only getting a small number of control_transfer returned warnings (if you get any at all) rather than continuously generating ones and ROS is otherwise performing normally, I would be inclined to regard the endpoint errors as momentary communication hiccups that can be safely ignored.

MartyG-RealSense, 2021-02-07:
Case closed due to no further comments received.

closed 2021-02-07 by MartyG-RealSense

JChrysanthemum, 2022-03-10:
Upgrade do not fix the problem for me.  RPI 4B+ Raspbian buster 5.10.63, D455

Built with LibRealSense v2.50.0

Running with LibRealSense v2.50.0

It happens `roslaunch realsense2_camera rs_camera.launch initial_reset:=true filters:=pointcloud`, I can get data at the begining.  
`ERROR [2535638016] (ds5-options.cpp:88) Asic Temperature value is not valid!` apperas after about 1 min,  then I get nothing.

Then, I will get same info in launch any realsense2_camera script. :(

But, a simple reboot will fix it.

MartyG-RealSense, 2022-03-10:
Hi @JChrysanthemum  If the pointcloud performs normally for a minute after launch then that suggests to me that your Pi 4B+ is not struggling to process the data, otherwise it would likely have performance problems from the start of streaming.  The USB system is also likely to be okay.

Does the error occur if you do not use the pointcloud filter?

JChrysanthemum, 2022-03-10:
Hi,@[MartyG-RealSense](https://github.com/MartyG-RealSense) 

Thx for your reply !

> The USB system is also likely to be okay.

Yes, official cable.
`Device USB type: 3.2`

> Does the error occur if you do not use the pointcloud filter?

Nope, It works fine. Only the fps is very low (no more than 10), and 300% cpu was consumed by nodelet (total 4 for a rpi 4)

JChrysanthemum, 2022-03-10:
@[MartyG-RealSense](https://github.com/MartyG-RealSense) 

> Could you check your Pi's memory usage to see whether the available memory is being rapidly consumed during the first minute (a "memory leak") please.

Okay, but I found there is enough mem for system.

Without pointcloud
` 7898.7 total,   6949.4 free` not change much overtime.

With pointcloud
` 7898.7 total,   6949.4 free` not change much overtime.

MartyG-RealSense, 2022-03-10:
Thank you very much for testing.  How does it perform if you set a custom stream configuration with a low stream resolution.

**roslaunch realsense2_camera rs_camera.launch initial_reset:=true filters:=pointcloud depth_width:=640 depth_height:=480 color_width:=640 color_height:=480**

JChrysanthemum, 2022-03-11:
Thanks for your time. 

> roslaunch realsense2_camera rs_camera.launch initial_reset:=true filters:=pointcloud depth_width:=640 depth_height:=480 depth_fps:=30 color_width:=640 color_height:=480 color_fps:=30

eum, really strange. Reboot do no help today. 

Everything works in realsense-viewer on pi.  And your command without `filters:=pointcloud` works.

[image]

with pointcloud, dead imediately.

```
[FATAL] [1646963915.029608247]: ASSERTION FAILED
	file = /opt/ros/melodic/include/ros/publisher.h
	line = 106
	cond = false
	message = 
[FATAL] [1646963915.030051716]: Call to publish() on an invalid Publisher
[FATAL] [1646963915.030195952]: 

[camera/realsense2_camera_manager-2] process has died [pid 2816, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/pi/.ros/log/b8361bd8-a0de-11ec-8f48-dca632b5f9fc/camera-realsense2_camera_manager-2.log].
log file: /home/pi/.ros/log/b8361bd8-a0de-11ec-8f48-dca632b5f9fc/camera-realsense2_camera_manager-2*.log
```

My librealsense installation follows [#7312](https://github.com/IntelRealSense/librealsense/issues/7312#issuecomment-694701288)
`cmake .. -DBUILD_PYTHON_BINDINGS:bool=true -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND=true -DPYTHON_EXECUTABLE=$(which python2)`

Maybe I could try on DepthCloud rather than PointCloud2.

JChrysanthemum, 2022-03-11:
Suprise suprise. Decrease the image size fixed the problem.

[image]

I find another pi user meets a similar problem.
[#1962](https://github.com/IntelRealSense/realsense-ros/issues/1962#issuecomment-872073630)

runable command here `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud depth_width:=424 color_width:=424 depth_height:=240 color_height:=240 depth_fps:=15 color_fps:=15 pointcloud_texture_stream:=RS2_STREAM_COLOR`

So I suggest you may add a simple instruction for pi users.

MartyG-RealSense, 2022-03-11:
It's great news that you found a solution, @JChrysanthemum - thanks very much for the update!

Trying a lower resolution / FPS when instability occurs is a common strategy, so it should not be necessary to provide it as Pi-specific advice.  Thanks for the suggestion.  :)