realsenseai/realsense-ros #1623: Problem running roslaunch

eltonhcchan, 2021-01-14:
I've tried to run roslaunch and got the following errors.

c:\catkin_ws>roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
... logging to C:\Users\Elton\.ros\log\d205d730-5679-11eb-bb36-842afdc7fec3\roslaunch-HP-13408.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://127.0.0.1:49320/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_rgb: False
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path:
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/rosbag_filename:
 * /camera/realsense2_camera/serial_no:
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method:
 * /camera/realsense2_camera/usb_port_id:
 * /rosdistro: noetic
 * /rosversion: 1.15.9

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [11972]
ROS_MASTER_URI=http://127.0.0.1:11311
setting /run_id to d205d730-5679-11eb-bb36-842afdc7fec3
process[rosout-1]: started with pid [16044]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [2928]
process[camera/realsense2_camera-3]: started with pid [1892]
[ INFO] [1610636681.996377300]: Initializing nodelet with 8 worker threads.
**[ERROR] [1610636682.763418900]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Failed to load library C:/catkin_ws/devel\bin/realsense2_camera.dll. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = C:/catkin_ws/devel\bin/realsense2_camera.dll)
[ERROR] [1610636682.763808200]: The error before refreshing the cache was: Failed to load library C:/catkin_ws/devel\bin/realsense2_camera.dll. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = C:/catkin_ws/devel\bin/realsense2_camera.dll)**
[FATAL] [1610636682.767714000]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
←[31m[camera/realsense2_camera-3] process has died [pid 1892, exit code 4294967295, cmd c:\opt\ros\noetic\x64\lib\nodelet\nodelet.exe load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=C:\Users\username\.ros\log\d205d730-5679-11eb-bb36-842afdc7fec3\camera-realsense2_camera-3.log].
log file: C:\Users\username\.ros\log\d205d730-5679-11eb-bb36-842afdc7fec3\camera-realsense2_camera-3*.log←[0m
^C
c:\catkin_ws>
c:\catkin_ws>[camera/realsense2_camera_manager-2] killing on exit
SUCCESS: The process with PID 2928 (child process of PID 13408) has been terminated.
[rosout-1] killing on exit
SUCCESS: The process with PID 16044 (child process of PID 13408) has been terminated.
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Anyone knows how to fix it? Anyone can successfully running ROS wrapper on Windows 10?

I am using realsense2_camera 2.2.21 and librealsense2 2.40 from vcpkg

MartyG-RealSense, 2021-01-14:
Hi @eltonhcchan   I researched your question and located a past case that had the same process-ending fatal error that you experienced when the pointcloud filter was used.

**Failed to load nodelet '/camera/realsense2_cameraof typerealsense2_camera/RealSenseNodeFactoryto managerrealsense2_camera_manager'
←[31m[camera/realsense2_camera-3] process has died**

**Doronhi** the RealSense ROS developer provides advice about this problem in point number '3' of the comment linked to below (skip past points 1 and 2).

eltonhcchan, 2021-01-15:
@MartyG-RealSense  oh yes, I installed librealsense v2.40.0 from vcpkg, but using RealSense ROS wrapper version 2.2.21.  As vcpkg is not yet updated to v2.41.0, are there any other method to install it?

MartyG-RealSense, 2021-01-15:
If you are using Windows then the method mentioned above of downloading and running an installer file from the librealsense SDK's **Releases** page is the easiest and most problem-free way to install.

The SDK can also be installed with **CMake**.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

The majority of that installation guide involves a method for adding metadata support.  This section may be skipped though, as metadata support can be easily added from within the RealSense Viewer program instead.

eltonhcchan, 2021-01-16:
@MartyG-RealSense , I choose to install the SDK using Windows installer. However, when I run the "catkin_make clean", the following errors are encountered.

-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at realsense-ros/realsense2_camera/CMakeLists.txt:41 (find_package):
  Could not find a package configuration file provided by "realsense2" with
  any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

I've tried to add path in the CMakeLists.txt but it cannot find another path again. I wonder if there is file that I can define all the required path. Unfortunately, I cannot locate it. Please advise.

MartyG-RealSense, 2021-01-16:
I had a past ROS installation case with the **Could not find a package configuration file provided by "realsense2" with
any of the following names** error on Windows in November 2020.  The **catkin_make clean** instruction has caused problems for others too.

https://github.com/IntelRealSense/realsense-ros/issues/1520#issuecomment-733179265

In the above link, the solution that the RealSense user found to get it working was to input the command below:

**rosdep install --from-paths src --ignore-src -r -y**

It looks as though that command would be input **before** inputting catkin_make clean

And then after **catkin_make clean** it looks as though the subsequent instructions to input on Windows would be:

**catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
catkin_make install
devel\setup.bat**

eltonhcchan, 2021-01-16:
@MartyG-RealSense 

It seems the rosdep install command revealed the fact that there is no definition of librealsense for Windows

c:\catkin_ws>rosdep install --from-paths c:\catkin_ws\src --ignore-src -r -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
realsense2_camera: No definition of [librealsense2] for OS [windows]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

I've tried to run catkin_make clean but still it cannot find the realsense cmake files. This problem was not seen when using vcpkg to install librealsense.

MartyG-RealSense, 2021-01-16:
I located a past case where this error occurred.

https://github.com/IntelRealSense/realsense-ros/issues/808

It sounds as though it is referring to the **package based** method of installing librealsense and the RealSense ROS wrapper together (**Method 1**) rather than building the wrapper from source code (**Method 2**, the method you have used).

eltonhcchan, 2021-01-17:
@MartyG-RealSense thx for your help. I've different combinations but still it cannot find the .make files. I think there may be some path definition missing. May try again later.

MartyG-RealSense, 2021-01-17:
Okay, please keep me updated.  Good luck!

MartyG-RealSense, 2021-01-23:
Hi @eltonhcchan  Do you have an update that you can provide about this case, please?   Thanks!

eltonhcchan, 2021-01-23:
@MartyG-RealSense I think I will try it later. BTW, I've found RTAP-Map working with L515 but need to lower the firmware.

MartyG-RealSense, 2021-01-23:
Thanks very much for the information @eltonhcchan  - good luck!

MartyG-RealSense, 2021-02-07:
Case closed due to no further comments received.

closed 2021-02-07 by MartyG-RealSense

sivaeinfo, 2021-05-04:
Hi, Actually my project is deep object pose estimation from (https://github.com/yehengchen/DOPE-ROS-D435) for object and human hand. For that I have tried, the below packages. Now I am facing few errors which I listed here
[image]
[image]
[image]

.
[image]

===================================================
Errors are 
--------------------------------------
i) roslaunch realsense2_camera rs_rgbd.launch  # Publishes RGB images to `/camera/color/image_raw`

process[camera/realsense2_camera_manager-1]: started with pid [3429]
process[camera/realsense2_camera-2]: started with pid [3430]
process[camera/color_rectify_color-3]: started with pid [3431]
process[camera/points_xyzrgb_hw_registered-4]: started with pid [3436]
[ INFO] [1620108763.382160336]: Initializing nodelet with 8 worker threads.
[ INFO] [1620108763.525868202]: RealSense ROS v2.2.7
[ INFO] [1620108763.525912472]: Running with LibRealSense v2.44.0
[ INFO] [1620108763.782599190]: getParameters...
[ INFO] [1620108763.829140477]: setupDevice...
[ INFO] [1620108763.829192188]: JSON file is not provided
[ INFO] [1620108763.829239499]: ROS Node Namespace: camera
[ INFO] [1620108763.829274609]: Device Name: Intel RealSense D435I
[ INFO] [1620108763.829299691]: Device Serial No: 923322073096
[ INFO] [1620108763.829319932]: Device FW version: 05.12.13.50
[ INFO] [1620108763.829349756]: Device Product ID: 0x0B3A
[ INFO] [1620108763.829367252]: Enable PointCloud: Off
[ INFO] [1620108763.829380406]: Align Depth: On
[ INFO] [1620108763.829404377]: Sync Mode: On
[ INFO] [1620108763.829524892]: Device Sensors: 
[ INFO] [1620108763.829617426]: Stereo Module was found.
[ INFO] [1620108763.829658363]: RGB Camera was found.
[ INFO] [1620108763.829740886]: Motion Module was found.
[ INFO] [1620108763.829818176]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1620108763.829846209]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1620108763.829882602]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1620108763.829919924]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1620108763.829949349]: Setting Dynamic reconfig parameters.
/opt/ros/melodic/lib/nodelet/nodelet: symbol lookup error: /home/bit/catkin_ws/devel/lib//librealsense2_camera.so: undefined symbol: _ZN20ddynamic_reconfigure19DDynamicReconfigure16registerVariableIiEEvRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEET_RKN5boost8functionIFvSA_EEES9_SA_SA_S9_
[camera/realsense2_camera_manager-1] process has died [pid 3429, exit code 127, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/bit/.ros/log/9b9f0040-ac9f-11eb-9fb1-d4bed9944d8b/camera-realsense2_camera_manager-1.log].
log file: /home/bit/.ros/log/9b9f0040-ac9f-11eb-9fb1-d4bed9944d8b/camera-realsense2_camera_manager-1*.log
[camera/points_xyzrgb_hw_registered-4] process has finished cleanly
log file: /home/bit/.ros/log/9b9f0040-ac9f-11eb-9fb1-d4bed9944d8b/camera-points_xyzrgb_hw_registered-4*.log
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/bit/.ros/log/9b9f0040-ac9f-11eb-9fb1-d4bed9944d8b/camera-realsense2_camera-2*.log

------------------------
ii) roslaunch dope dope.launch [config:=/path/to/my_config.yaml]  # Config file is optional; default is `config_pose.yaml`

process[dope-1]: started with pid [3678]
Traceback (most recent call last):
  File "/home/bit/catkin_ws/src/Deep_Object_Pose/nodes/dope", line 19, in <module>
    import tf.transformations
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf/__init__.py", line 30, in <module>
    from tf2_ros import TransformException as Exception, ConnectivityException, LookupException, ExtrapolationException
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_ros/__init__.py", line 38, in <module>
    from tf2_py import *
  File "/opt/ros/melodic/lib/python2.7/dist-packages/tf2_py/__init__.py", line 38, in <module>
    from ._tf2 import *
ImportError: dynamic module does not define module export function (PyInit__tf2)
[dope-1] process has died [pid 3678, exit code 1, cmd /home/bit/catkin_ws/src/Deep_Object_Pose/nodes/dope __name:=dope __log:=/home/bit/.ros/log/9b9f0040-ac9f-11eb-9fb1-d4bed9944d8b/dope-1.log].
log file: /home/bit/.ros/log/9b9f0040-ac9f-11eb-9fb1-d4bed9944d8b/dope-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
bit@bit-OptiPlex-990:~$

MartyG-RealSense, 2021-05-04:
Hi @sivaeinfo   It is inportant to match a particular RealSense ROS wrapper version as closely as possible with the librealsense version listed on that wrapper version's release notes.  Your log indicates that you are using the very ol ROS wrapper version 2.2.7 with the recent librealsense version 2.44.0.

For librealsense 2.44.0, ROS wrapper version **2.2.24** should be built for it.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.24

sivaeinfo, 2021-05-04:
Hi @MartyG-RealSense Will you explain how do I update the ROS wrapper 2.2.7 to ROS wrapper 2.2.2 in my package. Also how can I find the current ROS wrapper version in my linux system.

sivaeinfo, 2021-05-04:
Hi, Please help me to identify the ddynamic_reconfigure for ROS melodic.

MartyG-RealSense, 2021-05-04:
@sivaeinfo  Could you try using the command below please to install ddynamic_reconfigure for Melodic:

**sudo apt-get update apt-get install ros-melodic-ddynamic-reconfigure**

sivaeinfo, 2021-05-04:
Will you explain how do I update the ROS wrapper 2.2.7 to ROS wrapper 2.2.2 in my package. Also how can I find the current ROS wrapper version in my linux system.

MartyG-RealSense, 2021-05-04:
@sivaeinfo  I believe that you can upgrade by deleting the whole catkin workspace (**/catkin_ws**) and then following the wrapper installation instructions to create the /catkin_ws workspace folder again and build the wrapper within it.

The currently installed ROS wrapper version is indicated in the second half of the roslaunch log.

[image]

sivaeinfo, 2021-05-05:
I have made all the procedure in this package (https://github.com/yehengchen/DOPE-ROS-D435/tree/6125a8453c922892f9eac7faf8011661d52c875c). It works fine upto this steps.

**Installing Realsense-ros**
catkin workspace
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src/
Download realsense-ros pkg
git clone https://github.com/IntelRealSense/realsense-ros.git
cd realsense-ros/
git checkout `git tag | sort -V | grep -P "^\d+\.\d+\.\d+" | tail -1`
cd ..
Installing Realsense-ros
catkin workspace
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src/
Download realsense-ros pkg
git clone https://github.com/IntelRealSense/realsense-ros.git
cd realsense-ros/
git checkout `git tag | sort -V | grep -P "^\d+\.\d+\.\d+" | tail -1`
cd ..
**Download ddynamic_reconfigure**
cd src
git clone https://github.com/pal-robotics/ddynamic_reconfigure/tree/kinetic-devel
cd ..
cd src
git clone git clone https://github.com/ros/dynamic_reconfigure.git  (ROS Melodic)
cd ..
=============================
After I have entered this command. I have received following error.

**Pkg installation
catkin_init_workspace
cd ..
catkin_make clean**

How can I rectify this error.

MartyG-RealSense, 2021-05-05:
The installation of **ddynamic_reconfigure** looks like where the installation process may start going wrong.  It may be easier to install it using a sudo instruction instead:

**sudo apt-get install ros-melodic-ddynamic-reconfigure**