realsenseai/realsense-ros #1624: Error: An exception has been thrown: Device already streaming! How to correctly shut down / restart Intel Realsense D415  through terminal?

mso797, 2021-01-15:
I have the following camera and machine specifications.
Camera: Intel RealSense D415, Serial Number: 739112060258, Update Serial Number: 736313021040, Firmware Version: 05.12.09.00, USB Type: 3.2

OS: Ubuntu 16.04 LTS, Kernel: 4.4.0-198-generic, ROS Version: Kinetic

Every time that I turn on my robot, I get the following error.

`➜  ~ roslaunch realsense2_camera rs_camera.launch enable_color:=true
... logging to /home/mangokid/.ros/log/124e55be-5760-11eb-b7eb-f44d306f8357/roslaunch-wombat_cpu-1769.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://wombat_cpu:42651/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters:
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
[... 82 lines cut ...]
  what():  Device already streaming!
[camera/realsense2_camera_manager-2] process has died [pid 1811, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/mangokid/.ros/log/124e55be-5760-11eb-b7eb-f44d306f8357/camera-realsense2_camera_manager-2.log].
log file: /home/mangokid/.ros/log/124e55be-5760-11eb-b7eb-f44d306f8357/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/mangokid/.ros/log/124e55be-5760-11eb-b7eb-f44d306f8357/camera-realsense2_camera-3*.log`

I am able to resolve this error by physically unplugging/replugging the camera. Unfortunately, the camera is located deep inside of the robot and it is a rather tedious process. Is there a software only method that will allow me to reset the camera? I have tried all of the below solutions, none of them have worked:

1. `initial_reset:=true`

2. `import pyrealsense2 as rs2
ctx = rs2.context()
list = ctx.query_devices()
for dev in list:
      serial = dev.query_sensors()[0].get_info(rs2.camera_info.serial_number)
      # compare to desired SN
      dev.hardware_reset()`

3.  rs2::context ctx;
rs2::device dev = ctx.query_devices().front(); // Reset the first device
dev.hardware_reset();
rs2::device_hub hub(ctx);
dev = hub.wait_for_device();

3. Running linux scripts to reset usb port - https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line

mso797, 2021-01-15:
Upon further investigation, I've discovered that I am only able to resolve the issue by unplugging/replugging the type-c end of the USB cable. If I unplug/replug the normal usb end, the issue does not go away.

I guess that explains why resetting the usb port did not resolve the issue.

MartyG-RealSense, 2021-01-16:
Hi @mso797   As a starting point in investigating your case, I would recommend using RealSense ROs wrapper version **2.2.21** instead of the **2.2.8** version that your log states that you are using,  Wrapper 2.2.8 was designed for use with the very old librealsense version 2.25.0 from August 2019.  

It is important to match the RealSense ROS wrapper version as closely as possible with the librealsense version stated in the release notes for each ROS wrapper version.

[image]

MartyG-RealSense, 2021-01-22:
Hi @mso797    Do you require further assistance with this case, please?   Thanks!

mso797, 2021-01-22:
Hi, updating the ros wrapper didn't resolve the issue unfortunately. I ended up just connecting the camera through a usb 2.0 extension cable. This forces the camera to use usb 2.0 by default and prevent the error. But it also limits my usage of the camera. But it is still good enough for my application..

Thanks for your help!

MartyG-RealSense, 2021-01-22:
Thanks very much for the update!   If you are satisifed with the outcome, you can close the case with the **Close Issue** button underneath the comment box, unless you have further questions.   Thanks again!

closed 2021-01-29 by MartyG-RealSense