realsenseai/realsense-ros #1626: Customizing BaseRealSenseNode via composition/inheritance instead of modifying source

scottnothing, 2021-01-17:
In its current implementation `BaseRealSenseNode` prevents much in the way of customized behavior (such as some types of introspection, or changing sensor configs) due to its members being private. What would be the best way of having e.g. `sensors` be accessible via extension without having to modify or rewrite the BaseRealSenseNode? Or is this functionality prevented as a design choice?

Thanks

RealSenseSupport, 2021-01-19:
@scottnothing Can the config in launch files meet your request? What're the detailed configs do you want to configure? Thanks!

scottnothing, 2021-01-20:
Hi, thanks for the reply - the requirement is for mainly greater introspection, but could also allow custom behavior on frame callbacks 

- Customized diagnostics
- Getting sensor states e.g. exposure

RealSenseSupport, 2021-02-19:
@scottnothing Thanks for your feedback! Have submit your request to our engineering team. Any update will let you know.

RealSenseCustomerSupport, 2025-01-14:
----------------------------------------------

@scott-robotics Thanks for your suggestion! And sorry to tell you that we decide not to do this change after evaluation. Close this ticket accordingly.

closed 2025-01-16 by scott-robotics