realsenseai/realsense-ros #1629: T265 Camera not working on intel ros

dbgarasiya, 2021-01-17:
Camera Model | {T265 } |
| Firmware Version | {0.2.0.951} |
| Operating System & Version | {Ubuntu 18.04.5 LTS} |
| Kernel Version (Linux Only) | (Linux 5.4.0-42-generic) |
| Platform | PC |
| SDK Version | {2.41.0-0} |
| Language | {C++ } |
| Segment | {others } |

I am working on projects with two intel camera ( D435i and T265)
after installing robot_devkit when i play demo for both camera

D435i camera works but T265 camera not working , it gives error that device is not connected ( File name :T265_Error.txt)
[T265_Error.txt](https://github.com/IntelRealSense/realsense-ros/files/5825696/T265_Error.txt)

which gives in details.

my realsense-viewer command works for both camera ,

iamsavva, 2021-01-18:
oi, i get the same issue frequently when running `rs_d400_and_t265.launch`. Unplugging and replugging the cameras two or three times usually does it for me. What I usually do instead is I launch the cameras in separate terminal windows using `rs_camera.launch` and `rs_t265.launch` and then run another node that defines static publishers between the two.

dbgarasiya, 2021-01-19:
Thanks @iamsavva for showing interest  , but that is not solution for this problem

Best Regards

iamsavva, 2021-01-19:
> Thanks @iamsavva for showing interest , but that is not solution for this problem
> 
> Best Regards

have you tried launching t265 camera on its own? does the t265 camera pop up when you run `rs-enumerate-devices`?
also, i just realized that you are using realsense devices with ROS2 - never tried that, good luck.

dbgarasiya, 2021-01-20:
@**iamsavva** , i have already tested t265 camera alone with "rs-enumerate-devices" and "realsense-viewer" .
both commands works fine 

fist show all info. of camera (include serial number), and second will display both fisheye 1 and 2. display

closed 2026-10-23 by MartyG-RealSense