realsenseai/realsense-ros #1631: Running D435i on WSL (Ubuntu 18.04): No RealSense devices were found!

rk122019, 2021-01-18:
I'm fairly new to ROS and Linux, and I'm attempting to connect a D435i camera to ROS in Windows Subsystem for Linux (WSL, Ubuntu 18.04, Windows 10). I have ROS Melodic installed and followed the directions under Method 1 to install the realsense-ros tools. 

When I try to run `roslaunch realsense2_camera rs_camera.launch` in the terminal window and received:

[image]
[image]
[image]

It seems that the WSL isn't recognizing that there is a camera connected via USB. In the Windows 10 Device Manager, I can verify that the device is connected and is listed under "Universal Serial Bus controllers" as "USB Composite Device" (Port 14). I also have the Intel RealSense SDK installed on Windows 10, and the camera shows up in the RealSense Viewer with no problems. 

I've also been using the WSL to connect other (non-Intel RealSense) devices using "rosserial". WSL can connect to those devices with no problem as long as I specify the correct COM port (i.e. /dev/ttyS_). 

I also had trouble installing ROS directly onto Windows 10, so have not tried running realsense-ros directly from Windows 10.

In addition to the D435i, I also have a T265 which I tried connecting and running `roslaunch realsense2_camera rs_t265.launch`. I had the same result as with the D435i.

First, is the library compatible with Ubuntu being used on WSL? If so, are there any extra steps I may need to take to get the realsense-ros tools to recognize and connect to the cameras through USB?

dbgarasiya, 2021-01-19:
Have you tried simple relsense-viewer on both camera ?

MartyG-RealSense, 2021-01-19:
Hi @rk122019   I am not involved in T265 support, so please restrict the conversation to the D435i as indicated in the case title so that I am able to handle your case.   Thanks!

Intel's guide to installing ROS Melodic on Windows Subsystem For Linux (WSL) states that as WSL is based on Ubuntu, the normal Ubuntu installation process for ROS can be used.

https://software.intel.com/content/www/us/en/develop/blogs/working-with-recorded-camera-data-from-realsense-d400-series.html

If that is the case, you should be able to use the RealSense ROS wrapper's Ubuntu instructions for installing Librealsense, ROS and the RealSense ROS wrapper.  The above instruction guide states that the only difference may be in setting up the public key authentication, as described under the guide's **Install ROS** heading.

dbgarasiya, 2021-01-19:
Dear @MartyG-RealSense 
please suggest me from where I can get technical support for T265 camera ?

MartyG-RealSense, 2021-01-19:
hi @dbgarasiya  If it is a problem related to librealsense (e.g the RealSense Viewer), please create a new post using the **New Issue** button on the front page of the librealsense GitHub and include 'T265' in the case's title.

https://github.com/IntelRealSense/librealsense/issues/

If it is a RealSense ROS problem, please create a new post using the **New Issue** button on the front page of the RealSense ROS GitHub and include 'T265' in the case's title.

https://github.com/IntelRealSense/realsense-ros/issues/

dbgarasiya, 2021-01-19:
Thanks @**MartyG-RealSense** replying 
I have already created new post as New Issue with below link , but i have not got any reply on this issue
[https://github.com/IntelRealSense/realsense-ros/issues/1629]

please suggest any solution.

MartyG-RealSense, 2021-01-19:
If you have already posted your case then there is no further action that you need to take.  Please wait for an Intel support team member to respond.

dbgarasiya, 2021-01-19:
Thanks for help

rk122019, 2021-01-19:
@dbgarasiya  I tried running realsense-viewer in the WSL terminal. The command returned without having done anything (but it didn't throw any errors, either).

@MartyG-RealSense  Thank you for the confirmation. I went to the link you suggested and followed the instructions there, but I still have the same problem with the D435i ("No RealSense devices were found!"). Do you have any other suggestions?

Since I already had ROS installed, the only command I ran to install the Realsense library and ROS wrapper was `sudo apt-get install ros-melodic-realsense2-camera`. Would this command install all necessary dependencies, such as the librealsense library, or would I need to do that separately?

MartyG-RealSense, 2021-01-19:
@rk122019   Reading the Intel WSL guide again, I see that it covers installing Ubuntu and then installing the Melodic version of ROS (not the RealSense ROS wrapper, which is installed **after** ROS Melodic is installed.

So, if ROS Melodic is installed successfully then it is reasonable to assume that, as per the RealSense ROS wrapper instructions, installing the RealSense ROS wrapper as the next step should set up the camera to work with the Ubuntu system installed via WSL.

I assume that you have been able to test that ROS Melodic is working correctly on WSL before commencing installation of the RealSense ROS wrapper?

The **sudo apt-get install ros-melodic-realsense2-camera** command looks fine to me.  And you have Melodic installed already.  This command for Ubuntu should install librealsense and the RealSense ROS wrapper together.  The **Method 1** documentation states that this installation will include dependencies.

Are you launching RealSense ROS with a roslaunch instruction suited to the D435i? 

**roslaunch realsense2_camera rs_camera.launch**

rk122019, 2021-01-19:
@MartyG-RealSense  Yes, I've tested ROS Melodic with WSL using the "rosserial" library (for another application) and it works without a problem.

I tried plugging in the D435i only, then completely rebooting my computer, then opened Ubuntu/WSL and running the roslaunch command. The same issue still occurs.

MartyG-RealSense, 2021-01-19:
I do not often see the **No realsense devices were found!** error.  I believe though that it is caused by **librealsense** rather than the RealSense ROS wrapper.  So ROS and the RealSense ROS wrapper could be working, but the librealsense SDK is causing camera detection to fail.  As you have installed the ROS wrapper and librealsense at the same time using **Method 1** there is not anything that you could do to influence librealsense's installation.

I note that you mentioned earlier that you tried running the Ubuntu version of the RealSense Viewer in WSL by inputting **realsense-viewer** into the Ubuntu terminal and nothing happened.   That in itself is an indicator that there may be a problem with the WSL installation of librealsense.  You have already validated that the camera works correctly with the Viewer in Windows.

The only way forward for using ROS in Linux in your WSL installation may be to try building librealsense and the ROS wrapper separately using the wrapper's **Method 2** instructions.   Or attempt again to get ROS and the ROS wrapper working directly in Windows without WSL.

rk122019, 2021-01-20:
@MartyG-RealSense  Thanks so much for your help. One more thing I tried was to completely uninstall Ubuntu 18.04 from my computer, reinstall, and follow the instructions from scratch. I used [https://wiki.ros.org/melodic/Installation/Ubuntu](these instructions) to install ROS Melodic on the WSL, then tried both Method 1 and Method 2 to install everything again. I still receive the same error. 

I will continue trying different fixes and updated if I find a solution.

dbgarasiya, 2021-01-20:
@rk122019 
run below commands , you will get it work of realsense-viewer
[link.txt](https://github.com/IntelRealSense/realsense-ros/files/5840058/link.txt)

rk122019, 2021-01-20:
Thanks for the suggestion, @dbgarasiya. Unfortunately, that did not work. I had already run those commands when building from source, so they did not update anything in the WSL.

rk122019, 2021-01-21:
An update: I've given up on trying to get WSL working, and instead turned to installing the ROS wrapper in Windows 10 directly. It seems to be _almost_ working. However, when I run the `catkin_make -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release` command, I get some errors that I believe are related to CMake. Specifically `cl: Command line warning D9002 : ignoring unknown option '-std=c++0x'` and `cl: Command line warning D9002 : ignoring unknown option '-std=c++11'`. 

Everything still seems to build fine, and I can actually run `roslaunch realsense_camera2 rs_camera.launch` successfully now, but I'm not sure if these errors affect the build at all. 

Is there a way to fix this? Perhaps adjusting settings in Visual Studio 2019 to change CMake? Thanks for the help!

`[  4%] Building CXX object CMakeFiles/Project__setup_util.py_exec_install_python.dir/catkin_generated/add_python_executable/Project__setup_util.py_exec_install_python/_setup_util.cpp.obj
_setup_util.cpp
[  8%] Linking CXX executable catkin_generated\windows_wrappers\Project__setup_util.py_exec_install_python\_setup_util.exe
[  8%] Built target Project__setup_util.py_exec_install_python
[ 12%] Building CXX object ddynamic_reconfigure/CMakeFiles/ddynamic_reconfigure.dir/src/ddynamic_reconfigure.cpp.obj
cl : Command line warning D9002 : ignoring unknown option '-std=c++0x'
ddynamic_reconfigure.cpp
C:\catkin_ws\src\ddynamic_reconfigure\src\ddynamic_reconfigure.cpp(421): warning C4267: '=': conversion from 'size_t' to 'dynamic_reconfigure::Group_<std::allocator<void>>::_id_type', possible loss of data
C:\catkin_ws\src\ddynamic_reconfigure\src\ddynamic_reconfigure.cpp(476): warning C4267: '=': conversion from 'size_t' to 'dynamic_reconfigure::GroupState_<std::allocator<void>>::_id_type', possible loss of data
[ 16%] Linking CXX shared library C:\catkin_ws\devel\bin\ddynamic_reconfigure.dll
   Creating library C:\catkin_ws\devel\lib\ddynamic_reconfigure.lib and object C:\catkin_ws\devel\lib\ddynamic_reconfigure.exp
[ 16%] Built target ddynamic_reconfigure
[ 20%] Building CXX object ddynamic_reconfigure/CMakeFiles/fake_dynamic_reconfigure_server.dir/test/fake_dynamic_reconfigure_server.cpp.obj
cl : Command line warning D9002 : ignoring unknown option '-std=c++0x'
fake_dynamic_reconfigure_server.cpp
[ 25%] Linking CXX executable C:\catkin_ws\devel\lib\ddynamic_reconfigure\fake_dynamic_reconfigure_server.exe
[ 25%] Built target fake_dynamic_reconfigure_server
[ 29%] Building CXX object ddynamic_reconfigure/CMakeFiles/test_bool_dynamic_reconfigure_server.dir/test/test_bool_dynamic_reconfigure_server.cpp.obj
cl : Command line warning D9002 : ignoring unknown option '-std=c++0x'
test_bool_dynamic_reconfigure_server.cpp
[ 33%] Linking CXX executable C:\catkin_ws\devel\lib\ddynamic_reconfigure\test_bool_dynamic_reconfigure_server.exe
[ 33%] Built target test_bool_dynamic_reconfigure_server
[ 33%] Built target std_msgs_generate_messages_cpp
[ 33%] Built target sensor_msgs_generate_messages_cpp
[ 33%] Built target _realsense2_camera_generate_messages_check_deps_IMUInfo
[ 33%] Built target _realsense2_camera_generate_messages_check_deps_Extrinsics
[ 41%] Built target realsense2_camera_generate_messages_cpp
[ 41%] Built target std_msgs_generate_messages_eus
[ 41%] Built target sensor_msgs_generate_messages_eus
[ 54%] Built target realsense2_camera_generate_messages_eus
[ 54%] Built target std_msgs_generate_messages_lisp
[ 54%] Built target sensor_msgs_generate_messages_lisp
[ 62%] Built target realsense2_camera_generate_messages_lisp
[ 62%] Built target std_msgs_generate_messages_py
[ 62%] Built target sensor_msgs_generate_messages_py
[ 75%] Built target realsense2_camera_generate_messages_py
[ 75%] Built target std_msgs_generate_messages_nodejs
[ 75%] Built target sensor_msgs_generate_messages_nodejs
[ 83%] Built target realsense2_camera_generate_messages_nodejs
[ 83%] Built target realsense2_camera_generate_messages
[ 83%] Built target _catkin_empty_exported_target
[ 83%] Built target dynamic_reconfigure_generate_messages_nodejs
[ 83%] Built target dynamic_reconfigure_generate_messages_eus
[ 83%] Built target dynamic_reconfigure_generate_messages_lisp
[ 83%] Built target dynamic_reconfigure_generate_messages_cpp
[ 83%] Built target roscpp_generate_messages_cpp
[ 83%] Built target roscpp_generate_messages_eus
[ 83%] Built target dynamic_reconfigure_generate_messages_py
[ 83%] Built target rosgraph_msgs_generate_messages_py
[ 83%] Built target rosgraph_msgs_generate_messages_eus
[ 83%] Built target geometry_msgs_generate_messages_lisp
[ 83%] Built target actionlib_msgs_generate_messages_nodejs
[ 83%] Built target rosgraph_msgs_generate_messages_nodejs
[ 83%] Built target geometry_msgs_generate_messages_cpp
[ 83%] Built target nav_msgs_generate_messages_nodejs
[ 83%] Built target diagnostic_msgs_generate_messages_lisp
[ 83%] Built target nodelet_generate_messages_lisp
[ 83%] Built target roscpp_generate_messages_nodejs
[ 83%] Built target rosgraph_msgs_generate_messages_lisp
[ 83%] Built target tf_generate_messages_nodejs
[ 83%] Built target nav_msgs_generate_messages_cpp
[ 83%] Built target roscpp_generate_messages_lisp
[ 83%] Built target roscpp_generate_messages_py
[ 83%] Built target nav_msgs_generate_messages_py
[... 114 lines cut ...]
            T=sensor_msgs::CameraInfo_<std::allocator<void>>
        ]
[... cut ...]
C:\opt\ros\melodic\x64\include\sensor_msgs/impl/point_cloud2_iterator.h(129): warning C4267: '=': conversion from 'size_t' to 'sensor_msgs::PointCloud2_<std::allocator<void>>::_width_type', possible loss of data
C:\opt\ros\melodic\x64\include\sensor_msgs/impl/point_cloud2_iterator.h(130): warning C4267: '=': conversion from 'size_t' to 'sensor_msgs::PointCloud2_<std::allocator<void>>::_row_step_type', possible loss of data
[100%] Linking CXX shared library C:\catkin_ws\devel\bin\realsense2_camera.dll
   Creating library C:\catkin_ws\devel\lib\realsense2_camera.lib and object C:\catkin_ws\devel\lib\realsense2_camera.exp
[100%] Built target realsense2_camera`

rk122019, 2021-01-21:
Follow-up on the above issue: when running the roslaunch command, I've noticed that it is using LibRealSense v2.34.0, which is significantly behind the current v2.41.0 version. I have the most up-to-date SDK installed in my Program Files, but in order to get a successful build of the catkin workspace, I had to git clone "vcpkg" from [here](https://github.com/microsoft/vcpkg/releases/tag/2020.11-1). 

I'm guessing it's using the librealsense version built with this vcpkg (which is out of date) to build the catkin workspace. Any suggestions on how to make sure the catkin workspace uses the right librealsense version?

MartyG-RealSense, 2021-01-22:
Using SDK **2.34.0** is not advisable if it is possible to avoid doing so, as it had a problem with continuously generated timing errors for some RealSense users.

Another RealSense user had a problem with an out of date SDK version being installed by vcpkg a few weeks ago.  In their case, SDK **2.37.0** was installed.   In the link below they explain their solution, which enabled SDK **2.40.0** to be installed.

https://github.com/IntelRealSense/realsense-ros/issues/1509#issuecomment-753965795

rk122019, 2021-01-22:
Thank you so much, @MartyG-RealSense! The wrapper is starting a node now with the D435i and using librealsense v2.40.0. 

A one follow-up issue: I'm now trying to use the wrapper with a T265 camera (on its own, without the D435i) using `roslaunch realsense2_camera rs_t265.launch`. However, I have the same issue as I originally had with the WSL ("No RealSense devices were found!"), even though I'm running directly on Windows 10. The camera shows up in the RealSense Viewer and my device manager with no problems.

@MartyG-RealSense, I know you said you don't work with T265, but if you know of any similar issues or know someone who could help so I can use T265, I'd appreciate it!

MartyG-RealSense, 2021-01-22:
Great news @rk122019 that you succeeded with WSL and your D435i - thanks for the update!

I am unable to offer T265 guidance.  If booting the computer with only the T265 plugged in and the D435i unplugged does not work, I recommend creating a new case on this RealSense ROS forum and including 'T265' in the title of the case.

rk122019, 2021-01-27:
Since I got the D435i working, I'll close this by mentioning the steps I followed to get there (in case any other Windows 10 users have the same problem):

1. Install Visual Studio 2019. For me, I had to add the following workspaces during installation:
  - .NET desktop development
  - Desktop development with C++
  - Universal Windows Platform development
  - Visual Studio extension development
  - Linux development with C++
  - Windows 10 SDK (10.0.19041.0)

2. Setup ROS Melodic as indicated in the instructions on realsense-ros.
3. Install vcpkg on the C: drive
  - Create a folder in "C:" called "dev"
  - Open the ROS-enabled command prompt from step 2
  - Clone vcpkg [from here](https://github.com/microsoft/vcpkg) into the "dev" folder
  - Run ".\vcpkg\bootstrap-vcpkg.bat"

4. Re-install vcpkg in the ROS Melodic installation (without this step, vcpkg will install an outdated version of librealsense)
  - Navigate to C:\opt\ros\melodic\x64\tools
  - Remove the existing vcpkg folder
  - In the ROS-enabled command prompt, navigate to this same folder
  - Clone vcpkg again as in number (3) above and run the .bat command

5. Continue to follow "Method 2" (for Windows) on the Realsense ROS wrapper page
  - While still in "catkin_ws\src", also clone [ddynamic_reconfigure](https://github.com/pal-robotics/ddynamic_reconfigure)
  - Continue to build and install
  - Don't forget to run "devel\setup.bat"

In my experience, you may have to play around with the version of realsense-ros that you clone. Make sure it aligns with the version of librealsense that vcpkg installs. 

In my experience, this got `roslaunch realsense2_camera rs_camera.launch` to run on Windows 10, but I still run into the original issue ("No RealSense devices were found!") with `roslaunch realsense2_camera rs_t265.launch` when using T265.

closed 2021-01-27 by rk122019

MartyG-RealSense, 2021-01-27:
Thanks so much for sharing your extensively detailed D435i guide with the RealSense community @rk122019   :)

mgopshtein, 2021-05-03:
> using T265

Hi, did you mange to make T265 work with ROS on Windows? I'm getting the same "device not found" error.

And thanks for the detailed information, was very helpful!

rk122019, 2021-05-04:
I unfortunately did not find a solution for the T265 on Windows. I resorted to creating an Ubuntu dual boot on my computer and doing all of my T265 ROS work on Ubuntu 18.04. 

Best of luck!