realsenseai/realsense-ros #1632: ROS, Missing frames when using D435 and multiple devices problems

BIGFatming, 2021-01-20:
Here are some basic information: We use two Realsense D435 cameras, both of the devices connect to a same industrial PC. The OS is Ubuntu 16.04. We run this command in terminal to launch the cameras:

"roslaunch realsense2_camera rs_multiple_devices.launch serial_no_camera1:=909512070754 serial_no_camera2:=819612073319"

we run this comment to record images and form a rosbag:

"record /camera1/color/image_raw/compressed /camera1/depth/ime_rect_raw/compressedDepth /camera1/infra1/image_rect_raw/compressed /camera1/infra2/image_rect_raw/compressed /camera2/color/image_raw/compressed /camera2/depth/image_rect_raw/compressedDepth /camera2/infra1/image_rect_raw/compressed /camera2/infra2/image_rect_raw/compressed -O 'bag_name'.bag"

We did not changes any default parameter of the cameras.

We find problems when we check the recorded data, we decomposed the rosbag and extracted every images it contains. The FPS of the RGB images is only 15 (supposed to be 30) while the infra images can reach the 30FPS, but the FPS of the depth images is too low. As the following pictures show.
[image]
PIC1. the time stamp show the RGB FPS is only 15
[image]
PIC2. the infra can reach 30 FPS
[image]
PIC3. FPS of depth is horrible......

Additionally, the time stamp of two cameras is different, the PIC4 and PIC5 show this problem.
[image]
PIC4 The images captured by camera1
[image]
PIC5 The images captured by camera2

SO, what causes the problem of FPS?
AND, what causes the problem of time stamp? Are there any method to make multiple devices work synchronously？

MartyG-RealSense, 2021-01-20:
Hi @BIGFatming   A constant FPS can be enforced in two ways:

-  Disabling auto-exposure and setting manual exposure within a certain range.

https://github.com/IntelRealSense/librealsense/issues/1957#issuecomment-400715579

- Enabling auto-exposure and disabling the **Auto-Exposure Priority**

Auto-Exposure Priority can be enabled and disabled in ROS with a **rosparam** using the details in the link below,

https://github.com/IntelRealSense/realsense-ros/issues/1349#issuecomment-681695259

The link below features a case where a RealSense ROS user used these methods successfully to set FPS at 30 instead of 15.

https://github.com/IntelRealSense/realsense-ros/issues/1357

In regard to timestamps in your multicam setup, the case in the link below may be a useful reference.  The RealSense ROS user in that discussion had a multicam system and was experiencing erratic timestamps.   It was recommended to them that they disable **global timestamps**.

https://github.com/IntelRealSense/librealsense/issues/8003

BIGFatming, 2021-01-21:
I will try exposure settings and give a feedback ASAP. 
I checked some closed issues about timestamp, and saw #1451 mentioned a master-slave setting, I wonder if it works.

BIGFatming, 2021-01-21:
@MartyG-RealSense

 I disabled the auto_exposure_priority and the FPS of RGB images has reached 30. **BUT, the FPS of depth images is still <10**. 

Additionally, I set the **global timestamps to false** but the timestamps of two cameras is still different. I also **set inter_cam_sync_mode to 1(Master) for camera1 and 2(Slave) for camera2**, but still not working. I wonder if multiple cameras can really capture the images at the same time, which means the images captured by different cameras has corresponding timestamps.

BIGFatming, 2021-01-21:
@MartyG-RealSense 
I checked the website https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration you mentioned in #1454. Accroding to the website and your comment, if I need to **buy the wire to hardware sync the devices and also set the inter_cam_sync_mode to master and slave** so that I can make two devices work simultaneously?

BIGFatming, 2021-01-21:
@MartyG-RealSense 
I checked the website https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration you mentioned in #1451 . Accroding to the website and your comment, if I need to buy the wire to hardware sync the devices and also set the inter_cam_sync_mode to master and slave so that I can make two devices work simultaneously?

MartyG-RealSense, 2021-01-21:
Hi @BIGFatming   If you experience problems when using **roslaunch realsense2_camera rs_multiple_devices.launch** and specifying serial numbers, an alternate approach to ROS multicam on the same computer is to use **roslaunch realsense2_camera rs_camera.launch** and set up a separate terminal for each camera.  This process is described in Intel's ROS multiple camera tutorial:

https://www.intelrealsense.com/how-to-multiple-camera-setup-with-ros/

The **-all** condition of rosbag record can record data from all published topics.

http://wiki.ros.org/rosbag/Commandline#rosbag_record

The Inter Cam Sync Mode can be set from within ROS if you wish to sync the timestamp of slave cameras to the timing of a master camera, as described in the link in https://github.com/IntelRealSense/realsense-ros/issues/1451 

You would need to assemble sync cabling though if you wish to physically connect the cameras together, as described in the multiple camera white paper.

A more recent reference document for multiple camera sync is the **External Synchronization (Genlock)** white paper:

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras

BIGFatming, 2021-01-22:
@MartyG-RealSense 
Thank you and I am a bit clear about the Synchronization stuff. Now I am going to focus on the problem of the FPS of depth images.
I will try "roslaunch realsense2_camera rs_camera.launch" to setup two cameras separately and give you a feedback ASAP.

MartyG-RealSense, 2021-01-22:
Thanks very much @BIGFatming - I look forward to your test feedback.  Good luck!

BIGFatming, 2021-01-23:
@MartyG-RealSense 
The experiment results that using "roslaunch realsense2_camera rs_camera.launch" to setup two cameras separately is useless, the FPS of depth images is still <10.
And I also find out that the RGB images FPS is lowered to 15 although I have set the auto_exposure_priotiry to false, I also tried to use "roslaunch realsense2_camera rs_multiple_devices.launch" to launch just like what I did last time, but nothing changed. That is extremely weird since this problem should be solved last time! The differences between experiments this time and last time is when I launch the camera, there is a warning: 

WARNING: disk usage in log directory [/home/sti/.ros/log] is over 1GB. It's recommended that you use the 'rosclean' command.

might it be the reason why the RGB images frequency has been lowerd?

MartyG-RealSense, 2021-01-23:
FPS may halve or drop further than half if depth-color alignment is being performed, though I cannot see any alignment being done in your project.  

Going back to your opening message, I see that you are recording compressed depth.  I recall a case from November 2020 where a RealSense ROS user was experiencing extreme delay when using compression.

https://github.com/IntelRealSense/realsense-ros/issues/1484

BIGFatming, 2021-01-25:
@MartyG-RealSense 
Thanks for your suggestion, I set the **enable_sync:=true**, now I can get frames of different topics return with same timestamps.
I also checked the #1484 and set /camera/depth/image_rect_raw/compressedDepth/png_level to 1, now the depth_fps reaches 30.

Now, the only unsolved issue is the RGB_fps, it is still around 15. I set /camera/color/image_rect_raw/compressed/png_level to 1 and it dose not work. I have noticed that you mentioned a **compress_rate setting** in #1484 but I did not find it in launch documents and "rosrun rqt_reconfigure rqt_reconfigure", could you tell me how to change this setting in detail?

MartyG-RealSense, 2021-01-25:
Greatto hear that you made progress @BIGFatming 

My research of the **<param name="compressed_rate"  type="double" value="$(arg rate)"/>**line mentioned in https://github.com/IntelRealSense/realsense-ros/issues/1484 suggests that it is usually used exactly in that form, and implemented with the following steps:

1.  Edit the launch file and in the section of the file with **<arg>** instructions, add this line:

**<arg name="rate"  default="5"/>**

Change the rate value from '5' to whatever you want it to be.

2.  Then in the section of the launch file with **<param>** instructions, add this line:

**<param name="compressed_rate"  type="double" value="$(arg rate)"/>**

This line substitutes into the bracket the value of the **rate** arg that you defined earlier in the launch file in step 1.

***************

Since you achieved 30 FPS for RGB by disabling **Auto-Exposure Priority**, I would suggest using that False setting again for Auto-Exposure Priority if you are not currently using it.

MartyG-RealSense, 2021-01-26:
Oh yes, GItHub does not show scripting that is within arrow brackets unless formatted as a code quote.  I prefer bolding because of its better visibility.  The line was `<param name="compressed_rate"  type="double" value="$(arg rate)"/>`.  Sorry about that.

Please do let me know how your tests go once you have multiple cameras again.  Good luck!

BIGFatming, 2021-01-27:
@MartyG-RealSense 
The test goes well, images from each topic share same timestamps, and the frequency reaches 30. I think it is time to close the issue.
But there is an episode I think is worth reporting. At the beginning, I launch cameras using  **roslaunch realsense2_camera rs_multiple_devices.launch** then those problems happen again: lowered frequency, messy timestamps... so I close the process and then launch only one camera using **realsense2_camera rs_camera.launch** just like what I did yesterday. However, the RGB images frequency is lowered to 15, and this problem should be solved yesterday. So I totally restart the ros process by restarting the **roscore**. This time, I launch cameras separately in two terminals using **realsense2_camera rs_camera.launch** and everything is well, no problems ever happen again.
Anyway, again, Thanks for your patient assist.

MartyG-RealSense, 2021-01-27:
Thanks so much for your detailed test feedback that will benefit other readers of this case!  It is great to hear of your success.  :)

As suggested, I will close this case as you are satisfied with the outcome.  Thanks again.

closed 2021-01-27 by MartyG-RealSense