realsenseai/realsense-ros #1633: 'rs2::invalid_value_error' hwmon command 0x7d failed

grupot3, 2021-01-20:
These are the specifications

Jetson TX2
Firmware 5.12.7.100
v2.2.11 of the ROS package realsense2_camera
v2.31.0 of librealsense

In realsense-viewer, I've tried both the camera d435 and the camera d435i, and they both work. The camera d435 also works when running `roslaunch realsense2_camera rs_camera.launch `. 

However, when I run `roslaunch realsense2_camera rs_camera.launch ` with the d435i camera, I get the error 
`terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  hwmon command 0x7d failed. Error type: No data to return (-21).`

I've tried also different firmwares of the camera, use the dynamic calibration tool, and calibrate the IMU of the camera, but the error is still there. I haven't modified anything from the launch file

What should I do?

The full output of the roslaunch command is below

Thanks!
______________________________________________________
SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
 * /camera/realsense2_camera/json_file_path: 
 * /camera/realsense2_camera/linear_accel_cov: 0.01
 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame
 * /camera/realsense2_camera/pointcloud_texture_index: 0
 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /camera/realsense2_camera/pose_frame_id: camera_pose_frame
 * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic...
 * /camera/realsense2_camera/publish_odom_tf: True
 * /camera/realsense2_camera/publish_tf: True
 * /camera/realsense2_camera/rosbag_filename: 
 * /camera/realsense2_camera/serial_no: 
 * /camera/realsense2_camera/tf_publish_rate: 0.0
 * /camera/realsense2_camera/topic_odom_in: odom_in
 * /camera/realsense2_camera/unite_imu_method: 
 * /camera/realsense2_camera/usb_port_id: 
 * /rosdistro: kinetic
 * /rosversion: 1.12.17

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

ROS_MASTER_URI=XXXXXXXXXXXXXXXXXX [edited]

process[camera/realsense2_camera_manager-1]: started with pid [13121]
process[camera/realsense2_camera-2]: started with pid [13122]
[ INFO] [1611114096.941755765]: Initializing nodelet with 6 worker threads.
[ INFO] [1611114097.027678846]: RealSense ROS v2.2.11
[ INFO] [1611114097.027782494]: Running with LibRealSense v2.31.0
[ INFO] [1611114097.404687803]:  
 19/01 22:41:37,737 WARNING [547272782224] (types.cpp:49) hwmon command 0x4f failed. Error type: No data to return (-21).
[ INFO] [1611114097.794427636]: Device with serial number XXXXXXXXXX [edited] was found.

[ INFO] [1611114097.794715764]: Device with physical ID 2-1-3 was found.
[ INFO] [1611114097.794890132]: Device with name Intel RealSense D435I was found.
[ WARN] [1611114097.809165936]: Error extracting usb port from device with physical ID: 2-1-3
Please report on github issue at https://github.com/IntelRealSense/realsense-ros
[ INFO] [1611114097.822556397]: getParameters...
[ INFO] [1611114098.029999350]: setupDevice...
[ INFO] [1611114098.030073430]: JSON file is not provided
[ INFO] [1611114098.030127062]: ROS Node Namespace: camera
[ INFO] [1611114098.030169782]: Device Name: Intel RealSense D435I
[ INFO] [1611114098.030205430]: Device Serial No: XXXXXXXXXX [edited]
[ INFO] [1611114098.030240214]: Device physical port: 2-1-3
[ INFO] [1611114098.030272822]: Device FW version: 05.12.10.00
[ INFO] [1611114098.030307542]: Device Product ID: 0x0B3A
[ INFO] [1611114098.030341270]: Enable PointCloud: Off
[ INFO] [1611114098.030373974]: Align Depth: Off
[ INFO] [1611114098.030404214]: Sync Mode: Off
[ INFO] [1611114098.030957718]: Device Sensors: 
[ INFO] [1611114098.031396854]: Stereo Module was found.
[ INFO] [1611114098.031509014]: RGB Camera was found.
[ INFO] [1611114098.031559190]: Motion Module was found.
[ INFO] [1611114098.031645782]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1611114098.031687670]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1611114098.031723862]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1611114098.031760534]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1611114098.032187798]: num_filters: 0
[ INFO] [1611114098.032256886]: Setting Dynamic reconfig parameters.
 19/01 22:41:40,084 WARNING [547272782224] (types.cpp:49) hwmon command 0x7d failed. Error type: No data to return (-21).
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  hwmon command 0x7d failed. Error type: No data to return (-21).
[camera/realsense2_camera_manager-1] process has died [pid 13121, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/nvidia/.ros/log/fb28b25c-5ad0-11eb-b282-00044b8c44f2/camera-realsense2_camera_manager-1.log].
log file: /home/nvidia/.ros/log/fb28b25c-5ad0-11eb-b282-00044b8c44f2/camera-realsense2_camera_manager-1*.log

MartyG-RealSense, 2021-01-20:
Hi @grupot3   The IMU functions (**enable_gyro** and **enable_accel**) were not set to false by default in the RealSense ROS wrapper until wrapper version 2.2.15 onwards, and the log states that both of these IMU functions are True in your wrapper 2.2.11.

Could you try adding the instructions below to the end of your roslaunch statement to set accel and gyro to false please and see whether this resolves your **hwmon command 0x7d failed. Error type: No data to return (-21)** launch exit problem, please?

**enable_gyro:=false enable_accel:=false**

grupot3, 2021-01-20:
Hi @MartyG-RealSense, thanks for the help. I'm afraid the error still appears

I'm running

`roslaunch rs_camera.launch enable_gyro:=false enable_accel:=false `

The output is below:

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
 * /camera/realsense2_camera/infra_fps: 30
 * /camera/realsense2_camera/infra_height: 480
 * /camera/realsense2_camera/infra_width: 640
 * /camera/realsense2_camera/initial_reset: False
[... 42 lines cut ...]
[ INFO] [1611150591.836271745]: Device Name: Intel RealSense D435I
[ INFO] [1611150591.836304609]: Device Serial No: XXXXXXXXXXXXXXXXX
[ INFO] [1611150591.836339713]: Device physical port: 2-1-2
[ INFO] [1611150591.836373441]: Device FW version: 05.12.10.00
[ INFO] [1611150591.836406817]: Device Product ID: 0x0B3A
[ INFO] [1611150591.836438978]: Enable PointCloud: Off
[ INFO] [1611150591.836471586]: Align Depth: Off
[ INFO] [1611150591.836504738]: Sync Mode: Off
[ INFO] [1611150591.837046051]: Device Sensors: 
[ INFO] [1611150591.837510372]: Stereo Module was found.
[ INFO] [1611150591.837649412]: RGB Camera was found.
[ INFO] [1611150591.837706436]: Motion Module was found.
[ INFO] [1611150591.837801541]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1611150591.837845861]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1611150591.837882981]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1611150591.837922149]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1611150591.838330662]: num_filters: 0
[ INFO] [1611150591.838376966]: Setting Dynamic reconfig parameters.
 20/01 08:49:53,861 WARNING [547943870864] (types.cpp:49) hwmon command 0x7d failed. Error type: No data to return (-21).
terminate called after throwing an instance of 'rs2::invalid_value_error'
  what():  hwmon command 0x7d failed. Error type: No data to return (-21).
[camera/realsense2_camera_manager-1] process has died [pid 4341, exit code -6, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/nvidia/.ros/log/3d7f23f2-5b26-11eb-a478-00044b8c44f2/camera-realsense2_camera_manager-1.log].
log file: /home/nvidia/.ros/log/3d7f23f2-5b26-11eb-a478-00044b8c44f2/camera-realsense2_camera_manager-1*.log

Thanks!

MartyG-RealSense, 2021-01-20:
Thanks very much @grupot3 for trying.   Next, could you try adding an instruction to the roslaunch to perform a reset of the camer at launch:

**initial_reset:=true**

grupot3, 2021-01-20:
Same error appears with initial_reset:=true

grupot3, 2021-01-20:
After more debugging, it seems that the issue happens when calling `sensor.get_option(option)`, where `option` is such that 
`nh1.resolveName(option_name)=="/camera/stereo_module/emitter_on_off"`

A quick workaround was to add 
```
if(nh1.resolveName(option_name)=="/camera/stereo_module/emitter_on_off"){
continue;
}
```
before the `if (is_checkbox(sensor, option))` of `base_realsense_node.cpp`.

The camera is working now after that fix, but is there a better way to solve this?

MartyG-RealSense, 2021-01-21:
I am not personally aware of a better method than your own workaround.

closed 2021-02-05 by MartyG-RealSense