realsenseai/realsense-ros #1635: Rtabmap..cpp:1148::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 76684 is ignored!

punchnami, 2021-01-21:
| Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (05.12.09) | 
| Operating System & Version |   Linux (Ubuntu 18.04.5)   | 
| Kernel Version (Linux Only)    |  (4.9.14)                                         | 
| Platform                                 | NVIDIA Jetson NANO  |
| SDK Version                            |  2.0                          | 

After I followed all the instruction and  used the command
`roslaunch realsense2_camera opensource_tracking.launch `

Everything works just fine for a while with the **odo quality** values appear on the log then it gave the value of 0  with warning and error as seen on the image below

[image]

I tried executing the following command but the problem still occurs.

`roslaunch realsense2_camera opensource_tracking.launch enable_gyro:=true enable_accel:=true enable_infra:=true`

**Secondly**, to save the rosbag file, the rviz program should be opened while saving? (I was trying save the rosbag file while the rviz program was executing, it didn't stop and the file size was up to 40Gb so I Ctrl+c the terminal respectively)

Thank you in advance!
Any help is appreciated :)

MartyG-RealSense, 2021-01-22:
Hi @punchnami   Are you followng the instructions in the D435i SLAM guide that uses **roslaunch realsense2_camera opensource_tracking.launch**, please?  

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

If you are using this guide and are having problems with loss of tracking, then use of **RTABMAP** has been discussed in regard to resetting odometry with **relocalization**.

https://github.com/IntelRealSense/realsense-ros/issues/1529

punchnami, 2021-01-25:
Thank you for your response as always! @MartyG-RealSense , I tried to slow down the movement of my camera and it's in fact affected the odometry value and didn't break the sequences! 

And yes, I did the installation as instructed in Intel's D435i SLAM guide but there're another problem when I tried to play my recorded rosbag file with the following command 

`roscore >/dev/null 2>&1 &
rosparam set use_sim_time true
rosbag play my_bagfile_1.bag --clock
roslaunch realsense2_camera opensource_tracking.launch offline:=true`

Nothing seems to show up on the RViz.No map cloud and point cloud data as well as the images.

MartyG-RealSense, 2021-01-25:
Given that the roslaunch for this D435i SLAM tutorial now requires additional instructions to enable the IMU after ROS wrapper version **2.2.15**, it may be that you need to add the same terms to the offline roslaunch instruction to play back the bag.

roscore >/dev/null 2>&1 &
rosparam set use_sim_time true
rosbag play my_bagfile_1.bag --clock
roslaunch realsense2_camera opensource_tracking.launch offline:=true **enable_gyro:=true enable_accel:=true enable_infra:=true**

punchnami, 2021-01-25:
Oh I see, thank you so much @MartyG-RealSense !

closed 2021-01-25 by punchnami