realsenseai/realsense-ros #1636: Unable to locate package (Noetic version)

Plopao, 2021-01-21:
Hello! I've been trying to install the realsense-ros package from the command line using  the 

`sudo apt-get install ros-$ROS_VER-realsense2-camera` command 

(and yes, I did an `export ROS_VER=noetic` beforehand).

However, I'm getting this error:
`Unable to locate package ros-noetic-realsense2-camera`

Any ideas?

MartyG-RealSense, 2021-01-21:
Hi @Plopao   As a starting point in investigating your case, may I confirm that you have installed ROS Noetic on Ubuntu **20.04** before attempting to install the RealSense ROS wrapper with **Method 1**?  Thanks!

Plopao, 2021-01-21:
Yes I did! I'm about to try installing it with the other methods, but I'd prefer method 1 by far since it's a lot easier to install.

MartyG-RealSense, 2021-01-21:
I recalled this recent case about Noetic package availability:

https://github.com/IntelRealSense/realsense-ros/issues/1577

closed 2021-01-21 by Plopao