realsenseai/realsense-ros #1638: The requested device with serial number xxx and device name containing xx is NOT found. Will Try again.

joguiguma, 2021-01-22:
I try to run https://github.com/IntelRealSense/realsense-ros/tree/occupancy-mapping/occupancy  using my laptop, ubuntu 18 and ros melodic. I have T265 and  D435i

Running:
source devel/setup.bash
roslaunch occupancy occupancy_live_rviz.launch

There are two errors:

 21/01 21:46:54,032 ERROR [140533704140544] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 21/01 21:46:54,032 ERROR [140533704140544] (types.h:313) Unable to open device interface
 21/01 21:46:54,032 WARNING [140533704140544] (rs.cpp:306) null pointer passed for argument "device"
[ WARN] [1611283614.032974241]: Device 2/2 failed with exception: Unable to open device interface
[ERROR] [1611283614.033007845]: The requested device with serial number T265_SN and device name containing t265 is NOT found. Will Try again.
[ INFO] [1611283614.035876474]: Device with serial number 925122110642 was found.

[ INFO] [1611283614.035915640]: Device with physical ID 2-3-3 was found.
[ INFO] [1611283614.035932578]: Device with name Intel RealSense T265 was found.
[ INFO] [1611283614.036276561]: Device with port number 2-3 was found.
[ERROR] [1611283614.132034334]: The requested device with serial number D400_SN and device name containing d4.5 is NOT found. Will Try again.

When I run realsense.viewer both realsense are with good connection. 

What can I do?

joguiguma, 2021-01-22:
Ful terminal is:

roslaunch occupancy occupancy_live_rviz.launch
... logging to /home/joguiguma/.ros/log/c665ed5e-5cd4-11eb-b4ce-38baf8333bf5/roslaunch-joguiguma-Latitude-7490-4072.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.106:39271/

SUMMARY
========

PARAMETERS
 * /DepthOfInterestMax: 2.0
 * /DepthOfInterestMin: 0.2
 * /HeightOfInterestMax: 0.05
 * /HeightOfInterestMin: -0.05
 * /d400/realsense2_camera/accel_fps: 0
 * /d400/realsense2_camera/accel_frame_id: d400_accel_frame
 * /d400/realsense2_camera/accel_optical_frame_id: d400_accel_optica...
 * /d400/realsense2_camera/align_depth: True
 * /d400/realsense2_camera/aligned_depth_to_color_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_fisheye_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra1_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/aligned_depth_to_infra2_frame_id: d400_aligned_dept...
 * /d400/realsense2_camera/allow_no_texture_points: False
 * /d400/realsense2_camera/base_frame_id: d400_link
 * /d400/realsense2_camera/calib_odom_file: 
 * /d400/realsense2_camera/clip_distance: -2.0
 * /d400/realsense2_camera/color_fps: 30
 * /d400/realsense2_camera/color_frame_id: d400_color_frame
 * /d400/realsense2_camera/color_height: 480
 * /d400/realsense2_camera/color_optical_frame_id: d400_color_optica...
 * /d400/realsense2_camera/color_width: 640
 * /d400/realsense2_camera/confidence_fps: 30
 * /d400/realsense2_camera/confidence_height: 480
 * /d400/realsense2_camera/confidence_width: 640
 * /d400/realsense2_camera/depth_fps: 30
 * /d400/realsense2_camera/depth_frame_id: d400_depth_frame
 * /d400/realsense2_camera/depth_height: 480
 * /d400/realsense2_camera/depth_optical_frame_id: d400_depth_optica...
 * /d400/realsense2_camera/depth_width: 640
 * /d400/realsense2_camera/device_type: d4.5
 * /d400/realsense2_camera/enable_accel: False
 * /d400/realsense2_camera/enable_color: True
 * /d400/realsense2_camera/enable_confidence: True
 * /d400/realsense2_camera/enable_depth: True
 * /d400/realsense2_camera/enable_fisheye1: False
 * /d400/realsense2_camera/enable_fisheye2: False
 * /d400/realsense2_camera/enable_fisheye: False
 * /d400/realsense2_camera/enable_gyro: False
 * /d400/realsense2_camera/enable_infra1: False
 * /d400/realsense2_camera/enable_infra2: False
 * /d400/realsense2_camera/enable_infra: False
 * /d400/realsense2_camera/enable_pointcloud: False
 * /d400/realsense2_camera/enable_pose: False
 * /d400/realsense2_camera/enable_sync: False
 * /d400/realsense2_camera/filters: pointcloud
 * /d400/realsense2_camera/fisheye1_frame_id: d400_fisheye1_frame
 * /d400/realsense2_camera/fisheye1_optical_frame_id: d400_fisheye1_opt...
 * /d400/realsense2_camera/fisheye2_frame_id: d400_fisheye2_frame
 * /d400/realsense2_camera/fisheye2_optical_frame_id: d400_fisheye2_opt...
 * /d400/realsense2_camera/fisheye_fps: 30
 * /d400/realsense2_camera/fisheye_frame_id: d400_fisheye_frame
 * /d400/realsense2_camera/fisheye_height: 0
 * /d400/realsense2_camera/fisheye_optical_frame_id: d400_fisheye_opti...
 * /d400/realsense2_camera/fisheye_width: 0
[... 165 lines cut ...]
[ WARN] [1611335453.731839670]: ***************************************************
[ INFO] [1611335453.740823883]:  
[ INFO] [1611335453.745188802]: Device with serial number 912112073155 was found.

[ INFO] [1611335453.745218485]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb2/2-4/2-4:1.0/video4linux/video0 was found.
[ INFO] [1611335453.745232369]: Device with name Intel RealSense D435I was found.
[ INFO] [1611335453.745536033]: Device with port number 2-4 was found.
[ INFO] [1611335453.752423204]: Device with serial number 925122110642 was found.

[ INFO] [1611335453.752485524]: Device with physical ID 2-3-3 was found.
[ INFO] [1611335453.752511676]: Device with name Intel RealSense T265 was found.
[ INFO] [1611335453.752874208]: Device with port number 2-3 was found.
[ERROR] [1611335453.850843871]: The requested device with serial number T265_SN and device name containing t265 is NOT found. Will Try again.
^C[rviz-8] killing on exit
[occupancy_node-7] killing on exit
[t265_to_d400-6] killing on exit
[d400/realsense2_camera-5] killing on exit
[d400/realsense2_camera_manager-4] killing on exit
[t265/realsense2_camera-3] killing on exit
[t265/realsense2_camera_manager-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

doronhi, 2021-01-24:
You should set the parameters "serial_no_camera1" and "serial_no_camera2". They are still in their "example" form:
"T265_SN" and "D400_SN".

Try:
```
roslaunch occupancy occupancy_live_rviz.launch serial_no_camera1:=<Your T265 serial number> serial_no_camera2:=<Your D435 serial number>
```
Or modify the "occupancy/launch/cameras.launch" file accordingly.

closed 2021-01-25 by joguiguma

joguiguma, 2021-01-25:
> You should set the parameters "serial_no_camera1" and "serial_no_camera2". They are still in their "example" form:
> "T265_SN" and "D400_SN".
>
> Try:
>
> ```
> roslaunch occupancy occupancy_live_rviz.launch [...]

The problem was fixed modifying  "occupancy/launch/cameras.launch" introducing  the serial of both cameras. Thank you very much.