realsenseai/realsense-ros #1639: Depth image and color image do not have the same size

Schloern93, 2021-01-22:
Hallo, 

I make an object detection with the color image of a D435 camera and let output the found objects in boxes which are indicated as pixels.
Now I want to find the same pixels in the Point Cloud to locate the objects.
The problem is now the following. The camera lens of the depth image has a higher visual range than the color image (as you can see in the picture below).
Is there a parameter or is it possible to convert this ratio so that I can find the same pixels that I found in the color image in the depth image?

[image]
Thanks for the help!

MartyG-RealSense, 2021-01-22:
Hi @Schloern93   The D435 has a larger field of view (FOV) size on its depth sensor than the color sensor.  This means that when depth is aligned to color, the depth is mapped to the smaller color FOV size and so the outer regions of depth detail may get excluded from the image.

The D415 and D455 camera models have the same FOV size on both the color and depth sensors.

You could instead try aligning **color to depth** instead of depth to color.  The SDK C++ sample program **rs-measure** provides an example of aligning color to depth and then **deprojecting** 2D pixels to a 3D image.

https://github.com/IntelRealSense/librealsense/tree/master/examples/measure

Schloern93, 2021-01-23:
Hallo @MartyG-RealSense,

Thank you very much for the answer.
I was able to set the parameter 
`{'name': 'align_depth1', 'default': 'true', 'description': ''}`
in my launch file and got the correct depth image.
Everything works perfectly now.

closed 2021-01-23 by Schloern93

MartyG-RealSense, 2021-01-23:
Thanks very much @Schloern93 - I'm pleased to hear that you found a solution.   :)

dpbnasika, 2023-07-28:
> Hi @Schloern93 The D435 has a larger field of view (FOV) size on its depth sensor than the color sensor. This means that when depth is aligned to color, the depth is mapped to the smaller color FOV size and so the outer regions of depth detail may get excluded from the image.
> 
> The D415 and D455 camera models have the same FOV size on both the color and depth sensors.
> 
> You could instead try aligning **color to depth** instead of depth to color. The SDK C++ sample program **rs-measure** provides an example of aligning color to depth and then **deprojecting** 2D pixels to a 3D image.
> 
> https://github.com/IntelRealSense/librealsense/tree/master/examples/measure
> 
> If you prefer to use Python, the RealSense user **soarwing52** created a Python conversion of rs-measure:
> 
> https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py

Hi MartyG,

how do you set the color_to_depth option in ros launch file or through command line?, when i set align_depth to true, it shows me only the /camera/aligned_depth_to_color/image_raw topic, but what you mean to say looks different and also when i click on the 2d image pixel it always gives me a point that is some what offset with some pixels, i dont know if the color image and depth image give the same pixel ratio, i try to run it like this, could you able to show me how to set it the way you are telling?.

ros2 launch realsense2_camera rs_launch.py pointcloud.enable:=true rgb_camera.profile:=848,480,30 device_type:=d435i align_depth.enable:=true

Thank you

Best Regards

MartyG-RealSense, 2023-07-28:
Hi @dpbnasika  Whilst the librealsense SDK can align color to depth, the RealSense ROS wrapper does not support this feature and only aligns depth to color.

The effect of color to depth alignment is that the color image is resized to fit the depth image's size.  So on camera models such as D435 and D435i which have a color sensor with a smaller field of view than the depth sensor, the color image will be stretched to match the size of the larger depth image.  

With depth to color alignment (which the ROS wrapper provides), when the color sensor is smaller then the outer edges of the aligned image that were visible on the depth image will be cut off from the combined depth-color image.

dpbnasika, 2023-07-28:
> Hi @dpbnasika Whilst the librealsense SDK can align color to depth, the RealSense ROS wrapper does not support this feature and only aligns depth to color.
>
> The effect of color to depth alignment is [...]

So this gives your the offset of the 2d color image to 3d coordinate ?. Because when I click on the pixels I don’t really get the robot to go there in real world 3d coordinate but always there lies a offset and it goes few cm away from the target object. How do you think we can solve this with d435i camera or any other camera? Is there a solution for this readily available ?. And do you think having a d415 or d455 can give you good estimates of 2d point on the color image to a real 3d world coordinate ?.

MartyG-RealSense, 2023-07-28:
If an object is more than 3 meters away from the camera then depth measurement accuracy will start to noticably reduce beyond that distance with the D435 / D435i models.  

A D415 has around 2x the accuracy of D435 and 2x the image quality (less noise).  Accuracy will still start to decline beyond 3 meters though.

The D455 has 2x the accuracy of D435, so at 6 meters the D455 has the same accuracy that D435 has at 3 meters, making it ideal for long-range depth sensing.  D455 also has both the depth and RGB sensors mounted on the same 'stiffener' inside the camera, providing improved depth-color alignment.

dpbnasika, 2023-07-28:
> Depth measurement error starts at around 0 at the camera lenses and increases linearly as an observed object becomes further away from the camera. This phenomenon is called **RMS Error**.
> 
> If an object is more than 3 meters away from the camera then depth measurement accuracy will start to noticably reduce beyond that distance.
> 
> A D415 has around 2x the accuracy over distance of D435 and 2x the image quality (less noise). Accuracy will still start to decline beyond 3 meters with the D415 though.
> 
> The D455 has 2x the accuracy of D435, so at 6 meters the D455 has the same accuracy that D435 has at 3 meters, making it ideal for long-range depth sensing. D455 also has both the depth and RGB sensors mounted on the same 'stiffener' inside the camera, providing improved depth-color alignment.
> 
> In some situations, using one of the 'f' models (D435f, D435if or D455f) will provide improved accuracy because of filters equipped on the lenses (hence the 'f' in the model name).
> 
> https://www.intelrealsense.com/depth-camera-d455f/

I got the point, but for me the range is not beyond 3cm , I try to click on a pixel in the 2d image of a real object , and I pass that information to the deprojection from 2d pixel to 3d coordinate function and get the point, but the point that I get in 3d is not exactly the point of the target object but it is few cm away from the target object. Why is this so happening ?. I really want the 3d coordinates of the 2d pixel of the object I click on. With d435i camera it does not work..!! And I could not reach or get the real 3d coordinate of the object. and the resolution of my depth and color image I pass is 848x480 at 30fps.

MartyG-RealSense, 2023-07-28:
It may be worth trying the ROS2 wrapper's **show_center_depth.py** Python node script, which will provide the 3D distance measurement of the object at the center of the camera's view without having to click on the image.

https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera/scripts/show_center_depth.py

MartyG-RealSense, 2023-07-28:
The color and texture of the observed object will affect how accurately the camera can read depth information from it.  

For example, if the surface is colored dark gray or black then the camera will have greater difficulty depth-sensing it.  This is because it is a general physics principle (not specific to RealSense) that dark grey or black absorbs light and so makes it more difficult for depth cameras to read depth information from such surfaces.  The darker the color shade, the more light that is absorbed and so the less depth detail that the camera can obtain.

Surfaces that are reflective or are smooth and plain (with low texture detail or no texture) will also be more difficult to depth-sense accurately.  

If the instruction **get_distance()** is used in a script to obtain the distance in meters of a particular point after depth and color have been aligned then the results will be less accurate if depth intrinsics are used instead of color intrinsics.  When get_distance() is inaccurate, measurements will be accurate at the center of the image and become increasingly incorrect as you move outwards from the center towards the outer edges of the image.

prarthanasigedar, 2025-03-10:
@MartyG-RealSense Hello, I was going through this thread and I am facing a similar issue of getting the 3d coordinates from the camera. The calculated 3d position is off. I am using d435i.
 My object is within 1 meter of the camera and i am using ros2 real-sense wrapper. So, the depth frames have a different size as compared  to color frames. The resolution of depth frame is smaller relative to the colour frames:-

1) So, is there any way to match the resolution of depth in terms of colour or vice versa?
2) Also, in the above threads, it was written that depth sensor has a higher FOV. However, in the below screenshot I can see that the colour frame has higher FOV relative to the depth? Am i correct or am i missing something ?

[image]

MartyG-RealSense, 2025-03-10:
Hi @prarthanasigedar  In your kindly provided image, color has a higher **resolution** than depth.  The resolution size is not the field of view (FOV) size.  The FOV determines how much of the real-world scene the camera can see without being moved.  Detail in the scene that is outside of the field of view is not included on the image.

In the RealSense ROS2 wrapper you can define a custom resolution for the depth and color so that the resolutions match by configuring a **profile**, like in the launch instruction below.

`ros2 launch realsense2_camera rs_launch.py depth_module.depth_profile:=1280x720x30 rgb_camera.color_profile:=1280x720x30`