realsenseai/realsense-ros #1640: parameters in config file ignored

AndreasAZiegler, 2021-01-22:
I'm running an Ubuntu 18.04 with ROS2 eloquent installed and use the eloquent branch of this ros driver.

When I want start the node with (how I used to do it with the previous unofficial ROS2 driver)
```
ros2 run realsense2_camera realsense2_camera_node --params-file `ros2 pkg prefix safety-field-intrusion-detector-ros`/share/safety-field-intrusion-detector-ros/config/d435i_new.yaml -r __ns:=/realsense1
```
with ```cat `ros2 pkg prefix safety-field-intrusion-detector-ros`/share/safety-field-intrusion-detector-ros/config/d435i_new.yaml```
```
realsense1:
  camera:
    ros__parameters:
        serial_no: '_841512070768' # d435i
        initial_reset: true
        base_frame_id: d435i_link

        align_depth: false
        enable_pointcloud: false
        enable_fisheye: false
        enable_fisheye1: false
        enable_fisheye2: false
        enable_pose: false

        enable_color: true
        color_fps: 30.0
        color_height: 480
        color_width: 640
            
        enable_depth: true
        depth_fps: 30.0
        depth_height: 720
        depth_width: 1280

        enable_infra: false
        enable_infra1: false
        enable_infra2: false

        enable_gyro: false

        enable_accel: false
```

The node starts without considering the ```serial_no``` in the config file.

When I start the node with
```
ros2 run realsense2_camera realsense2_camera_node --params-file `ros2 pkg prefix safety-field-intrusion-detector-ros`/share/safety-field-intrusion-detector-ros/config/d435i_new.yaml -r __ns:=/realsense1 --ros-args -p serial_no:=_841512070768
```
the serial given in the terminal is taken. How can I use the parameters in a yaml-config file?

MartyG-RealSense, 2021-01-23:
Hi @AndreasAZiegler  I must emphasize that I am not a ROS2 specialist and so my knowledge of ROS2 launchi configuration is limited.  I will do my best to be of help though.

Looking at the list of settings that you want to apply, it may be unnecessary to have that full list as the majority of the settings in your list should already be set to those desired statuses by default (at least, in the ROS1 wrapper).

The main settings that stand out to me as possibly needing custom configuration due to differing from the defaults are:

-  Depth width / height
-  initial_reset
-  base_frame_id
-  serial number

These few remaining settings could conceivably be placed in a **ros2 run** statement so that a yaml configuration file was not required.

doronhi, 2021-01-24:
Are the other parameters taken into account? initial_reset for example?
I tested your file on my eloquent machine and it works for me. Maybe the path resulting from 
`` `ros2 pkg prefix safety-field-intrusion-detector-ros`/share/safety-field-intrusion-detector-ros/config/d435i_new.yaml ``
Does not yield the intended file?

AndreasAZiegler, 2021-01-25:
@MartyG-RealSense I now start the node with the launch file
```
ros2 launch realsense2_camera rs_launch.py camera_name:=realsense1 config_file:=/home/mtr/ros/safety_field_detector_ws/install/safety-field-intrusion-detector-ros/share/safety-field-intrusion-detector-ros/config/d435i_new.yaml serial_no:=_841512070768
```
However, I still have to specify the ```serial_no``` as otherwise it takes the first camera (of the two connected) and uses it.

@doronhi The path should be fine. I tried it with an absolute path and also copied the yaml file to my home directory and used the absolute path (```/home/mtr/d435i_new.yaml```) all without success.

Are there any other ways I can further trouble shoot. I have another node which does read it's config file and also the previous ROS2 RealSense driver does read it's config file. Therefore, I don't think it is something general ROS2 eloquent related.

doronhi, 2021-01-26:
Hi @AndreasAZiegler, I think you are in a better position to investigate than I am since I can't reproduce. What are the differences between the 2 nodes you have - the one that reads the config file and the other that does not?
Do you mean that the node that does not read the serial_no reads other values from that config file or none at all?

MartyG-RealSense, 2021-02-02:
Hi @AndreasAZiegler  Do you require further assistance with this case, please?   Thanks!

AndreasAZiegler, 2021-02-02:
I didn't have time to work on it the last couple of days. But I plan to work on it again this week. I will let you know, if I need further assistance.

MartyG-RealSense, 2021-02-02:
Okay, thanks very much for the update.  I will keep this case open for a further time period.  Good luck!

AndreasAZiegler, 2021-02-15:
I don't know what I did wrong. I is working now.

closed 2021-02-15 by AndreasAZiegler

MartyG-RealSense, 2021-02-15:
Great news @AndreasAZiegler  - thanks for the update!