realsenseai/realsense-ros #1641: How to use realsense D435 to get the topic:/camera/color/Image_raw/compressed?

GitZzw, 2021-01-23:
I only got these topics:
/camera/color/camera_info
/camera/color/image_raw
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/extrinsics/depth_to_color
/camera/motion_module/parameter_descriptions
/camera/motion_module/parameter_updates
/camera/realsense2_camera_manager/bond
/camera/rgb_camera/auto_exposure_roi/parameter_descriptions
/camera/rgb_camera/auto_exposure_roi/parameter_updates
/camera/rgb_camera/parameter_descriptions
/camera/rgb_camera/parameter_updates
/camera/stereo_module/auto_exposure_roi/parameter_descriptions
/camera/stereo_module/auto_exposure_roi/parameter_updates
/camera/stereo_module/parameter_descriptions
/camera/stereo_module/parameter_updates
/diagnostics
/rosout
/rosout_agg
/tf
/tf_static

MartyG-RealSense, 2021-01-23:
Hi @GitZzw  Are you using the **rs_camera** roslaunch instruction please?

**roslaunch realsense2_camera rs_camera.launch**

GitZzw, 2021-01-24:
I tryed:
roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true enable_infra1:=true enable_infra2:=true
and I got there errors:
[screen.log](https://github.com/IntelRealSense/realsense-ros/files/5861740/screen.log)

MartyG-RealSense, 2021-01-24:
In your log you have multiple **control_transfer returned error, index: 768, error: No data available, number: 61** warnings after launch.  If this error is continuously generating then it can indicate a serious communication problem with the device.

The first test step that I would recommend is to add an instruction to the end of your roslaunch statement to reset the camera at launch and see whether that improves performance:

**initial_reset:=true**

GitZzw, 2021-01-24:
Thank you for your patience, so D435 camera do not contain an IMU component?
And I have tried all the advised command in https://github.com/IntelRealSense/realsense-ros/issues/1271, but still have multiple control_transfer returned error.

MartyG-RealSense, 2021-01-24:
That is correct, the D435 model does not have an IMU component inside it.  The D435i model does.

Please add the commands below to your roslaunch instruction to set the depth and color FPS speed to '15' to test whether using 15 FPS instead of 30 reduces the control_transfer returned warnings:

**depth_fps:=15 color_fps:=15**

closed 2021-01-25 by GitZzw