realsenseai/realsense-ros #1642: "No RealSense devices were found!" with T265 plugged in

rk122019, 2021-01-23:
After working through issues I posted in #1631, I'm now running the ROS wrapper (2.2.20) on Windows 10 using Librealsense v2.40.0 and ROS Melodic. I am able to successfully connect a D435i to my computer and run `roslaunch realsense2_camera rs_camera.launch` without any problems.

However, now when I try to plug in a T265 (on its own) and run `roslaunch realsense2_camera rs_t265.launch`, I get the same error as from #1631: "No RealSense devices were found!". The camera is recognized by Windows 10 in my Device Manager, and I can also run the RealSense Viewer successfully with the T265. 

I'm curious if anyone has any suggestions on how to resolve this problem and get the ROS wrapper to recognize my T265 camera. Thanks!

mgopshtein, 2021-05-03:
Have the same problem with ROS noetic, any solutions yet?

rk122019, 2021-05-04:
I unfortunately did not find a solution for the T265 on Windows. I resorted to creating an Ubuntu dual boot on my computer and doing all of my T265 ROS work on Ubuntu 18.04.