realsenseai/realsense-ros #1645: [ERROR] [1611477295.803276284]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.

qinhongtju, 2021-01-24:
[ERROR] [1611477295.803276284]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1611477295.803533615]: The error before refreshing the cache was: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1611477295.804208035]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera-3] process has died [pid 3704, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/pi/.ros/log/074822a6-5e1f-11eb-b2c9-dca6329ad46e/camera-realsense2_camera-3.log].
log file: /home/pi/.ros/log/074822a6-5e1f-11eb-b2c9-dca6329ad46e/camera-realsense2_camera-3*.log

At present, SDK2.0 is easy to use, but Realsense
After the ros installation is complete, execute: roslaunch realsense2_camera rs_camera.launch, there is this problem

MartyG-RealSense, 2021-01-24:
Hi @qinhongtju   Is the problem resolved if you do a clean re-installation of the ROS wrapper, please?

https://github.com/IntelRealSense/realsense-ros/issues/745#issuecomment-500394506

qinhongtju, 2021-01-26:
There is no problem with my ros wrapper. I have reinstalled it, but the problem still exists.

qinhongtju, 2021-01-26:
The motherboard model I use is Raspberry Pi 4B

MartyG-RealSense, 2021-01-26:
Are you installing on **Raspbian** OS on the Pi 4?  If so, I would recommend following the **eleccelerator** guide:

https://eleccelerator.com/wiki/index.php?title=Raspbian_Buster_ROS_RealSense

qinhongtju, 2021-02-01:
> Are you installing on **Raspbian** OS on the Pi 4? If so, I would recommend following the **eleccelerator** guide for installing ROS and the RealSense ROS wrapper on Pi 4 and Raspbian Buster:
> 
> https://eleccelerator.com/wiki/index.php?title=Raspbian_Buster_ROS_RealSense

I have adjusted according to the link you provided, but the problem still exists. The camera driver SDK 2.0 can be used normally, but after installing the ros package, it cannot be executed normally. Do you know the general reason?
The model of my device is:  Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux

MartyG-RealSense, 2021-02-01:
Others have encountered the error that was listed in your opening comment.

https://github.com/IntelRealSense/realsense-ros/issues/464

The website **Titanwolf** covers this error regarding installing RealSense ROS on Pi 3b.  In the section of their guide titled **Error measures** they suggest using the command below to resolve it.

**git clone https://github.com/icarpis/realsense.git**

https://titanwolf.org/Network/Articles/Article?AID=22d3e462-bffa-4972-87e7-e2d94dd52081#gsc.tab=0

qinhongtju, 2021-02-03:
> Others have encountered the error that was listed in your opening comment.
>
> #464
>
> The website **Titanwolf** covers this error regarding installing RealSense ROS on Pi 3b. In the section of their [...]

I refer to this link, but an execution error occurred 
- Could NOT find pcl_ros (missing: pcl_ros_DIR)
-- Could not find the required component 'pcl_ros'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /home/pi/catkin_ws/devel/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "pcl_ros" with any
  of the following names:

    pcl_rosConfig.cmake
    pcl_ros-config.cmake

  Add the installation prefix of "pcl_ros" to CMAKE_PREFIX_PATH or set
  "pcl_ros_DIR" to a directory containing one of the above files.  If
  "pcl_ros" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  realsense/realsense_camera/CMakeLists.txt:39 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/pi/real_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/real_ws/build/CMakeFiles/CMakeError.log".
make: *** [Makefile:1456：cmake_check_build_system] 错误 1
Invoking "make cmake_check_build_system" failed
 
“git clone https://github.com/icarpis/realsense.git”  Which folder should this be cloned in？
Thank you very much for your reply ！

MartyG-RealSense, 2021-02-03:
If you receive these error messages:

**Could not find the required component 'pcl_ros'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
Could not find a package configuration file provided by "pcl_ros" with any of the following names:

pcl_rosConfig.cmake
pcl_ros-config.cmake**

Then the eleccelerator guide suggests a procedure for fixing it.

https://eleccelerator.com/wiki/index.php?title=Raspbian_Buster_ROS_RealSense#Example_Problem

qinhongtju, 2021-02-04:
> If you receive these error messages:
>
> **Could not find the required component 'pcl_ros'. The following CMake error indicates that you either need to install the package with the same name or change [...]

I tried the method you provided, but there are still compilation errors. 
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   chrono
--   date_time
--   atomic
CMake Error at perception_pcl/pcl_ros/CMakeLists.txt:8 (find_package):
  By not providing "FindPCL.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "PCL", but
  CMake did not find one.

  Could not find a package configuration file provided by "PCL" with any of
  the following names:

    PCLConfig.cmake
    pcl-config.cmake

  Add the installation prefix of "PCL" to CMAKE_PREFIX_PATH or set "PCL_DIR"
  to a directory containing one of the above files.  If "PCL" provides a
  separate development package or SDK, be sure it has been installed.

-- Configuring incomplete, errors occurred!
See also "/home/pi/real_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/real_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

MartyG-RealSense, 2021-02-04:
When the Boost libraries are being checked, I noticed that the final library checked before the error occurs is **atomic**.  This makes me wonder whether you are experiencing an old problem with librealsense Pi builds that ws fixed by adding a reference to **-latomic** to the **CMakeLists.txt** file:

https://github.com/IntelRealSense/librealsense/issues/4272#issuecomment-573445307

closed 2021-02-16 by MartyG-RealSense