realsenseai/realsense-ros #1648: GetIMU Info in realsense-ros

peter5232, 2021-01-26:
I want to use the IMU of RealSenseD455 in ros , but I can't find which can open the IMU so that I can get the IMU info .So can someone tell me what I can do ? And in "IMUInfo.msg" ,waht do all parameters mean?

cco-seasony, 2021-01-26:
For the ros2 foxy distribution you would need to enable the gyro and accelerometer. One way to do so is by setting these parameters on the sample launch file provided.
Something like this:

`ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true`

This should make the camera node start publishing messages about the gyro and the accelerometer. Notice that I tested it with ROS2 Foxy and the Realsense D435i.

I would suggest you mention which distro of ROS you are using.

MartyG-RealSense, 2021-01-26:
Hi @peter5232  As kindly mentioned above by @cco-seasony you can enable the IMU topics by adding the instructions below to the end of the roslaunch instruction:

**enable_gyro:=true enable_accel:=true**

In regard to explanations of the data, noise_variance and bias_variance IMU values, I recommend the documentation link below that provides definitions for these motion intrinsics.

https://github.com/IntelRealSense/librealsense/blob/master/src/media/readme.md#motion-intrinsic

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a significant time period.

closed 2026-10-23 by MartyG-RealSense