realsenseai/realsense-ros #1649: ROS Wrapper Not Showing Topics

dlee640, 2021-01-26:
Hello. I have been working to enable ros-wrapper for **Intel realsense D435 device** on **Ubuntu 18.04**. **I have issues with rostopic and rosnode not showing up, despite the fact that build was successful and the fact that I was able to run roslaunch realsense2_camera rs_camera.launch without errors.**

Here is the overview of the problem:

Upon typing: $ roslaunch realsense2_camera rs_camera.launch, I get:

unitree@nx:~$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/unitree/.ros/log/5aa94cb0-6011-11eb-ac1b-48b02d0790cc/roslaunch-nx-13423.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nx:44581/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
[... 79 lines cut ...]
[ INFO] [1611692157.603136390]: Expected frequency for depth = 30.00000
[ INFO] [1611692157.611618633]: Expected frequency for color = 30.00000
[ INFO] [1611692157.617215754]: setupStreams...
[ INFO] [1611692157.618916722]: insert Depth to Stereo Module
[ INFO] [1611692157.619074645]: insert Color to RGB Camera
[ INFO] [1611692157.685289770]: SELECTED BASE:Depth, 0
[ INFO] [1611692157.711732555]: RealSense Node Is Up!
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/unitree/.ros/log/5aa94cb0-6011-11eb-ac1b-48b02d0790cc/camera-realsense2_camera-2*.log

**Weird thing is, I do not get full list of nodes and topics that I am supposed to be getting.**

**unitree@nx:~$ rosnode list
/camera/realsense2_camera_manager
/rosout
 
unitree@nx:~$ rostopic list
/rosout
/rosout_agg**

Sometimes, I do get
$ /camera/realsense2_camera
and the full list of topics described on readme.md doc, but they are very temporary (only available for like 3 seconds, then they disappear)

**What could be wrong? My realsense_viewer works fine, so it doesn't seem to be an issue related to camera and PC connection.**

doronhi, 2021-01-26:
The following line in the log: "[camera/realsense2_camera-2] process has finished cleanly" says that the node is dead, therefore, no node in the list and no topics.
As for why did the node crash there are no helping messages I can see, except that it is running with LibRealSense v2.41.0 while it was built with LibRealSense v2.40.0.
Please rebuild realsense2_camera to get that fixed.

MartyG-RealSense, 2021-01-26:
Hi @dlee640   I second the advice of @doronhi - if librealsense is updated after the RealSense ROS wrapper has been built then the ROS wrapper should be built again after the new librealsense version has been installed.

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a significant time period.

closed 2026-10-23 by MartyG-RealSense