realsenseai/realsense-ros #1651: D435 IR source emitter setting

BIGFatming, 2021-01-27:
When we check infra images captured by D435 camera, we found out there are many light spots caused by laser emitter. Shown in following picture.
[image]
I run rosrun rqt_reconfigure rqt_reconfigure and checked **stereo_module**, then I saw this tip
[image]
I tried to add 
`<rosparam>`
`  </camera/stereo_module/emitter_enabled: 3>`
`</rosparam>`
in launch file, but later I realized param emitter_enabled is a bool value.
So:
1. Dose the "LED" shown in the tip mean infra-red LED? whether D435 support this option? If it supports, how to enabled it.
2. Since the accuracy of depth image depends on the laser. If I use LED instead of laser, will the depth image be affected?

MartyG-RealSense, 2021-01-27:
Hi @BIGFatming   The LED setting (emitter option 3) is related to the IR Emitter but never saw any use as far as I know after the launch of the librealsense SDK.  There is no detailed definition for what the option does.  I recommend that you avoid using it, and also avoid emitter option '2' (Auto Laser).

Whilst the camera can use the semi-random dot pattern emitted by the camera projector to analyze surfaces for dept detail (by using the dots cast onto surfaces as a texture source), the camera can also use **ambient light** to read depth detail from a scene if that scene is well lit.  The darkness of your IR image suggests that the scene is not strongly lit though, so this may not be a viable alternative for your project.

If the Laser Power is reduced then the depth image does become more sparse in detail. On the infrared image, it results in the IR image becoming darker.  This can be compansated for on the IR image by increasing the value of the **Gain** setting of the stereo module to brighten it.  Increasing Gain will not restore lost depth detail when Laser Power is reduced or the emitter is turned off if you are also using a depth image as well as an infrared one though.  Whilst increasing the Gain brightens the IR image, it can make the depth image progressively worse as Gain is increased further.

If the IR emitter is causing excessive brightness near to the camera lens and you also wish to dispense with the dots, you could consider using a patternless external **IR Illuminator** lamp as the IR light source for the camera instead.

https://github.com/IntelRealSense/librealsense/issues/2000

There is also the option of using an external LED pattern projectot instead of the built-in one in the camera.  External pattern projectos can be freely moved around and shaken without significantly affectinf the camera.  Intel's white-paper document about projectors has further detals about this,

https://dev.intelrealsense.com/docs/projectors

If changing the camera model is an option, the D415 supports a camera preset called **Left Imager Color w/o IR Pattern** for removing the dot pattern from the IR image.

https://dev.intelrealsense.com/docs/d400-series-visual-presets#section-preset-table

BIGFatming, 2021-01-28:
Thanks for your detailed answer. We will discuss it.

MartyG-RealSense, 2021-02-07:
Hi @BIGFatming  Do you have an update about your case that you can provide, please?   Thanks!

closed 2021-02-13 by MartyG-RealSense