realsenseai/realsense-ros #1652: Running poincloud stream and color stream on different resolutions

apoorvcn47, 2021-01-27:
I am using L515 on Foxy. Which supports pointcloud stream at 640X480 and color stream at 1920X1080. I want pointcloud with color texture. When I set color_width and color_height at 640 and 480, it gives me following warning and color stream never turns on:

[WARN] [1611786931.026906040] [RealSenseCameraNode]: Given stream configuration is not supported by the device!  Stream: Color, Stream Index: 0, Width: 640, Height: 480, FPS: 30, Format: None      

When I set color_width and color_height at 1920 and 1080, it gives me following warning and pointcloud stream never turns on:

[WARN] [1611787046.127299813] [RealSenseCameraNode]: Given stream configuration is not supported by the device!  Stream: Depth, Stream Index: 0, Width: 1920, Height: 1080, FPS: 30, Format: Z16    

Is there anything I am missing? Following are my parameters:

[{'name': 'camera_name',                  'default': 'camera', 'description': 'camera unique name'},
                           {'name': 'serial_no',                    'default': '', 'description': 'choose device by serial number'},
                           {'name': 'usb_port_id',                  'default': '', 'description': 'choose device by usb port id'},
                           {'name': 'device_type',                  'default': '', 'description': 'choose device by type'},
                           {'name': 'config_file',                  'default': '', 'description': 'yaml config file'},
                           {'name': 'enable_pointcloud',            'default': 'true', 'description': 'enable pointcloud'},
                           {'name': 'unite_imu_method',             'default': '', 'description': '[copy|linear_interpolation]'},                           
                           {'name': 'json_file_path',               'default': '', 'description': 'allows advanced configuration'},                           
                           {'name': 'output',                       'default': 'screen', 'description': 'pipe node output [screen|log]'},                           
                           {'name': 'depth_width',                  'default': '640', 'description': 'depth image width'},                           
                           {'name': 'depth_height',                 'default': '480', 'description': 'depth image height'},                           
                           {'name': 'enable_depth',                 'default': 'true', 'description': 'enable depth stream'},
                           {'name': 'color_width',                  'default': '1920', 'description': 'color image width'},                           
                           {'name': 'color_height',                 'default': '1080', 'description': 'color image height'},                           
                        #    {'name': 'color_width',                  'default': '640', 'description': 'color image width'},                           
                        #    {'name': 'color_height',                 'default': '480', 'description': 'color image height'},                           
                           {'name': 'enable_color',                 'default': 'true', 'description': 'enable color stream'},
                           {'name': 'infra_width',                  'default': '640', 'description': 'infra width'},
                           {'name': 'infra_height',                 'default': '480', 'description': 'infra width'},
                           {'name': 'enable_infra1',                'default': 'false', 'description': 'enable infra1 stream'},
                           {'name': 'enable_infra2',                'default': 'false', 'description': 'enable infra2 stream'},
                           {'name': 'infra_rgb',                    'default': 'false', 'description': 'enable infra2 stream'},
                           {'name': 'fisheye_width',                'default': '848', 'description': 'fisheye width'},
                           {'name': 'fisheye_height',               'default': '800', 'description': 'fisheye width'},
                           {'name': 'enable_fisheye1',              'default': 'false', 'description': 'enable fisheye1 stream'},
                           {'name': 'enable_fisheye2',              'default': 'false', 'description': 'enable fisheye2 stream'},
                           {'name': 'confidence_width',             'default': '640', 'description': 'depth image width'},                           
                           {'name': 'confidence_height',            'default': '480', 'description': 'depth image height'},                           
                           {'name': 'enable_confidence',            'default': 'true', 'description': 'enable depth stream'},
                           {'name': 'fisheye_fps',                  'default': '30.', 'description': ''},                           
                           {'name': 'depth_fps',                    'default': '30.', 'description': ''},                           
                           {'name': 'confidence_fps',               'default': '30.', 'description': ''},                           
                           {'name': 'infra_fps',                    'default': '30.', 'description': ''},                           
                           {'name': 'color_fps',                    'default': '30.', 'description': ''},                           
                           {'name': 'gyro_fps',                     'default': '400.', 'description': ''},                           
                           {'name': 'accel_fps',                    'default': '250.', 'description': ''},    
                           {'name': 'color_qos',                    'default': 'DEFAULT', 'description': 'QoS profile name'},    
                           {'name': 'confidence_qos',               'default': 'DEFAULT', 'description': 'QoS profile name'},    
                           {'name': 'depth_qos',                    'default': 'DEFAULT', 'description': 'QoS profile name'},    
                           {'name': 'fisheye_qos',                  'default': 'DEFAULT', 'description': 'QoS profile name'},    
                           {'name': 'infra_qos',                    'default': 'DEFAULT', 'description': 'QoS profile name'},    
                        #    {'name': 'color_qos',                    'default': 'SENSOR_DATA', 'description': 'QoS profile name'},    
                        #    {'name': 'confidence_qos',               'default': 'SENSOR_DATA', 'description': 'QoS profile name'},    
                        #    {'name': 'depth_qos',                    'default': 'SENSOR_DATA', 'description': 'QoS profile name'},    
                        #    {'name': 'fisheye_qos',                  'default': 'SENSOR_DATA', 'description': 'QoS profile name'},    
                        #    {'name': 'infra_qos',                    'default': 'SENSOR_DATA', 'description': 'QoS profile name'},    
                           {'name': 'enable_gyro',                  'default': 'false', 'description': ''},                           
                           {'name': 'enable_accel',                 'default': 'false', 'description': ''},                           
                        #    {'name': 'pointcloud_texture_stream',    'default': 'RS2_STREAM_ANY', 'description': 'testure stream for pointcloud'},                           
                           {'name': 'pointcloud_texture_stream',    'default': 'RS2_STREAM_COLOR', 'description': 'testure stream for pointcloud'},                           
                           {'name': 'pointcloud_texture_index',     'default': '0', 'description': 'testure stream index for pointcloud'},                           
                           {'name': 'enable_sync',                  'default': 'true', 'description': ''},                           
                           {'name': 'align_depth',                  'default': 'true', 'description': ''},                           
                           {'name': 'filters',                      'default': 'colorizer', 'description': ''},                           
                           {'name': 'clip_distance',                'default': '-2.', 'description': ''},                           
                           {'name': 'linear_accel_cov',             'default': '0.01', 'description': ''},                           
[... cut ...]
                           {'name': 'calib_odom_file',              'default': '', 'description': ''},                           
                           {'name': 'topic_odom_in',                'default': '', 'description': 'topic for T265 wheel odometry'},
                           {'name': 'tf_publish_rate',              'default': '0.0', 'description': 'Rate of publishing static_tf'},
                           {'name': 'rosbag_filename',              'default': '', 'description': 'A realsense bagfile to run from as a device'},
                          ]

RealSenseSupport, 2021-01-29:
L515 Depth resolutions supported (which applies to confidence stream and infrared):
1024x768 @ 30fps
640x480 @ 30fps
320x240 @ 30fps

L515 RGB resolutions supported:
960x540
1280x720
1920x1080

These can be found in the datasheet:  https://dev.intelrealsense.com/docs/lidar-camera-l515-datasheet

RealSenseSupport, 2021-02-05:
Is there anything else needed on this thread?

*If we don’t hear from you in 7 days, this issue will be closed.

apoorvcn47, 2021-02-06:
Sounds good. Please close this issue

closed 2021-02-25 by RealSenseSupport