realsenseai/realsense-ros #1653: Unable to run rtabmap with RealSense D455

EricWiener, 2021-01-27:
I'm trying to run rtabmap using the RealSense and I am having a lot of difficulties. I believe the issue is that the RealSense is not opening for some reason, but I'm not sure. I would really appreciate any advice.

Here's the launch file I'm using:
```
<launch>
    <!-- 
          Source: https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_rtabmap.launch
          NOTICE: Installation of rtabmap is required for using this launch file:
                  For installation type:
                        apt-get install ros-kinetic-rtabmap-ros
    -->
    <arg name="device_type"    		      default="d4.5"/>	<!-- Note: using regular expression. match D435, D435i, D415... -->
    <arg name="serial_no"    			  default="" />
    <arg name="camera"                    default="D455"/>		<!-- Note: Replace with camera name -->
    <arg name="clip_distance"             default="-2"/>
    <arg name="use_rviz"                  default="true"/>
    <arg name="use_rtabmapviz"            default="false"/>
    
    <include file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="device_type"             value="$(arg device_type)"/>
        <arg name="serial_no"               value="$(arg serial_no)"/>
        <arg name="camera"                  value="$(arg camera)"/>
        <arg name="clip_distance"           value="$(arg clip_distance)"/>
        <arg name="enable_pointcloud"       value="true"/>
	<arg name="initial_reset"	    value="true"/>
    </include>

    <include file="$(find rtabmap_ros)/launch/rtabmap.launch">           
 	 <arg name="rtabmap_args"       value="--delete_db_on_start"/>
            <arg name="depth_topic"        value="/$(arg camera)/aligned_depth_to_color/image_raw"/>
            <arg name="frame_id"           value="$(arg camera)_link"/>
            <arg name="visual_odometry"    value="false"/>
            <arg name="odom_topic"         value="/imu/data"/>
            <arg name="rgb_topic"          value="/$(arg camera)/color/image_raw"/>
            <arg name="camera_info_topic"  value="/$(arg camera)/color/camera_info"/>
            <arg name="queue_size"         value="200"/>
            <arg name="rviz"               value="$(arg use_rviz)"/>
            <arg name="rtabmapviz"         value="$(arg use_rtabmapviz)"/>
    </include>
</launch>
```

Here's the error I get:
```
roslaunch realsense_scan.launch 
... logging to /root/.ros/log/8f3b56aa-60d2-11eb-acab-b30829e21680/roslaunch-cpr-mic09-1361.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cpr-mic09:33177/

SUMMARY
========

PARAMETERS
 * /D455/realsense2_camera/accel_fps: 250
 * /D455/realsense2_camera/accel_frame_id: D455_accel_frame
 * /D455/realsense2_camera/accel_optical_frame_id: D455_accel_optica...
 * /D455/realsense2_camera/align_depth: False
 * /D455/realsense2_camera/aligned_depth_to_color_frame_id: D455_aligned_dept...
 * /D455/realsense2_camera/aligned_depth_to_fisheye1_frame_id: D455_aligned_dept...
 * /D455/realsense2_camera/aligned_depth_to_fisheye2_frame_id: D455_aligned_dept...
 * /D455/realsense2_camera/aligned_depth_to_fisheye_frame_id: D455_aligned_dept...
 * /D455/realsense2_camera/aligned_depth_to_infra1_frame_id: D455_aligned_dept...
 * /D455/realsense2_camera/aligned_depth_to_infra2_frame_id: D455_aligned_dept...
 * /D455/realsense2_camera/allow_no_texture_points: False
 * /D455/realsense2_camera/base_frame_id: D455_link
 * /D455/realsense2_camera/calib_odom_file: 
 * /D455/realsense2_camera/clip_distance: -2.0
[... 159 lines cut ...]
[ INFO] [1611787258.964632275]: /rtabmap/rtabmap: subscribe_scan_descriptor = false
[ INFO] [1611787258.964920757]: /rtabmap/rtabmap: queue_size    = 200
[ INFO] [1611787258.964994825]: /rtabmap/rtabmap: approx_sync   = true
[ INFO] [1611787258.965214622]: Setup depth callback
[ INFO] [1611787258.982740198]: 
/rtabmap/rtabmap subscribed to (approx sync):
   /imu/data \
   /D455/color/image_raw \
   /D455/aligned_depth_to_color/image_raw \
   /D455/color/camera_info
[ INFO] [1611787259.024727549]: Approximate time sync = true
[ INFO] [1611787259.271229965]: Parameters are not saved! (No configuration file provided...)
rtabmap: Saving database/long-term memory... (located at /root/.ros/rtabmap.db)
rtabmap: Saving database/long-term memory...done! (located at /root/.ros/rtabmap.db, 0 MB)
[rtabmap/rtabmap-3] process has died [pid 1382, exit code -11, cmd /opt/ros/melodic/lib/rtabmap_ros/rtabmap --delete_db_on_start rgb/image:=/D455/color/image_raw depth/image:=/D455/aligned_depth_to_color/image_raw rgb/camera_info:=/D455/color/camera_info rgbd_image:=rgbd_image_relay left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud scan_descriptor:=/scan_descriptor user_data:=/user_data user_data_async:=/user_data_async gps/fix:=/gps/fix tag_detections:=/tag_detections odom:=/imu/data imu:=/imu/data __name:=rtabmap __log:=/root/.ros/log/8f3b56aa-60d2-11eb-acab-b30829e21680/rtabmap-rtabmap-3.log].
log file: /root/.ros/log/8f3b56aa-60d2-11eb-acab-b30829e21680/rtabmap-rtabmap-3*.log
^C[points_xyzrgb-5] killing on exit
[D455/realsense2_camera-2] killing on exit
[D455/realsense2_camera_manager-1] killing on exit
[rviz-4] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

taogashi, 2021-01-28:
1. You may not want to set `enable_pointcloud` to true, which could degrade the performance very much.
2. `/imu/data` may not have enough infomation for running rtabmap, which node publish this topic?
3. As to the `process died`, no idea :stuck_out_tongue_closed_eyes:, may be some compilation issue? try add gdb command to `launch-prefix`

EricWiener, 2021-01-28:
@taogashi thanks so much for the help! I will try not enabling pointcloud. `/imu/data` is coming from the Husky robot. I will also try using GDB and update on the progress. Thank you!

MartyG-RealSense, 2021-01-28:
Hi @EricWiener   It is also worth adding that your log states that **enable_gyro** and *enable_accel** are set to False.  After ROS wrapper version 2.2.15, the IMU topics are disabled by default.  So if you need to access IMU topics, you should include the instructions below in your roslaunch statement or set gyro and accel to True in the launch file:

**enable_gyro:=true enable_accel:=true**

EricWiener, 2021-01-28:
@MartyG-RealSense thanks for the advice! I'm using the D455 only and not the tracking camera, so my IMU information is coming from the Husky robot. Is this still necessary to add these parameters?

Thank you for sharing that tutorial! That will be very helpful

MartyG-RealSense, 2021-01-28:
The D455 has a built-in IMU, and so using an external IMU on your Husky robot may not be necessary if the D455 is mounted to the robot.

Generally, the gyro and accel enable parameters are used with the built-in IMU of RealSense models that are equipped with one.   I am not aware of RealSense ROS cases where an external non-RealSense IMU has been used.

MartyG-RealSense, 2021-02-05:
Hi @EricWiener  Do you require further assistance with this case, please?  Thanks!

EricWiener, 2021-02-05:
@MartyG-RealSense sorry it's working now. I'll close the issue. Thanks for your help!

closed 2021-02-05 by EricWiener

MartyG-RealSense, 2021-02-05:
No problem at all.  Great to hear that it's working for you - thanks for the update!