realsenseai/realsense-ros #1655: Update realsense-ros version to 3.1.3 in meta-ros

sapoluri, 2021-01-29:
Please update the realsense-ros version in meta-ros or help me with the usual process by which you do the update so I can contribute.

MartyG-RealSense, 2021-01-29:
Hi @sapoluri   Could you provide more details please about what you are aiming to achieve?  It is possible that there is confusion arising from the fact that there have been two separate ROS2 wrappers for RealSense.  One of them was published on the main **Intel** GitHub and is now obsolete and no longer updated.  The other wrapper (the one that is actively updated and is the currently supported one) is here on the **realsense_ros** GitHib.  These two wrappers are not related to each other.

The RealSense ROS wrapper that should be used is the one on the **realsense_ros** GitHub.  The version **3.1.3** that you mention belongs to this wrapper.

https://github.com/IntelRealSense/realsense-ros/releases/tag/3.1.3

@doronhi - the developer of this wrapper - should be able to advise you further if you let us know in a comment below what your needs are.  Thanks!

sapoluri, 2021-01-29:
Hi @MartyG-RealSense, thank you for the response. I am trying to create a linux image using yocto dunfell release. I have included meta-ros/meta-ros2-foxy into my config files and built the linux image. The version of librealsense2 that I get is v2.34 and realsense-ros is v2.0.8. I need to upgrade librealsense2 to v2.39 and I wanted to upgrade realsense-ros version to v3.1.1 or higher.

I wanted the yocto scripts updated for everyone to use the updated realsense-ros and librealsense2. I am not sure who is responsible for updating those scripts but I would think it is important for Realsense's ease of use with embedded linux, that updating them would be necessary.

Who can I work with to get this done?

MartyG-RealSense, 2021-01-29:
Thanks very much @sapoluri for providing these details.  @doronhi the RealSense ROS wrapper developer will be the best person to discuss this subject with.  I have tagged them into this discussion.

It is important to match ROS wrapper version numbers as closely as possible with the librealsense version that is recommended for a particular wrapper version in the release notes.  If you are using **Foxy** then because the Foxy wrapper on this GitHub is new at the time of writing this, version **2.41.0** of the librealsense SDK that is recommended to be matched to Foxy wrapper version **3.1.3**.

sapoluri, 2021-01-29:
I will contact Doron. Thanks Marty!

closed 2021-01-29 by sapoluri