realsenseai/realsense-ros #1656: [camera/realsense2_camera_manager-2] process has died (Connection refused)

waterfl0w, 2021-01-30:
Hello,
when I try to run the `rs_camera.launch` file I run into the issue that the camera manager just dies.

This is run on Ubuntu 20.04 - 5.8.0-7630-generic ROS Noetic. The camera is a D435I.

The output is:
```
... logging to /home/jan/.ros/log/ba537b98-6338-11eb-a4a4-9ba919efef09/roslaunch-hydrogen-17118.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hydrogen:32905/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/confidence_fps: 30
 * /camera/realsense2_camera/confidence_height: 480
 * /camera/realsense2_camera/confidence_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters:
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
[... 294 lines cut ...]
  File "/usr/lib/python3.8/xmlrpc/client.py", line 1308, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python3.8/http/client.py", line 1250, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.8/http/client.py", line 1010, in _send_output
    self.send(msg)
  File "/usr/lib/python3.8/http/client.py", line 950, in send
    self.connect()
  File "/usr/lib/python3.8/http/client.py", line 921, in connect
    self.sock = self._create_connection(
  File "/usr/lib/python3.8/socket.py", line 808, in create_connection
    raise err
  File "/usr/lib/python3.8/socket.py", line 796, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

[rosmaster.master][INFO] 2021-01-30 21:22:46,011: -PUB [/rosout_agg] /rosout http://hydrogen:37991/
[rosmaster.master][INFO] 2021-01-30 21:22:46,012: -SUB [/rosout] /rosout http://hydrogen:37991/
[rosmaster.master][INFO] 2021-01-30 21:22:46,012: -SERVICE [/rosout/get_loggers] /rosout rosrpc://hydrogen:56543
[rosmaster.master][INFO] 2021-01-30 21:22:46,012: -SERVICE [/rosout/set_logger_level] /rosout rosrpc://hydrogen:56543
[rosmaster.master][INFO] 2021-01-30 21:22:46,013: -CACHEDPARAM [/rosout/omit_topics] by /rosout
[rosmaster.main][INFO] 2021-01-30 21:22:46,094: keyboard interrupt, will exit
[rosmaster.main][INFO] 2021-01-30 21:22:46,094: stopping master...
[rospy.core][INFO] 2021-01-30 21:22:46,094: signal_shutdown [atexit]
```

MartyG-RealSense, 2021-01-31:
Hi @waterfl0w   The error message in the roslaunch log - **Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!** - is what I would focus on first.  This suggests that there is a kernel related conflict.  The librealsense SDK only currently officially supports up to **kernel 5.4** rather than the 5.8 kernel that you are using.

After official support for Ubuntu 20.04 was introducied in librealsense version **2.40.0**, the SDK release notes suggested building librealsense with the **RSUSB backend** method.  This installation method requires an internet connection but is not dependent on Linux versions or kernel versions and does not require patching.

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2400

If librealsense is rebuilt then the RealSense ROS wrapper should also be built again afterwards.

waterfl0w, 2021-01-31:
Thank you. This solved the issue.

closed 2021-01-31 by waterfl0w

MartyG-RealSense, 2021-01-31:
Great news @waterfl0w - thanks for the update!