realsenseai/realsense-ros #1658: I want to know the contents of depth data

Hiroaki-K4, 2021-02-01:
Hi.
I'm receiving data from realsense-ros with a callback function on a different node. The code below.

```
void cameraCallback(const sensor_msgs::ImageConstPtr& rgb_msg, const sensor_msgs::ImageConstPtr& depth_msg)
{
  ROS_DEBUG("[YoloObjectDetector] USB image received.");

  cv_bridge::CvImagePtr rgb_image;
  cv_bridge::CvImagePtr depth_image;

  try {
    rgb_image = cv_bridge::toCvCopy(rgb_msg, sensor_msgs::image_encodings::BGR8);
    depth_image = cv_bridge::toCvCopy(depth_msg, sensor_msgs::image_encodings::TYPE_16UC1);
  } catch (cv_bridge::Exception& e) {
    ROS_ERROR("cv_bridge exception: %s", e.what());
    return;
  }

  if (rgb_image) {
    {
      imageHeader_ = rgb_msg->header;
      camImageCopy_ = rgb_image->image.clone();
      depthImageCopy_ = depth_image->image.clone();// std::cout << std::fixed << std::setprecision(19) << depthImageCopy_
      std::cout << std::setprecision(19) << depthImageCopy_ << std::endl;
    }
    ~~~~~~
  }
~~~~~~~~~~~~~~~~~
}
```
DepthImageCopy_ (cv :: Mat) is output on the terminal with std :: cout.
The output result is as below.

```
[4, 973, 973, 760, 354, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1093, 1093, 1091, 283, 56, 1003, 1078, 1075, 1071, 1063, 1057, 658, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 215, 828, 483, 484, 830, 830, 832, 832, 832, 832, 831, 830, 310, 421, 99, 790, 787, 786, 783, 779, 778, 777, 775, 772, 770, 769, 768, 768, 767, 767, 766, 765, 764, 763, 761, 760, 759, 759, 759, 758, 758, 757, 756, 755, 754, 754, 753, 752, 751, 751, 750, 750, 749, 749, 749, 748, 747, 746, 745, 743, 742, 741, 740, 740, 739, 738, 738, 738, 738, 738, 738, 737, 737, 737, 735, 736, 735, 191, 0, 0, 0, 0, 48, 627, 662, 661, 661, 661, 661, 660, 659, 658, 657, 657, 656, 657, 659, 660, 661, 661, 661, 661, 660, 657, 655, 653, 651, 648, 411, 644, 642, 642, 642, 642, 642, 643, 645, 646, 647, 648, 649, 649, 649, 649, 649, 649, 649, 649, 650, 651, 653, 654, 655, 657, 577, 0 .....................]
```
What does 0 mean? Does that mean 0m? Or will it be 0 when it exceeds some threshold?
Thanks!

MartyG-RealSense, 2021-02-01:
Hi @Hiroaki-K4   The first point that I would raise, having researched **setprecision** (the number of decimal points to provide number results at), is whether you meant to set it to '19' decimal points:  Did you mean to set it to '9' please?

**std::cout << std::setprecision(19)**

http://www.cplusplus.com/reference/iomanip/setprecision/

Does using 9 instead of 19 noticably change the output?

Hiroaki-K4, 2021-02-03:
I missed it.
I changed it to setprecision (9) and tried again, but the result did not change.

MartyG-RealSense, 2021-02-03:
I notice that part of your code has '//' in front of it.  This usually suggests a section of code that has been commented out so that the program ignores it.  Is the **std::cout << std::fixed << std::setprecision(19) << depthImageCopy_** part of the script meant to be ignored please?

[image]

Hiroaki-K4, 2021-02-09:
yes.
I forgot to delete it.

MartyG-RealSense, 2021-02-09:
I do not have much experience of the kind of script that you are using but after researching it, my understanding is that it is copying RGB and depth image data into sensor messages.

I still do not understand the meaning of the numbers that are output by **std :: cout** though unfortunately, so I am not able to tell if they are correct or not.  I do apologize.

Hiroaki-K4, 2021-02-13:
ok, thank you for the survey.

closed 2021-02-13 by Hiroaki-K4