realsenseai/realsense-ros #1660: Aligned Depth error - not representative in Real World space

derektan95, 2021-02-01:
Hi @doronhi and @MartyG-RealSense, 

I recently performed a calibration using `realsense-viewer` to improve my depth data from my Intel Realsense d435i. I used the following to obtain my depth cloud. 

`roslaunch -realsense2_camera rs_camera.launch align_depth:=true`

The red lines corresponds to laser scan from a 2d lidar that is very accurate. As expected, the depth cloud from `raw depth` topic lines up with the lidar scan well. 

[image]

Unfortunately, I observed something strange. When I subscribed to `aligned_depth_to_color` topic, the depth cloud seemed stretched out and very unaligned: 

[image]

Any insights to why this is so? I need the `aligned_depth_to_color` data to be both aligned to my RGB image, and yet representative of real world space. Thanks. 

-Derek

MartyG-RealSense, 2021-02-01:
Hi @derektan95   It looks as though the depth is being aligned to the size of the color field of view (FOV), which is a smaller size than the depth FOV on the D435 and D435i camera models.  

The D415 and D455 camera models have the same FOV size for both depth and color,

derektan95, 2021-02-02:
Ah i see. I must have gotten some configuration wrong. I will look into it. Thank you so much!

closed 2021-02-02 by derektan95

MartyG-RealSense, 2021-02-02:
Thanks very much @derektan95 for the update - good luck!