realsenseai/realsense-ros #1661: focal length and pixel pitch of realsense D455

Hiroaki-K4, 2021-02-03:
Hi.
What are the focal length and pixel pitch values of realsense D455?
Thanks!

MartyG-RealSense, 2021-02-03:
Hi @Hiroaki-K4   Does the information below from the February 2021 edition of the 400 Series data sheet document meet your needs please?

[image]

Hiroaki-K4, 2021-02-07:
yes, thank you!

closed 2021-02-07 by Hiroaki-K4

MartyG-RealSense, 2021-02-07:
Great to hear @Hiroaki-K4 - thanks very much for the update!

UserFLO, 2021-07-28:
Hi @MartyG-RealSense , what is the pixel size ? When i change resolution, that value changes too ? Thanks

MartyG-RealSense, 2021-07-28:
My understanding is that the pixel size on RealSense cameras is fixed and that as resolution is changed, the number of pixels in an image changes but not the size of the pixel itself.

https://github.com/IntelRealSense/librealsense/issues/6682

UserFLO, 2021-07-28:
Ok Thanks, can you confirm me the pixel size please ?

UserFLO, 2021-07-28:
@MartyG-RealSense if the pixel size doesn't change, does the focal length change with resolution ?

MartyG-RealSense, 2021-07-28:
Focal length is fixed, as confirmed in the comment linked to below.

https://github.com/IntelRealSense/librealsense/issues/7814#issuecomment-810695564

Pixel width and height as represented by fx and fy do change though depending on the currently selected resolution (sorry, I know it can be a confusing subject!)

You can find the fx and fy for a particular resolution on a particular RealSense camera model using the **rs-enumerate-devices** tool by launching it in calibration mode with **rs-enumerate-devices -c**

[image]

The link below also has examples of obtaining fx and fy with scripting.

https://github.com/IntelRealSense/librealsense/issues/2930

UserFLO, 2021-07-28:
Ok I understand. Sorry i have an another question. The focal length is given in pixel units. I want this results in meters. How can i do that ?
I know that 1 pixel = 3*10^-6 so i can have the focal length in meters with this methods ? Thanks

MartyG-RealSense, 2021-07-28:
Focal length for sensors is provided in **mm** in the data sheet document for the 400 Series cameras.  

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

You can quickly browse through the different types of sensors to find the focal length for a particular sensor by doing a 'find' operation in the data sheet for the term **focal length** 

Divide the mm value by 1000 to find the meter equivalent if you require the value in that measurement scale (e.g 1.88 / 1000 = 0.00188).

[image]

UserFLO, 2021-07-28:
If i take your first capture (in black). I know that a resolution of 1280*720 has a focal length = 644 pixel. I would like have this result in meters, (convert pixel to meters)

MartyG-RealSense, 2021-07-28:
I will refer your above question to Intel's @agrunnet as it is outside of my knowledge.

UserFLO, 2021-07-28:
I think it's the right method. Thank you

MartyG-RealSense, 2023-12-05:
Hi @pfcouto  1.88 mm refers to the focal length of the sensor hardware component, which is a fixed length that cannot be changed.  On the D435i model, 1.88 mm is the focal length of the **color** sensor, whilst the focal length of its left and right IR sensors is **1.93 mm**.

There is also an **image** focal length, whose pixel values (represented by fx and fy) can vary depending on which resolution is selected.  Information about the image focal length's pixel values can be found at the documentation link below.  The link explains that fx and fy pixel values are allowed to be different, though are commonly.close in value.

https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20#intrinsic-camera-parameters